espressif/arduino-esp32 #8766: Question about timer and interrupt crash

fabianoriccardi, 2023-10-14:
### Board

Lolin D32

### Device Description

Just the Lolin D32

### Hardware Configuration

no wiring

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

The program is basically the example RepeatTimer, but I have inserted a `Serial.println("") at line 18 in ISR. I know this is not a good practise, however I don't get why this code crashes.

Is `ARDUINO_ISR_ATTR` required to run this example?

### Sketch

```cpp
hw_timer_t * timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

void ARDUINO_ISR_ATTR onTimer(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  isrCounter++;
  lastIsrAt = millis();
  portEXIT_CRITICAL_ISR(&timerMux);
  // Give a semaphore that we can check in the loop
  xSemaphoreGiveFromISR(timerSemaphore, NULL);
  // It is safe to use digitalRead/Write here if you want to toggle an output

  Serial.println("Ciao");
}

void setup() {
  Serial.begin(115200);

  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Use 1st timer of 4 (counted from zero).
  // Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more
  // info).
  timer = timerBegin(0, 80, true);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer, true);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter)
  timerAlarmWrite(timer, 1000000, true);

  // Start an alarm
  timerAlarmEnable(timer);
}

void loop() {
  // If Timer has fired
  if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE){
    uint32_t isrCount = 0, isrTime = 0;
    // Read the interrupt count and time
    portENTER_CRITICAL(&timerMux);
    isrCount = isrCounter;
    isrTime = lastIsrAt;
    portEXIT_CRITICAL(&timerMux);
    // Print it
    Serial.print("onTimer no. ");
    Serial.print(isrCount);
    Serial.print(" at ");
    Serial.print(isrTime);
    Serial.println(" ms");
  }
}
```
```

### Debug Message

```plain
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
[    16][D][esp32-hal-cpu.c:244] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[    44][W][esp32-hal-timer.c:226] timerAttachInterruptFlag(): EDGE timer interrupt is not supported! Setting to LEVEL...
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1). 

Core  1 register dump:
PC      : 0x4008ac36  PS      : 0x00060535  A0      : 0x80089baa  A1      : 0x3ffbf19c  
A2      : 0x3ffb8314  A3      : 0x3ffb81a4  A4      : 0x00000004  A5      : 0x00060523  
A6      : 0x00060523  A7      : 0x00000001  A8      : 0x3ffb81a4  A9      : 0x00000018  
A10     : 0x3ffb81a4  A11     : 0x00000018  A12     : 0x00000004  A13     : 0x00060523  
A14     : 0x007bf318  A15     : 0x003fffff  SAR     : 0x0000000a  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400864b9  LEND    : 0x400864c9  LCOUNT  : 0xfffffffe  
Core  1 was running in ISR context:
EPC1    : 0x400dbedf  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace: 0x4008ac33:0x3ffbf19c |<-CORRUPTED

Core  0 register dump:
PC      : 0x4008adcf  PS      : 0x00060035  A0      : 0x800897d3  A1      : 0x3ffbea3c  
A2      : 0x3ffbf318  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023  
A6      : 0x00060021  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0xffffffff  
A10     : 0x3ffc1fa0  A11     : 0x00000000  A12     : 0x3ffc1f9c  A13     : 0x00000007  
A14     : 0x007bf318  A15     : 0x003fffff  SAR     : 0x0000001d  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x4008adcc:0x3ffbea3c |<-CORRUPTED

ELF file SHA256: 10c5f306d70bf32c
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-10-15:
You can use ets_printf() to print to serial in an interrupt.  Safe, but still not recommended because it is slow.  You have a short window to complete your work in an ISR.  If you don't actually need precise timing, and would like to be able to interact with hardware, you are better off using the FreeRTOS timer, which will run your code in the current stack.  See the [Ticker](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ticker/examples/Blinker/Blinker.ino) library for an example.

fabianoriccardi, 2023-10-15:
I didn't know `ets_printf()`... Is it part of esp-idf? Thanks for the hint, it works. 

Unfortunately I need to manage a low latency task, printfs are not needed but sometimes they may help.

May you point me why Serial.println() doesn't work? Is it because it is not placed in IRAM and the flash cache is disabled?

fabianoriccardi, 2023-10-15:
> delay is exceeding the ~300 microseconds allowed for an ISR to complete.

I didn't know this specific limitation, thanks!

closed 2023-10-15 by fabianoriccardi