espressif/arduino-esp32 #8767: Arduino Nano ESP32 not compiling due to issues with __FlashStringHelper

jpoll962, 2023-10-15:
### Board

Arduino Nano ESP32

### Device Description

Here is the link to the exact device: [link](https://store.arduino.cc/products/nano-esp32)

Board:
- Name: Arduino Nano ESP32
- SKU: ABX00092

Microcontroller: u-blox NORA-W106 (ESP32-S3)

### Hardware Configuration

Currently, I have an RFID Module plugged into the Arduino Nano ESP32.

Power: 
The Arduino Nano ESP32 is powered via the USB-C Connector to my computer.

Wiring: 
| Nano ESP32 | PN5180-NFC|
| :--------: | :-------: |
| VIN | +5V |
| 3.3V | +3.3V |
| D7 | RST |
| D10| NSS |
| D12 (COPI) | MOSI |
| D11 (CIPO) | MISO |
| D13 (SCK) | SCK |
| D9 | BUSY |
| GND | GND |

RFID Module:
- PN5180-NFC
-  [Link](https://www.amazon.com/Rakstore-PN5180-Sensor-ISO15693-Frequency/dp/B09G2KXPRL/ref=sr_1_3?crid=1MLHRYJ25P64V&keywords=PN5180+rfid+reader&qid=1697333808&sprefix=pn5180+rfid+reader%2Caps%2C113&sr=8-3)

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

Default

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am attempting to use an Arduino Nano ESP32 board instead of an Arduino UNO REV3. The code included has successfully run on an Arduino UNO REV3, with the only alterations being to the PN5180_NSS, PN5180_BUSY, and PN5180_RST definitions.

From my attempts at resolving the issue, I gathered the issue may lie somewhere in the `const __FlashStringHelper *strerror(ISO15693ErrorCode errno);` portion of the error and how Arduino Nano ESP32 reads flash memory.

The code is the example pulled from the following GitHub repository: [Link](https://github.com/ATrappmann/PN5180-Library/tree/master)

[... 267 lines cut ...]
C:\Users\<username>\OneDrive\Documents\Arduino\libraries\PN5180_Library/PN5180ISO15693.h:67:30: note: candidate: 'const __FlashStringHelper* PN5180ISO15693::strerror(ISO15693ErrorCode* (*)())'
   const __FlashStringHelper *strerror(ISO15693ErrorCode errno);
                              ^~~~~~~~
C:\Users\<username>\OneDrive\Documents\Arduino\libraries\PN5180_Library/PN5180ISO15693.h:67:30: note:   no known conversion for argument 1 from 'ISO15693ErrorCode' to 'ISO15693ErrorCode* (*)()'
C:\Users\<username>\AppData\Local\Temp\.arduinoIDE-unsaved2023914-29884-6tr55t.8t2qo\PN5180-Library\PN5180-Library.ino:238:37: error: no matching function for call to 'PN5180ISO15693::strerror(ISO15693ErrorCode&)'
       Serial.println(nfc.strerror(rc));
                                     ^
In file included from C:\Users\<username>\AppData\Local\Temp\.arduinoIDE-unsaved2023914-29884-6tr55t.8t2qo\PN5180-Library\PN5180-Library.ino:76:
C:\Users\<username>\OneDrive\Documents\Arduino\libraries\PN5180_Library/PN5180ISO15693.h:67:30: note: candidate: 'const __FlashStringHelper* PN5180ISO15693::strerror(ISO15693ErrorCode* (*)())'
   const __FlashStringHelper *strerror(ISO15693ErrorCode errno);
                              ^~~~~~~~
C:\Users\<username>\OneDrive\Documents\Arduino\libraries\PN5180_Library/PN5180ISO15693.h:67:30: note:   no known conversion for argument 1 from 'ISO15693ErrorCode' to 'ISO15693ErrorCode* (*)()'

exit status 1

Compilation error: no matching function for call to 'PN5180ISO15693::strerror(ISO15693ErrorCode&)'
```

### Other Steps to Reproduce

I have gotten this code to work on an Arduino UNO REV3 and Arduino UNO Wi-Fi REV2.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

jpoll962, 2023-10-16:
I mistakenly mixed up my CIPO and COPI pins. Properly swapping these two fixed the issue.

P-R-O-C-H-Y, 2023-10-16:
Closing as fixed by reporter.

closed 2023-10-16 by P-R-O-C-H-Y