espressif/arduino-esp32 #8770: XIAO Esp32-S3 cannot connect to Wifi networks and cannot generate AP

viniciusbarozzi, 2023-10-15:
### Board

XIAO Esp32-S3

### Device Description

[image]
[image]

### Hardware Configuration

Standard board

### Version

v2.0.14

### IDE Name

Arduino IDE 2.2.1

### Operating System

Win10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hello, I have several XIAO Esp32-s3 and most of them cannot connect to any Wifi network or open AP for WifiManager...
I've already tried some solutions, but without success.
- Update Arduino board and IDE package
-Board and IDE package downgrade
- Antenna replacement
- Several different Wifi networks
- Several .ino examples including WifiManager and simpler examples.
And several other modifications.

I'm asking for your help so we can try to resolve this.

At first I'm starting to believe that it could be the Esp32 package for Arduino made available by Espressif....

### Sketch

```cpp
#include "WiFi.h"

// Replace with your network credentials
const char* ssid = "XXXXXXXXXX";
const char* password = "XXXXXXXXX";

void initWiFi() {
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  WiFi.setTxPower(WIFI_POWER_8_5dBm);
  Serial.print("Connecting to WiFi ..");
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print('.');
    delay(1000);
  }
  Serial.println();
  Serial.println(WiFi.localIP());
}

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  delay(100);

  initWiFi();
}

void loop() {
  
}
```

### Debug Message

```plain
[   318][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): ConfigurConnecting to WiFi ....[  1978][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: BarozziEngineering, BSSID: 28:ee:52:27:4b:0c, Reason: 2
[  1979][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  1987][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  1993][D][WiFiGeneric.cpp:1081] _eventCallback(): WiFi Reconnect Running
[  2001][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
..[  3718][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: BarozziEngineering, BSSID: 28:ee:52:27:4b:0c, Reason: 2
[  3718][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  3726][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  3733][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
[  3741][V][WiFiGeneric.cpp:97] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
..[  5459][V][WiFiGeneric.cpp:362] _arduino_event_cb(): STA Disconnected: SSID: BarozziEngineering, BSSID: 28:ee:52:27:4b:0c, Reason: 2
[  5460][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
[  5467][W][WiFiGeneric.cpp:1061] _eventCallback(): Reason: 2 - AUTH_EXPIRE
[  5474][D][WiFiGeneric.cpp:1085] _eventCallback(): WiFi AutoReconnect Running
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-10-15:
S3 is working fine in general, it is NOT the Arduino framework

viniciusbarozzi, 2023-10-16:
@Jason2866 Do you have any idea what could be happening to cause these problems?

SuGlider, 2023-10-16:
@viniciusbarozzi  - does it work is `WiFi.setTxPower(WIFI_POWER_8_5dBm);` is commented out?

viniciusbarozzi, 2023-10-16:
@SuGlider It - doesn't work even without reducing the TX power. This was one of the alternatives I tried to solve it.

SuGlider, 2023-10-16:
I saw a similar issue in the past. It was related to the WiFi Security mode used in the WiFi Router/AP.
The minimum security mode for the network is WPA2. 
Please try using this:

```
WiFi.setMinSecurity(WIFI_AUTH_WEP); // Lower min security to WEP.
// or
WiFi.setMinSecurity(WIFI_AUTH_WPA_PSK); // Lower min security to WPA.
```

Check this issue: https://github.com/espressif/arduino-esp32/issues/8558
Trouble shooting documentation: https://github.com/espressif/arduino-esp32/blob/master/docs/source/troubleshooting.rst#wi-fi

Another possible issue could be that the WiFi SSID and PWD are both case sensitive. 
Check the upper/lower case letters in both.

viniciusbarozzi, 2023-10-16:
@SuGlider - If I configure WiFi.setMinSecurity(WIFI_AUTH_WEP), will the esp be able to connect in WEP and WPA, or only in WEP?

lbernstone, 2023-10-16:
minSecurity will have nothing to with setting up an AP.
What antenna do you have on here?  Perhaps it is not properly impedance matched?  If you remove the connector, put the device very close to your laptop wireless and set up an AP, can you see the SSID on the laptop?

SuGlider, 2023-10-16:
> @SuGlider - If I configure WiFi.setMinSecurity(WIFI_AUTH_WEP), will the esp be able to connect in WEP and WPA, or only in WEP?

Yes, this is the minimum. It will connect to any higher security level.
The problem is that if your WLAN is running WEP, by default, ESP32 won't connect to it, because it will demand a higher WiFi security level.

SuGlider, 2023-10-16:
> minSecurity will have nothing to with setting up an AP. What antenna do you have on here? Perhaps it is not properly impedance matched? If you remove the connector, put the device very close to your laptop wireless and set up an AP, can you see the SSID on the laptop?

Reading the issue example code, I understand that @viniciusbarozzi is setting up ESP32 as STA.

viniciusbarozzi, 2023-10-16:
@lbernstone - I'm using esp as a STA, to connect to my router. The antenna I'm using is the standard one that comes with the Seeed Studio V1.2 board.
[image]

viniciusbarozzi, 2023-10-16:
> > @SuGlider- Se eu configurar WiFi.setMinSecurity(WIFI_AUTH_WEP), o esp conseguirá se conectar em WEP e WPA, ou apenas em WEP?
> 
> Sim, este é o mínimo. Ele se conectará a qualquer nível de segurança superior. O problema é que se a sua WLAN estiver rodando WEP, por padrão, o ESP32 não se conectará a ela, pois exigirá um nível de segurança WiFi mais alto.

I'm going to test this option, but I remember having already tried something similar. But maybe on another level.

viniciusbarozzi, 2023-10-17:
The solution was to add WiFi.setMinSecurity(WIFI_AUTH_WEP) to the code. After that everything worked correctly. Thank you to those involved.

neorock07, 2024-12-18:
I have same problem before, my xiao esp32s3 couldn't connect to my cellphone hotspot, fortunately I could fix this issue, here my suggestion :
- If you upload code using ESP32 board manager 3.x.x version, you should change to version 2.x.x (2.0.11 version is tested works), go to `tools > board > board manager` search esp32 by espresif and install 2.0.11 version ...it must be fix your problem too.
- check your antenna, unplug-plug it 
- change Arduino IDE version (prefer to use 2.2.2 nights version)
- this is the last step if you still couldn't get fix your esp32, return to seller and ask to the new one. 

I hope you could fix that issue...thx

Kup135, 2025-01-10:
I tried everything suggested here and the Xiaos ESP32-S3 are still not connecting. I've found that after each of them connected to a network once, they stopped connecting again - making the system super unreliable. Has anyone else had a similar problem?

emalihin, 2025-01-20:
I've discovered a new issue with my esp32s3 from seeed. my device has worked flawlessly for a number of months, until I decided to change its firmware a bit, but nothing to do with wifi. since the first recent reflash in 90% of cases wifi only connects if i hold the antenna solder joint (same setup as [here](https://github.com/espressif/arduino-esp32/issues/8770#issuecomment-1765078731)). 

I've replaced the esp32s3 with the new board, replaced the antenna - no joy.

me-no-dev, 2025-01-20:
I have the board on my desk, my router uses WPA2 and I have no problems connecting to it or creating AP. With that said, this is not the best board out there. It has bad thermals and that is even more visible if the camera is attached. To the point where it can not stay running without active cooling (fan and radiator).

Jason2866, 2025-01-20:
We have some feedback of Tasmota users and this board. Well the experience is most other hardware boards designs are better. In short I would not buy...

emalihin, 2025-01-20:
any recommendations for similarly sized S3 boards?

me-no-dev, 2025-01-20:
@emalihin there is [this board on Amazon](https://www.amazon.de/-/en/Waveshare-Development-ESP32-S3FH4R2-Castellated-Applications-ESP32-S3-Mini-Dev-Board/dp/B0CHYHGYRH/ref=sr_1_13?crid=13WF9JBSDB00L&dib=eyJ2IjoiMSJ9.xojDFdXI1yM6rxSdk63rHysm2od3H2YfTrKXQh3go4bRxdTB1iJ8Dgod7dxO_SHLw0-jJx7QpiVfmB1sZAEcrhJDthKWkz6mX6774YDX4zOWGVqufads5IpnN8-P5A1_5Vpo_SQv-n10ONgdJTFw5wuJXu-OriMNfBeOFQxM6sCDOaGKWbb_Aot_9BUuONMZoC4xVl_zt9DGlpBjbgWuPjyJMzQg3OssMHbCELTU-tY.kziEwbKWZ4VDZxr_V_3gZXyp_IJe9d42wxkJVLlKvRs&dib_tag=se&keywords=esp32-s3&nsdOptOutParam=true&qid=1737405480&sprefix=esp32-s3%2Caps%2C109&sr=8-13&th=1) seems well received, but I know nothing beyond that. M5Stack also have some small S3 boards like [this one](https://shop.m5stack.com/products/m5stamp-esp32s3-module)

emalihin, 2025-01-20:
@me-no-dev thank you i'll check these out!

Curiously, while searching online for wifi issues for this board I came across a recommendation to lower the TX power. I set `WiFi.setTxPower(WIFI_POWER_2dBm);` and every of 10 restarts so far connected to wifi no probs.

me-no-dev, 2025-01-20:
bad antenna + the fact that more power equals more current and more heat. This is my guess. Something about it is not right. It's not using Espressif module, so the quality of everything is not known. Could be bad impedance matching too...

Jason2866, 2025-01-20:
The small boards design are on the limit for keeping the needed distance of the antenna (keep free zone). Not the only board which has problems with. 
Lolin and Lilygo did fail with there first small C3 boards

emalihin, 2025-01-20:
Makes sense. I like the fact that M5STAMPS3 has an on-board antenna, so I'll try that next as my design doesn't allow a more remote external antenna placement.

fabrizioPucci, 2025-04-08:
Hello, I have 6 internally produced boards with the same ESP32S3 Wroom 1 device. 
Some of them can easily connect to the PC hotspot, and some can't.
I tried reducing power, touching the antenna, and many other strategy and nothing worked.
Then I tried to use the Samsung S22 2.4GHz hotspot, and now, after being connected to the Samsung HotSpot, all of them are working fine.
In the past (if I remember correctly), I think that I have read something related to an internal memory with some wifi related parameters that could create some problem, and after the corrected connection, they have been fixed. 
Anyway, I cannot find it again, so maybe my experience is related to something else.