espressif/arduino-esp32 #8771: Timer Alarm doesn't fire if I set the same Alarm Value

fabianoriccardi, 2023-10-15:
### Board

Lolin D32

### Device Description

no other hardware

### Hardware Configuration

no wiring

### Version

v2.0.14

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

The timer alarm doesn't trigger if I set the Alarm Value at the previously set value. Of course, I reset the Timer Counter before writing the Alarm Timer.

In the following sketch, it triggers only one, and then it stops.

The sketch is almost similar to RepeatTimer, without autoreload. If you change the Line 23 `initialPeriod += 0;` to `initialPeriod += 1;`, the sketch works as expected (i.e. it prints once per second).

NOTE: this is just an experiment, in my application the real Alarm value is calculated dynamically, and it may be equal to the previous one.

### Sketch

```cpp
hw_timer_t * timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

int initialPeriod = 1000000;

void onTimer1(){
    // Increment the counter and set the time of ISR
    portENTER_CRITICAL_ISR(&timerMux);
    isrCounter++;
    lastIsrAt = millis();
    portEXIT_CRITICAL_ISR(&timerMux);

    // Give a semaphore that we can check in the loop
    xSemaphoreGiveFromISR(timerSemaphore, NULL);
    // It is safe to use digitalRead/Write here if you want to toggle an output

    timerWrite(timer, 0);
    initialPeriod += 0;
    timerAlarmWrite(timer, initialPeriod, false);
    timerAlarmEnable(timer);
    timerStart(timer);
}

void loop() {
  // If Timer has fired
  if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE){
    uint32_t isrCount = 0, isrTime = 0;
    // Read the interrupt count and time
    portENTER_CRITICAL(&timerMux);
    isrCount = isrCounter;
    isrTime = lastIsrAt;
    portEXIT_CRITICAL(&timerMux);
    // Print it
    Serial.print("onTimer no. ");
    Serial.print(isrCount);
    Serial.print(" at ");
    Serial.print(isrTime);
    Serial.println(" us");
  }
}

void setup() {
  Serial.begin(115200);

  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Use 1st timer of 4 (counted from zero).
  // Set 80 divider for prescaler (see ESP32 Technical Reference Manual for more
  // info).
  timer = timerBegin(0, 80, true);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer1, true);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter)
  timerAlarmWrite(timer, 1000000, false);
  // Start an alarm
  timerAlarmEnable(timer);
  
  timerStart(timer);
}
```

### Debug Message

```plain
onTimer no. 1 at 1045 us
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-10-17:
@fabianoriccardi Can you test it with the latest dev release 3.0.0-alpha2?
The Timer got refactored also because of changes in ESP-IDF 5.x, so there is a new API. Be sure to use new RepeatTimer example where the new functions are used.

fabianoriccardi, 2023-10-17:
On v3.0.0-alpha2 it works as expected (a print every second). Below the code I have tested:

```
/*
 Repeat timer example

 This example shows how to use hardware timer in ESP32. The timer calls onTimer
 function every second. The timer can be stopped with button attached to PIN 0
 (IO0).

 This example code is in the public domain.
 */

// Stop button is attached to PIN 0 (IO0)
#define BTN_STOP_ALARM    0

hw_timer_t * timer = NULL;
volatile SemaphoreHandle_t timerSemaphore;
portMUX_TYPE timerMux = portMUX_INITIALIZER_UNLOCKED;

volatile uint32_t isrCounter = 0;
volatile uint32_t lastIsrAt = 0;

int initialPeriod = 1000000;

void ARDUINO_ISR_ATTR onTimer(){
  // Increment the counter and set the time of ISR
  portENTER_CRITICAL_ISR(&timerMux);
  isrCounter = isrCounter + 1;
  lastIsrAt = millis();
  portEXIT_CRITICAL_ISR(&timerMux);
  // Give a semaphore that we can check in the loop
  xSemaphoreGiveFromISR(timerSemaphore, NULL);
  // It is safe to use digitalRead/Write here if you want to toggle an output
  timerWrite(timer, 0);
  initialPeriod += 0;
  timerAlarm(timer, initialPeriod, false, 0);
}

void setup() {
  Serial.begin(115200);

  // Set BTN_STOP_ALARM to input mode
  pinMode(BTN_STOP_ALARM, INPUT);

  // Create semaphore to inform us when the timer has fired
  timerSemaphore = xSemaphoreCreateBinary();

  // Set timer frequency to 1Mhz
  timer = timerBegin(1000000);

  // Attach onTimer function to our timer.
  timerAttachInterrupt(timer, &onTimer);

  // Set alarm to call onTimer function every second (value in microseconds).
  // Repeat the alarm (third parameter) with unlimited count = 0 (fourth parameter).
  timerAlarm(timer, 1000000, false, 0);
}

void loop() {
  // If Timer has fired
  if (xSemaphoreTake(timerSemaphore, 0) == pdTRUE){
    uint32_t isrCount = 0, isrTime = 0;
    // Read the interrupt count and time
    portENTER_CRITICAL(&timerMux);
    isrCount = isrCounter;
    isrTime = lastIsrAt;
    portEXIT_CRITICAL(&timerMux);
    // Print it
    Serial.print("onTimer no. ");
    Serial.print(isrCount);
    Serial.print(" at ");
    Serial.print(isrTime);
    Serial.println(" ms");
  }
  // If button is pressed
  if (digitalRead(BTN_STOP_ALARM) == LOW) {
    // If timer is still running
    if (timer) {
      // Stop and free timer
      timerEnd(timer);
      timer = NULL;
    }
  }
}
```

P-R-O-C-H-Y, 2023-10-18:
@fabianoriccardi Thank you for testing on version 3.0.0-alpha2 and verifying that its already fixed there. I am closing this issue marking as solved.

closed 2023-10-18 by P-R-O-C-H-Y

fabianoriccardi, 2023-10-18:
Won't you solve it on branch v2.x?

P-R-O-C-H-Y, 2023-10-18:
Our main focus is now on 3.0.0 version and if there it works, there is no action needed. On 2.X branch the last 2.0.14 release is supposed to be the last version and it will be updated only for critical bugs, security issues, etc..