espressif/arduino-esp32 #8775: ESP32-CAM coredump partition error

Algonsi, 2023-10-16:
### Board

ESP32-CAM

### Device Description

ESP32-CAM

### Hardware Configuration

Nothing more

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows 11 

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

1500000

### Description

I'm programming an ESP32-CAM and I think I don't need the coredump feature. So I don't want to reserve a place for coredump in the partition table.
How can I configure coredump in Arduino framework in order to disable it and not have the error 'No coredump partition found!"?

### Sketch

```cpp
I use a partition table without coredump partition
```

### Debug Message

```plain
E() esp_core_dump_flash: No core dump partition found!
```

### Other Steps to Reproduce

No problem for compiling. Just error message when launching run

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-10-16:
Take a look here: [¶](https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/partition_table.html?highlight=partion%20table#partition-table)

Algonsi, 2023-10-16:
Thank you Vojtech. I look there but I can't find the answer to my question.
I want to **disable** core dump feature to avoid to reserve place in partition table.
It must be enabled by default or it would not show the error.

lbernstone, 2023-10-17:
I think this must be a platformio issue.  I don't get this error in Arduino IDE.

Algonsi, 2023-10-18:
> I think this must be a platformio issue. I don't get this error in Arduino IDE.

Do you how to disable the core dump feature in ESP32 in Arduino framework?

lbernstone, 2023-10-18:
If you don't use coredump, it won't be enabled.  AFAIK, esp-insights is the only piece that requires it.

Algonsi, 2023-10-18:
It must be enabled by default or it would not show the error. And I have not enabled it in config or program

ChrisAnthenum, 2023-10-18:
I had this same issue. Also my Adafruit feather (mini N2R4) was showing up as an ESP32-c3 for some reason… anyway I got it working by going through the factory reset process on Adafruit and changing the partition scheme under tools to default 4MB with spiffs. Seems like the latest doesn’t like TinyUF2… idk

VojtechBartoska, 2023-11-29:
Hello @Algonsi, can we close this issue?

me-no-dev, 2023-11-30:
@Algonsi coredump is not enabled, only partition was added for it, because it is used by some of the modules. Remove the coredump partition from your chosen `partitions.csv`, put it in your sketch folder and recompile.

Algonsi, 2023-11-30:
Thank you me-no-dev
I use a partition table without coredump partition
No problem for compiling. Program runs ok, but  I'd like to get rid of the error message when launching program:
"E() esp_core_dump_flash: No core dump partition found!"
The key to the problem is that it complains that it can't find the coredump partition

me-no-dev, 2023-11-30:
we can not support both ways unfortunately. It can be done in IDF, but here we use a static configuration that can not be altered.

Algonsi, 2023-12-01:
There is the answer
Thank you!

MorTnMor, 2023-12-27:
@Algonsi Ummm，sorry? Where is the answer? I can't find it

Algonsi, 2023-12-28:
"It can be done in IDF, but here we use a static configuration that can not be altered." i.e. there is no solution

uraannnn, 2024-04-28:
solved the problem?I have the same problem.