espressif/arduino-esp32 #8781: esp_camera.h lib with SC031GS camera, not enabled for Arduino.

Eduardo-Brandt, 2023-10-18:
### Related area

Software Arduino support for camera

### Hardware specification

ESP32-S3

### Is your feature request related to a problem?

Have anyone tried to use the SC031GS camera with the esp_camera.h library in Arduino?

I always get the following camera error:

> E (23) camera: Camera probe failed with error 0x105(ESP_ERR_NOT_FOUND)
> Camera init failed with error 0x105Camera ERROR!
> Camera capture failed.
> Camera capture failed.
> Camera capture failed.

I realized that in the config switch is turned ON for it, but in the ESP-IDF build for Arduino-ESP32 2.0.11 (which is the version currently used in PlatformIO), this is turned off:

https://github.com/espressif/arduino-esp32/blob/2.0.11/tools/sdk/esp32/sdkconfig#L1810
```
CONFIG_OV2640_SUPPORT=y
CONFIG_OV3660_SUPPORT=y
CONFIG_OV5640_SUPPORT=y
CONFIG_GC2145_SUPPORT=y
CONFIG_GC032A_SUPPORT=y
CONFIG_GC0308_SUPPORT=y
CONFIG_BF3005_SUPPORT=y
CONFIG_BF20A6_SUPPORT=y
# CONFIG_SC101IOT_SUPPORT is not set   <<<<<<<<<<<<<<<<<<<< this
CONFIG_SC030IOT_SUPPORT=y
# CONFIG_SC031GS_SUPPORT is not set
# CONFIG_SCCB_HARDWARE_I2C_PORT0 is not set
CONFIG_SCCB_HARDWARE_I2C_PORT1=y
CONFIG_SCCB_CLK_FREQ=100000
# CONFIG_GC_SENSOR_WINDOWING_MODE is not set
CONFIG_GC_SENSOR_SUBSAMPLE_MODE=y
CONFIG_CAMERA_TASK_STACK_SIZE=2048
CONFIG_CAMERA_CORE0=y
# CONFIG_CAMERA_CORE1 is not set
# CONFIG_CAMERA_NO_AFFINITY is not set
CONFIG_CAMERA_DMA_BUFFER_SIZE_MAX=32768
```

**I ask, is it possible to compile this lib with SC031GS enabled for use in Arduino on Platformio?**

Can the community fix this issue, please?
Thanks a lot!!

### Describe the solution you'd like

Compiling esp_camera.h library with SC031GS enabled for use in Arduino on Platformio?

### Describe alternatives you've considered

I can´t change esp_camera.h lib in Arduino as it is precompiled. 

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2023-10-18:
Highly unlikely it is still disabled in core 2.0.14 https://github.com/espressif/arduino-esp32/blob/2.0.14/tools/sdk/esp32/sdkconfig#L1816

Platformio releases the original release builds from espressif

Eduardo-Brandt, 2023-10-19:
Have anyone any suggestion or workaround idea to put this SC031GS camera working with Arduino?

VojtechBartoska, 2023-10-19:
Hi, @me-no-dev should know more details.

closed 2025-01-15 by Parsaabasi