espressif/arduino-esp32 #8786: assert failed: multi_heap_free multi_heap_poisoning.c:253

hemeleerse, 2023-10-20:
### Board

M5Stack-Core2

### Device Description

Nothing connected

### Hardware Configuration

Nothing connected

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

240

### PSRAM enabled

yes

### Upload speed

115200

### Description

System restarts in a random way. Can't link the restart to any activity, action, input.
Backtrace shows link to the AsyncWebServer I'm running on the device.
Any hint on where to start looking is welcome.

### Sketch

```cpp
I can 't isolate the part that generates the issue
```

### Debug Message

```plain
CORRUPT HEAP: Bad head at 0x3ffd0018. Expected 0xabba1234 got 0xad040002

assert failed: multi_heap_free multi_heap_poisoning.c:253 (head != NULL)

Backtrace:0x40083b61:0x3ffd96200x4008fb6d:0x3ffd9640 0x400954ed:0x3ffd9660 0x40095133:0x3ffd9790 0x40084021:0x3ffd97b0 0x4009551d:0x3ffd97d0 0x400f2baa:0x3ffd97f0 0x400f2bb9:0x3ffd9810 0x400e190b:0x3ffd9830 0x400e1e7e:0x3ffd9850 0x400e1e89:0x3ffd9870 0x400e0c92:0x3ffd9890 0x400e2d75:0x3ffd98b0 0x400de7fb:0x3ffd98d0 0x400de807:0x3ffd98f0 0x4019846e:0x3ffd9910 0x4019878e:0x3ffd9940 0x401987a4:0x3ffd9960 0x4019892e:0x3ffd9980 

  #0  0x40083b61:0x3ffd9620 in panic_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:402
  #1  0x4008fb6d:0x3ffd9640 in esp_system_abort at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:128
  #2  0x400954ed:0x3ffd9660 in __assert_func at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/assert.c:85
  #3  0x40095133:0x3ffd9790 in multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:253
      (inlined by) multi_heap_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/multi_heap_poisoning.c:245
  #4  0x40084021:0x3ffd97b0 in heap_caps_free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/heap/heap_caps.c:340
  #5  0x4009551d:0x3ffd97d0 in free at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/newlib/heap.c:39
  #6  0x400f2baa:0x3ffd97f0 in String::invalidate() at C:/Users/erikh/.platformio/packages/framework-arduinoespressif32@src-f2ea83e2545300b10a69ff44ef9dc6cd/cores/esp32/WString.cpp:175
  #7  0x400f2bb9:0x3ffd9810 in String::~String() at C:/Users/erikh/.platformio/packages/framework-arduinoespressif32@src-f2ea83e2545300b10a69ff44ef9dc6cd/cores/esp32/WString.cpp:159
  #8  0x400e190b:0x3ffd9830 in AsyncAbstractResponse::~AsyncAbstractResponse() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebResponseImpl.h:42
  #9  0x400e1e7e:0x3ffd9850 in AsyncFileResponse::~AsyncFileResponse() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebResponses.cpp:478
  #10 0x400e1e89:0x3ffd9870 in AsyncFileResponse::~AsyncFileResponse() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebResponses.cpp:481
  #11 0x400e0c92:0x3ffd9890 in AsyncWebServerRequest::~AsyncWebServerRequest() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebRequest.cpp:89 (discriminator 1)
  #12 0x400e2d75:0x3ffd98b0 in AsyncWebServer::_handleDisconnect(AsyncWebServerRequest*) at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebServer.cpp:102 (discriminator 1)
  #13 0x400de7fb:0x3ffd98d0 in AsyncWebServerRequest::_onDisconnect() at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebRequest.cpp:228
  #14 0x400de807:0x3ffd98f0 in std::_Function_handler<void (void*, AsyncClient*), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#6}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at .pio/libdeps/m5stack-core2/ESP Async WebServer/src/WebRequest.cpp:74
      (inlined by) _M_invoke at c:\users\erikh\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:297
  #15 0x4019846e:0x3ffd9910 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at c:\users\erikh\.platformio\packages\toolchain-xtensa-esp32\xtensa-esp32-elf\include\c++\8.4.0\bits/std_function.h:687
  #16 0x4019878e:0x3ffd9940 in AsyncClient::_fin(tcp_pcb*, signed char) at .pio/libdeps/m5stack-core2/AsyncTCP/src/AsyncTCP.cpp:888
      (inlined by) AsyncClient::_fin(tcp_pcb*, signed char) at .pio/libdeps/m5stack-core2/AsyncTCP/src/AsyncTCP.cpp:885
  #17 0x401987a4:0x3ffd9960 in AsyncClient::_s_fin(void*, tcp_pcb*, signed char) at .pio/libdeps/m5stack-core2/AsyncTCP/src/AsyncTCP.cpp:1195
  #18 0x4019892e:0x3ffd9980 in _async_service_task(void*) at .pio/libdeps/m5stack-core2/AsyncTCP/src/AsyncTCP.cpp:162
      (inlined by) _async_service_task at .pio/libdeps/m5stack-core2/AsyncTCP/src/AsyncTCP.cpp:194
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-10-21:
You have a String object that is disappearing before you can use it.  Don't pass objects between functions.  Treat anything out of your local context as a constant (or put mutexes on them).  Finish your handler routines quickly.

mrengineer7777, 2023-11-02:
Looks like a variable has gone out of scope or a pointer no longer refers to valid memory.  If you are doing async data processing, you may need to malloc memory for storage and free the memory once you are done.  I have to do that on certain messages when they are qued and processed on a different thread.

TD-er, 2023-11-22:
Is it possible your ESP receives a connection request from some client, tries to allocate some memory to handle it and then the client no longer replies or the network connection of the ESP gets disconnected?

Other use cases where you end up with crashes in the String destructor is when you either memcpy a String object (which you shouldn't do for ovbious reasons) or construction failed of the String object.
Typically this last one may occur when using `emplace_back` on std:: containers which hold objects with a `String` member.

I can also imagine strange things may happen with classes/structs using inheritance and calling member functions which do stuff on `String` objects from their constructor/destructor.

EmbeddedDevver, 2024-08-01:
Same issue with WiFi and AsyncTCP lib. Can't isolate it yet, sporadically happens....

TD-er, 2024-08-01:
Can you try to make a number of calls (e.g. frequent refreshing of a webpage) to decrease available free memory, then leave it alone and just periodically print the free memory on the serial port.

I think you will see this memory be freed after ~ 2 minutes.
If this is what is happening, then it probably has nothing to do with AsyncTCP.

Ni2000co, 2024-08-20:
I had the same problem but with WiFi from Arduino. I could only instance the WiFiClient once so I passed it through a function parameter for other classes to use and that created a copy of the WiFiClient. The problem is that WiFiClient uses async functions and the copy of WiFiClient got in the way of the other WiFiClient.
I then passed the WiFiClient as an adress with & and not as a copy and then everything worked.

closed 2025-01-15 by Parsaabasi