espressif/arduino-esp32 #8788: The littleFS example for PlatformIO is not working anymore :-(

FlightControl-User, 2023-10-21:
### Related area

Example not working

### Hardware specification

Arduino Nano ESP32

### Is your feature request related to a problem?

I have experienced that the example for the littleFS is not working properly. Maybe it's me but the littleFS example compiles but does not mount the littleFS file system! It is also very confusing and it requires a more decent explanation of how th example is built and how it works. And more important, what actions are to be done on the Arduino device itself!

Please check, i'm happy to share my experiences if any.

### Describe the solution you'd like

A better explanation or a review of the issue.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Jason2866, 2023-10-21:
You find a working example here https://github.com/Jason2866/platform-espressif32/tree/Tasmota/20x/examples/arduino-littlefs

VojtechBartoska, 2023-10-23:
Hi, seems this is not related directly with Arduino core, did you solve your problem @FlightControl-User by provided link above?

closed 2024-01-31 by VojtechBartoska