espressif/arduino-esp32 #8790: [ledC] Please add Arduino support for gamma curve fade

TD-er, 2023-10-22:
### Related area

IDF5.x

### Hardware specification

ESP32

### Is your feature request related to a problem?

Found hardware gamma fade support for ledC in the ESP-IDF 5.1 code.
See: 
- Example code: https://github.com/espressif/esp-idf/blob/8fc8f3f47997aadba21facabc66004c1d22de181/examples/peripherals/ledc/ledc_gamma_curve_fade/main/ledc_gamma_curve_fade_example_main.c#L189C21-L189C52
- Library code: https://github.com/espressif/esp-idf/blob/master/components/driver/ledc/include/driver/ledc.h#L580

### Describe the solution you'd like

The new Arduino implementation now finally manages pin to channel mapping in 1 place, so it makes sense to also use this abstraction layer for other ledc functions in the IDF code.

What I would like to see:
- Function to pre-compute a gamma curve, analogue to `ledc_fill_multi_fade_param_list`  (preferrably using an array of `ledc_fade_param_config_t` as this doesn't need extra memory allocations when using it)
- Function to set an array of pre-computed values like the `ledc_set_multi_fade` function.  (this technically allows to also set different intervals per LUT value)
- Function to set an user defined array of pre-computed values, so someone can define their own LUT.
- Function to 'inspect' the set fade cycle, like the `ledc_read_fade_param` function. Maybe a bit like the Arduino function `ledcRead`
- Function to de-allocate the computed LUT array and preferrably this can also be called when calling to detach.

Typical use cases for this:
- Fade LEDs, which need a gamma curve per LED color
- Ramp up/down a servo
- Function generator to use the ESP as a lab tool to generate reproducible curves

Some nice ideas:

An array of `ledc_fade_param_config_t` as being used in the `ledc_set_multi_fade` function can be manually tweaked.
For example you could change the `step_num` to optimize a curve or extend its run time.
Like pre-program the ramp-up, stable value, ramp-down so you still have high fade step-frequency when values change and not using very long arrays.

If this isn't going to be added, can you please make a function to get the current `speed_mode` and `channel` of an attached pin?
This way we can still keep using one central administration of which pin is attached to what timer and channel and also use the currently (in Arduino) unsupported ledc functions.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-10-22:
@P-R-O-C-H-Y  PTAL

P-R-O-C-H-Y, 2023-10-25:
Just a quick note, that this feature is only supported on ESP32C6 and ESP32H2.

TD-er, 2023-10-25:
Oh too bad, didn't know that.
I also saw that `SOC_LEDC_GAMMA_CURVE_FADE_SUPPORTED` is apparently set for the P4.
Would be nice to also have some kind of RTOS scheduled callback wrapper to process a next value for those platforms not supporting a hardware LUT.

PR #11464 (2025-06-16): feat(LEDC): Add Gamma Fade support and enhance auto channel/timer selection for multi-group
## Description of Change
This pull request introduces several updates to the LEDC (LED Controller) functionality in the ESP32 HAL, focusing on improving compatibility, expanding functionality, and enhancing error handling. The most significant changes include replacing hardcoded channel group sizes with `SOC_LEDC_CHANNEL_NUM`, adding support for gamma curve fades, and improving error handling in key LEDC operations.

### New Gamma Curve Fade Functionality:
* Added gamma curve fade support with customiza [... cut]
closed 2025-06-16 by me-no-dev