espressif/arduino-esp32 #8791: platform.txt treat warnings N O T as errors

StefanL38, 2023-10-22:
### Board

all ESP32

### Device Description

all ESP32

### Hardware Configuration

all

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40

### PSRAM enabled

no

### Upload speed

115200

### Description

If this GitHub thing is able to show ALL issues on a SINGLE page or if each ISSUE-page would have a link how to limit sarch to THIS projects issues
I would do checking the issues

I checked page 1 to 5. and then stopped.
Haven't found this issue so started a new one.

In the file
...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\platform.txt

there are some settings that make the compiler treat warnings as errors that cause the compiler to abort with an error-message.

compiler.warning_flags.more=-Wall -Werror=all
compiler.warning_flags.all=-Wall -Werror=all -Wextra

this should be changed to
compiler.warning_flags.more=-Wall
compiler.warning_flags.all=-Wall -Wextra

best regards Stefan 

### Sketch

```cpp
sketch
```

### Debug Message

```plain
FUCK UP with having to fill up all these windows even if they are not relevant to post an issue

That is the opposite of making it easy top contribute
```

### Other Steps to Reproduce

sdljflksjdflkj

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-22:
You are not inspiring us much to help you... but anyway... why not just select lower warning level? It's easy to overlook a warning if it does not cause the compilation to fail, so for higher error levels we have opted to treat warnings as errors. As benefit, there is rarely a warning in our codebase.

closed 2023-10-22 by me-no-dev

aWZHY0yQH81uOYvH, 2024-02-15:
I would agree that this behavior is annoying. I like to develop with warnings turned up so the compiler can help catch bugs I may accidentally write. When developing, often you might comment out some code to test things, but then have your compilation fail with `-Werror=unused-but-set-variable`, which is very annoying. I want to know about that, but not have it stop the entire compilation; it should be a *warning*.

> It's easy to overlook a warning if it does not cause the compilation to fail

The default AVR Arduino core doesn't have `-Werror`, and I would argue that if the user has explicitly asked for warnings to be generated by changing the setting in the Arduino IDE, they would be looking out for them in the compilation output. Having a warning-free core codebase is a different issue, and IMO shouldn't be enforced through this user-facing option.