espressif/arduino-esp32 #8792: undefined reference to `vtable for fs::FileImpl' in debug builds with -O0 flag.

yizakat, 2023-10-23:
### Board

ESP32-S3

### Device Description

esp32proS3 : Unexpected Maker.

### Hardware Configuration

Not relevant.

### Version

v2.0.14

### IDE Name

VSCode - ESP-IDF extension.

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am using Arduino as a component linked to an ESP-IDF project.  I get the following linker error when using the SPIFFS file system object and building with the O0 (no optimisation) sdkconfig setting.

Linker error "undefined reference to `vtable for fs::FileImpl'"

This has been noted in other forums/git repos, usually PlatformIO or Arduino IDE context and appears to be related to a change in the behaviour of the GCC toolchain.  Switching optimisation back to Og resolves the problem but makes the debugger unusable, presumably most people on those forums are not using GDB.

I am using the matching version of the ESP-IDF tooling for the Arduino component which packages the following toolchain version: esp-2021r2-patch5-8.4.0

Is there a later (or earlier) version I should be using that resolves this issue?  I can work my way back but it would be quicker if there is a known point at which the problem was introduced.

### Sketch

```cpp
#include <Arduino.h>
#include <SPIFFS.h>

const static char *TAG="TEST";

extern "C" void app_main(void)
{
    initArduino();    

    if(!SPIFFS.begin(true))
    {
        ESP_LOGE(TAG,"Error enabling SPIFFS filesystem\n");
    }
    else
    {
        ESP_LOGI(TAG,"SPIFFS mounted OK\n");
    }
}
```

### Debug Message

```plain
[5/7] Linking CXX executable main.elf
FAILED: main.elf
cmd.exe /C "cd . && C:\Users\jason\.espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\main.elf.rsp -o main.elf  && cd ."      
c:/users/jason/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: esp-idf/arduino-esp32/libarduino-esp32.a(vfs_api.cpp.obj):(.literal._ZN2fs8FileImplD5Ev[_ZN2fs8FileImplD5Ev]+0x0): undefined reference to `vtable for fs::FileImpl'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1
```

### Other Steps to Reproduce

RTTI is disabled.

Changing the app debug flags to 0g resolves the issue, so presumably optimises the problem out somehow.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-23:
@igrr any clues why this would be happening?

Jason2866, 2023-10-23:
Will happen if the PR https://github.com/espressif/arduino-esp32/pull/8768 is not applied.

yizakat, 2023-10-23:
That looks the same, as soon as any optimisation is enabled the problem goes away, but I'm using 4.4.6 and the PR refers specifically to 5.1.  I went with 4.4.6 as it aligned with the version mentioned in the 2.0.14 Arduino Release.

What are the options here?  Should I switch to 5.1?  The project I'm working on is not time-boxed so I can probably afford time to work on a non-release branch. 

My apologies if the answers are obvious, I'm new to the ESP32, IDF (and Arduino).  I usually stomp around in the ARM Cortex world with CMSIS ;-) for work.

yizakat, 2023-10-23:
This is the same but the version I'm using is older 4.4.6, so this has actually been around a while.   As noted, it only manifests if there is no optimisation which presumably optimises out the unused/unreferenced methods without the pure-virtual decl.

I've applied the PR locally and all fine now.

closed 2023-10-23 by yizakat