espressif/arduino-esp32 #8795: Reboot issue using ESP32C3 MCU with Arduino IDE

benniflash, 2023-10-23:
### Board

Custom board with ESP32C3 F4 MCU

### Device Description

ESP32C3 F4 MCU used to communicate via SPI and I2C with several sensors. issue when trying to communicate via I2C with only one sensor.

### Hardware Configuration

GPIO 8 SDA GPIO 5 SCL

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

- 👋 Hi, I’m @benniflash
- 👀 I’m using Arduino IDE with Esp32 C3 MCU, in order to get the outpuit of several STMicroelectronics and Sensirion's sensors. Everything worked fine, until I added a new sensors, using the same I2C lines used for the others.

In order to communicate via I2C i used GPIO 5 and GPIO 8, whose, as I've just said, work correctly for the other sensors.

Do you have any idea about this error? The MCU reboot every time I try to read some specific registers of that sensor. 

### Sketch

```cpp
#define INT_DRDY  7

const byte led_gpio = 20;
//Inizializzazione I2C per TMOS, LPS22DF, SHT40, MAG E SGP40 (?) 
#include <Wire.h>
#define SDA 8
#define SCL 5

#define IIS2MDC_ADDRESS 0x1E

void ReadByte_i2c(uint8_t slave_addr, uint8_t reg, uint8_t bytestoread, uint8_t *data_out);
void get_mag_data(uint8_t *data_mag);

void get_mag_data(uint8_t *data_mag){
  //Read TMOS data output
  ReadByte_i2c(IIS2MDC_ADDRESS, 0x69, 7, data_mag);
}

 
 while(1){
  delay(500);
  get_mag_data(data_mag);
}

this is the part of the code involved in my issue (not my entire sketch)
```

### Debug Message

```plain
- I get this error:

Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

ELF file SHA256: c365a5e532cb6c93

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382508
SPIWP:0xee
mode:DIO, clock div:2
load:0x3fcd5810,len:0x38c
load:0x403cc710,len:0x6ac
load:0x403ce710,len:0x25f4
entry 0x403cc710
E (104) esp_core_dump_flash: Core dump data check failed:
Calculated checksum='4b08dbc1'
Image checksum='73e2708b'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

VojtechBartoska, 2023-10-23:
Hi, just to make sure, you are on master branch which means 3.0.0-aplha2, righ?

benniflash, 2023-10-23:
Yes

SuGlider, 2023-10-31:
>Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0). 

I just can tell that this is a WatchDog Timer (WDT) problem.
Some task may be getting running for too long and not returning or allowing other tasks to run.

Not enough information to help...

lbernstone, 2023-10-31:
Very likely caused by a I2C device stretching.
https://github.com/espressif/arduino-esp32/issues?q=is%3Aissue+i2c+stretching

VojtechBartoska, 2023-12-11:
Closing as expired doe to no answer from the author, if needed you can reopen.

closed 2023-12-11 by VojtechBartoska