espressif/arduino-esp32 #8796: 3.0.0 version Migration Related issues

VojtechBartoska, 2023-10-23:


TD-er, 2023-10-23:
I think this is also related: #8774

everslick, 2023-10-27:
The OneWire lib does not compile anymore, because `GPIO` is not defined. I tried including `driver/gpio.h` but this did not solve it.

```
#if ESP_ARDUINO_VERSION_MAJOR >= 3
#include <driver/gpio.h>
#else
#include <driver/rtc_io.h>
#endif
```

```
libs/OneWire/OneWire.h:167:17: error: 'GPIO' was not declared in this scope; did you mean 'PI'?
  167 |         return (GPIO.in >> pin) & 0x1;
      |                 ^~~~
      |                 PI
```

me-no-dev, 2023-10-27:
> The OneWire lib does not compile anymore, because GPIO is not defined.

Probably needs particular include. Some have changed in IDF5

everslick, 2023-10-28:
I have difficulty porting the following code to IDF5. Please look at the line below **FIXME! HELP NEEDED!**. 

```
error: cannot convert 'wifi_interface_t' to 'esp_netif_t*' {aka 'esp_netif_obj*'}
  esp_netif_get_ip_info(WIFI_IF_AP, &info);
```

Snippet:

```
#ifdef ESP32
#ifdef IDF5
  typedef esp_netif_ip_info_t ip_info_t;
#else
  typedef tcpip_adapter_ip_info_t ip_info_t;
#endif
#endif
#ifdef ESP8266
  typedef struct ip_info ip_info_t;
#endif

static ip_info_t ap_ip_info(void) {
  ip_info_t info = { 0 };

#ifdef ESP32
#ifdef IDF5
  // FIXME! HELP NEEDED!
  // esp_netif_get_ip_info(WIFI_IF_AP, &info);
#else // !IDF5
  tcpip_adapter_get_ip_info(TCPIP_ADAPTER_IF_AP, &info);
#endif // !IDF5
#endif // ESP32

#ifdef ESP8266
  wifi_get_ip_info(SOFTAP_IF, &info);
#endif

  return (info);
}
```

Are `wifi_interface_t` and `esp_netif_t` incompatible types or can I just take the address of the former and cast it to the latter?

lbernstone, 2023-10-28:
@everslick https://github.com/espressif/esp-idf/blob/v5.1/examples/protocols/http_server/captive_portal/main/dns_server.c#L152

everslick, 2023-10-29:
Thanks for all the help so far. Things went relatively smoothly until now and I have a firmware that boots. Congrats to the devs for that release, great work!!!

One issue left is, that my firmware `aborts()` on the first boot after an update. (meaning: I flash a new build via OTA and reboot). Every boot after that first one is fine. It is always only the first boot after a firmware change that aborts. I think it is related to WiFi. Maybe it does some radio recalibration and runs out of stack? On the serial console it dumps:

```
[09:37:54] NET:  initializing WiFi stack (STA=1, AP=1)
[09:37:54] NET:  configuring network using DHCP
[09:37:54] NET:  unhandled WiFi event code '0'

Stack smashing protect failure!

Backtrace: 0x40083901:0x3ffb40c0 0x4008fde5:0x3ffb40e0 0x40083912:0x3ffb4100 0x400d9f72:0x3ffb4120 0x400daabb:0x3ffb41a0 0x400dabfe:0x3ffb4230 0x400d47a7:0x3f
fb4250 0x400d4663:0x3ffb4270 0x400d3640:0x3ffb4290

ELF file SHA256: 524fe429d4c52cd7

Rebooting...
```

The lines with the timestamps are from my firmware. The decoded stack is as follows:

```
0x40083901: panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:452
0x4008fde5: esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:84
0x40083912: __stack_chk_fail at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/stack_check.c:28
0x400d9f72: sta_connect() at /home/clemens/Devel/ESP/emonio-fw/src/net.cpp:673
0x400daabb: sta_start() at /home/clemens/Devel/ESP/emonio-fw/src/net.cpp:719
0x400dabfe: net_init() at /home/clemens/Devel/ESP/emonio-fw/src/net.cpp:1426
 (inlined by) net_init() at /home/clemens/Devel/ESP/emonio-fw/src/net.cpp:1365
0x400d47a7: emonio_init() at /home/clemens/Devel/ESP/emonio-fw/src/emonio.cpp:63
0x400d4663: setup() at /home/clemens/Devel/ESP/emonio-fw/build/3.0.99/xuan/eng/alpha/emonio.ino.cpp:5
0x400d3640: loopTask(void*) at /home/clemens/Devel/ESP/emonio-fw/sdk/Arduino-ESP32-3.0.0/cores/esp32/main.cpp:55
```

I tried to increase the LOOP_STACK_SIZE from 12KB to 16KB (I already set it to 12KB for CORE 2.0.0), but this does not change anything.

Any ideas about that?

everslick, 2023-10-29:
OK, it's the call to `WiFi.setHostname()` that smashes the stack. but only on first boot.

`setHostname()` is infamous for needing to be called in different places relative to `begin()` and `connect()` depending on platform (ESP8266 and ESP32) and IDF version. At least in my experience and after doing a lot of experimentation.

Is there a recommended place to call `setHostname()` for CORE 3.0.0 ?

everslick, 2023-10-29:
One VERY peculiar thing I just witnessed: Calling `yield()` in a very tight loop seems to crash with exception 29 (LoadProhibited), while PC points to a few lines above the call to `yield()` which makes it especially difficult to debug. Luckily I have `yield()` wrapped in my own `system_yield()` so I can switch codepaths between versions. Like:

```
void system_yield(void) {
#ifdef IDF5
  delay(1);
#else
  yield();
#endif

#ifdef HAVE_WATCHDOG
  watchdog_feed();
#endif
}
```

When the tight loop, `yield()` is called in, gets slowed down enough (with e.g. debug prints) it does not crash.

me-no-dev, 2023-10-30:
@everslick I am rewriting/restructuring the whole network stack in order to decouple WiFi from the rest of the network functions. Many things have changed and will be fixed/rewritten to work. Some APIs will change/get deprecated, but it's for the better.

everslick, 2023-10-30:
> @everslick I am rewriting/restructuring the whole network stack in order to decouple WiFi from the rest of the network functions. Many things have changed and will be fixed/rewritten to work. Some APIs will change/get deprecated, but it's for the better.

Yes, I've seen the issue for it. I'm very much looking forward to it!

While we are at it. One of the things I do not really like about the Arduino API is the omnipresent instantiation of singleton like APIs. We have NO_GLOBAL_INSTANCES for that, but this is not always implemented for all interfaces. In case of `WiFi`, it is so baked in, that nothing works anymore if the global `WiFi` object is not there. Maybe, if possible, this is something to consider, when reworking the API.

me-no-dev, 2023-10-30:
WiFi would be able to not be global, but beware a new `Network` class will be, though it's very light and has only events and dns queries.

me-no-dev, 2023-10-31:
@Jason2866 IPv6 is planned for 3.0.0/1 it's coming for sure :)

kapyaar, 2024-01-02:
Not sure if this belongs here, w5500+ETH using latest git version, installed manually on windows 11. Running Example ETH_W5500_Arduino_SPI. I have two boards, very possible they have two different versions of W5500. One works fine, but one gives the following very randomly.

```
E (127) w5500.mac: W5500 version mismatched, expected 0x04, got 0xfe
E (128) w5500.mac: emac_w5500_init(795): vefiry chip ID failed
E (128) esp_eth: esp_eth_driver_install(228): init mac failed
E (3033) w5500.mac: w5500_send_command(204): send command timeout
E (3033) w5500.mac: emac_w5500_receive(675): issue RECV command failed
E (3034) w5500.mac: frame read from module failed
E (4139) w5500.mac: w5500_send_command(204): send command timeout
E (4139) w5500.mac: emac_w5500_receive(675): issue RECV command failed
E (4140) w5500.mac: frame read from module failed
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00380038  PS      : 0x00060930  A0      : 0x00000000  A1      : 0x3ffb5140  
A2      : 0x3ffb4960  A3      : 0x00380038  A4      : 0x40110378  A5      : 0x3ffb5ac4  
A6      : 0x000005f2  A7      : 0x00000000  A8      : 0x80110778  A9      : 0x3ffb5100  
A10     : 0x3ffb59d0  A11     : 0x3ffb5ac4  A12     : 0x0000fffc  A13     : 0x0000fffc  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00380038  LBEG    : 0x40089948  LEND    : 0x4008995e  LCOUNT  : 0xffffffff  

Backtrace: 0x00380035:0x3ffb5140
ELF file SHA256: 2dd9d9d7fcebe907

Rebooting...
ets Jun  8 2016 00:22:57
```

If I repower the board, Sometimes it starts working. Other times, it may give a different set of error messages, like this one. 

```
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1256
load:0x40078000,len:13832
load:0x40080400,len:4
load:0x40080404,len:3048
entry 0x40080590
E (127) w5500.mac: w5500_reset(302): reset timeout
E (128) w5500.mac: emac_w5500_init(793): reset w5500 failed
E (128) esp_eth: esp_eth_driver_install(228): init mac failed
E (3033) w5500.mac: w5500_send_command(204): send command timeout
E (3033) w5500.mac: emac_w5500_receive(675): issue RECV command failed
E (3034) w5500.mac: frame read from module failed
E (4139) w5500.mac: w5500_send_command(204): send command timeout
E (4139) w5500.mac: emac_w5500_receive(675): issue RECV command failed
E (4140) w5500.mac: frame read from module failed
Guru Meditation Error: Core  0 panic'ed (InstrFetchProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x00380038  PS      : 0x00060930  A0      : 0x00000000  A1      : 0x3ffb5140  
A2      : 0x3ffb4960  A3      : 0x00380038  A4      : 0x40110378  A5      : 0x3ffb5ac4  
A6      : 0x000005f2  A7      : 0x00000000  A8      : 0x80110778  A9      : 0x3ffb5100  
A10     : 0x3ffb59d0  A11     : 0x3ffb5ac4  A12     : 0x0000fffd  A13     : 0x0000fffd  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00380038  LBEG    : 0x40089948  LEND    : 0x4008995e  LCOUNT  : 0xffffffff  

Backtrace: 0x00380035:0x3ffb5140
ELF file SHA256: 2dd9d9d7fcebe907

Rebooting...
ets Jun  8 2016 00:22:57
```

And when it works, it connects and works fine, goes like this.

```
ETH Started
ETH Connected
ETH Got IP: 'eth0'
eth0: <UP,100M,FULL_DUPLEX,AUTO>
      ether 36:AB:95:6B:07:90 phy 0x1
      inet 192.168.2.30 netmask 255.255.255.0 broadcast 192.168.2.255
      gateway 192.168.2.1 dns 192.168.2.1

1
connecting to arduino.tips
HTTP/1.1 200 OK
Date: Tue, 02 Jan 2024 18:41:16 GMT....
```
And the rest of the contents are printed from /asciilogo.txt. I use wificlientsecure to do https.

Does  https://github.com/espressif/esp-idf/issues/11331 has any bearing on this?

me-no-dev, 2024-01-04:
@kostaond could you please comment on @kapyaar issue? Here the "bad" version is 0xFE

kostaond, 2024-01-04:
@kapyaar how is the W5500 to the ESP32? Is it custom board or is it connected via wires? It seems to me like some SPI connection issue. You can also try to decrease SPI CLK frequency to see if it helps.

kapyaar, 2024-01-05:
@kostaond @me-no-dev Board is good, it is an assembled pcb, but the issue is totally on our side. This ETH section is part of a larger board, and the new version had a few additional SPI chips, CS of one of those chips was floating, likely responding to requests meant for W5500. Took me a while to think outside my section of the board, Fixed that part, and the ETH part works consistently. Really sorry. I have an ETH question, but unrelated to this. To keep this thread clean, let me know if you want me to delete my post and comment, I will be happy to do that.

kostaond, 2024-01-05:
@kapyaar if you have Ethernet related question, please ask at [forum](https://www.esp32.com/).

TD-er, 2024-01-05:
@kapyaar 
One tip for designing your board with Ethernet.
Please include a ferrite bead in the power line to the ETH power and make sure you only have a single point where GND of Ethernet and the rest is connected.
In my own designs I used a beefy 0 Ohm resistor (0805 or 1206) to connect both grounds.

You can have a look at the schematics provided by Olimex to get inspiration on how to implement the power nets when using Ethernet.
N.B. this is regardless of the used Ethernet chip as this is just typical for all Ethernet modules/chips.

TD-er, 2024-01-06:
Found another issue, where the DHCP service doesn't seem to work when starting softAP.
See: https://github.com/espressif/arduino-esp32/issues/9069

everslick, 2024-01-06:
I2C HAL fails to compile if HAL LOCKS are disabled.
See: #9073

luc-github, 2024-02-18:
Hello in latest git / 3.0.0 alpha I cannot find the API equivalent API of 
```
/**
 * @brief Compatible version of former tcpip_adapter API of esp_netif_get_sta_list
*/
esp_err_t tcpip_adapter_get_sta_list(const wifi_sta_list_t *wifi_sta_list, tcpip_adapter_sta_list_t *tcpip_sta_list);
```
to get IP of connected clients

It seems now a component in IDF :
 https://github.com/espressif/esp-idf/blob/release/v5.1/components/esp_wifi/include/esp_wifi_ap_get_sta_list.h
 
 But cannot find it in latest git 
 
 Will this API be ported / available in arduino for final 3.0.0 ?

me-no-dev, 2024-02-19:
@luc-github no. IDF 5.x has many changes since 4.x. There should be something else that would work the same way though

luc-github, 2024-02-19:
so you mean `esp_err_t esp_wifi_ap_get_sta_list_with_ip(const wifi_sta_list_t *wifi_sta_list, wifi_sta_mac_ip_list_t *wifi_sta_ip_mac_list);` I linked ?
but I cannot find it in latest git, so it is may be a miss ?

lucasssvaz, 2024-03-27:
@P-R-O-C-H-Y @me-no-dev Should we add the changes of the network refactoring to the migration guide ?

Jason2866, 2024-03-27:
@lucasssvaz not one of the tagged, but from my side a clear YES. This changes have broken currently before well running Tasmota with alpha 3.0.0 completely ;-)

me-no-dev, 2024-03-27:
@Jason2866 can you elaborate a bit more on what actually broke on your end? We did rename a couple of IPv6 methods, but that should be mostly it. I tried to keep all other old methods working as before.

me-no-dev, 2024-03-28:
@Jason2866 this is unexpected. Can you give more info on how you get to `WiFi.disconnect(true)`? Do you call it first on boot, or do you call other WiFi APIs before that? I want to replicate the problem.

Jason2866, 2024-03-28:
@me-no-dev  
@s-hadinger is investigating this. His finding so far
> struggling with the Core3 Wifi. It connects but Tasmota considers it's not connected because the status remains stuck on `WL_STOPPED`.
It looks like the `STAClass` never receives `WIFI_EVENT_STA_START`/`ARDUINO_EVENT_WIFI_STA_START` that changes the status from `WL_STOPPED` to `WL_DISCONNECTED` because the `STAClass` is initialized too late, long after STA more was set and long after event `WIFI_EVENT_STA_START`.

> I'm still struggling whether it uses `STAClass` or `WiFiSTAClass` (or worse both).

me-no-dev, 2024-03-28:
@Jason2866 this is hard to believe, because event listeners are attached before STA is started. WiFiSTAClass is there just for compatibility and it always calls the STAClass in the background. STAClass is the one that manages the interface and receives and handles it's events. https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/STA.cpp#L293-L302

If you can provide a chain of calls that can reproduce an issue, I can investigate further as well.

Jason2866, 2024-03-28:
@me-no-dev Thx, we will do. We need some time.

s-hadinger, 2024-03-28:
I traced back the chain of calls (and simplified a bit).

```C
// We have a global static singleton (actually we go through a WiFi32 wrapper but it does not change the chain of calls)
WiFiClass WiFi;

// then we make calls in this order:
WiFi.persistent(false);
WiFi.enableIPv6(true);
WiFi.hostname(TasmotaGlobal.hostname);  // ESP32 needs this here (before WiFi.mode) for core 2.0.0
WiFi.forceSleepWake(); // Make sure WiFi is really active.

WiFi.mode(WIFI_STA);
-> just after this call, the ARDUINO_EVENT_WIFI_STA_START event is fired but there is no listener because `STA` instance is not initialized yet

WiFi.setSleepMode(WIFI_NONE_SLEEP);       // Disable sleep
WiFi.setOutputPower((float)(Settings->wifi_output_power) / 10);
WiFi.setAutoReconnect(true);
WiFi.hostname(TasmotaGlobal.hostname);  // ESP8266 needs this here (after WiFi.mode)

WiFi.begin(SettingsText(SET_STASSID1 + Settings->sta_active), SettingsText(SET_STAPWD1 + Settings->sta_active), channel, Wifi.bssid);
--> this is only now that `WiFiSTAClass::begin` and `STAClass::begin` are called and hence register the event listener
```

As you can see above, since we never called `WiFi.begin`, `STAClass::begin` was never called and the listener is not yet registered. When `WiFi.mode(WIFI_STA);` is called, the event `ARDUINO_EVENT_WIFI_STA_START` is lost.

Does that mean that we should call `WiFi.begin()` (with no parameters) before doing `WiFi.mode(WIFI_STA);`, and later call again `WiFi.begin(...)` with all STA parameters ?

s-hadinger, 2024-03-28:
@me-no-dev To give more details about event handlers:

The event handler is registered only when calling `STAClass::begin(bool tryConnect)`

https://github.com/espressif/arduino-esp32/blob/654aeada0a4d88595c31a68c62d339a09794c381/libraries/WiFi/src/STA.cpp#L291-L294

This registers `_sta_event_cb` as LwIP handler, which in turns converts to Arduino event in `_onStaEvent`.

Since `_sta_event_cb` is not registered when we call `WiFi.mode(WIFI_STA);`, the event `WIFI_EVENT_STA_START` is never sent to the `Network` instance with type `ARDUINO_EVENT_WIFI_STA_START`

s-hadinger, 2024-03-28:
One last thing I forgot to mention, we have a `sleep(200)` after `WiFi.mode(WIFI_STA);`.

When looking at examples like:
```C
void setup(){
    Serial.begin(115200);
    WiFi.disconnect(true);
    WiFi.onEvent(WiFiEvent);  // Will call WiFiEvent() from another thread.
    WiFi.mode(WIFI_MODE_APSTA);
    //enable ap ipv6 here
    WiFi.softAPenableIPv6();
    WiFi.softAP(AP_SSID);
    //enable sta ipv6 here
    WiFi.enableIPv6();
    WiFi.begin(STA_SSID, STA_PASS);
}
```

would it be possible that `WiFi.begin()` is called fast enough that LwIP didn't generate before this call?

me-no-dev, 2024-03-28:
let me think about it a bit. The problem is that `WiFi.mode(WIFI_MODE_APSTA);` would start the radio but not attach AP/STA to callbacks. If you do not use `mode` but call `begin` or `softAP`, then things would be OK. I'll think about how to attach also through `mode`

Jason2866, 2024-03-29:
another #9435

Jason2866, 2024-03-29:
@me-no-dev Currently Lib Builder is running including the PR. Will report asap

Jason2866, 2024-03-29:
@me-no-dev nope still crashes with #9436
```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x401668e9  PS      : 0x00060b30  A0      : 0x80138e54  A1      : 0x3ffb1b80  
A2      : 0x00000000  A3      : 0x0000002b  A4      : 0x00000001  A5      : 0x3ffc9790  
A6      : 0x00000020  A7      : 0x00000000  A8      : 0x4012e340  A9      : 0x3ffb1b60  
A10     : 0x3ffb1bb0  A11     : 0x3ffb1b88  A12     : 0x3ffc9790  A13     : 0x3ffd4320  
A14     : 0x00000000  A15     : 0x00060323  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4008e691  LEND    : 0x4008e6a1  LCOUNT  : 0xfffffffd  

Backtrace: 0x401668e6:0x3ffb1b80 0x40138e51:0x3ffb1bf0 0x401635bc:0x3ffb1ce0 0x40167cd5:0x3ffb1d00 0x400e17ed:0x3ffb1d20 0x400e3769:0x3ffb1e90 0x400da4f7:0x3ffb2220 0x4015a3c3:0x3ffb2290
```

me-no-dev, 2024-03-29:
@Jason2866 this is a different issue than what @s-hadinger outlined above. I have an idea for the one you are testing, but I need to first reproduce it. I'm starting/stopping/deleting... I can not get into your situation

Jason2866, 2024-03-29:
@me-no-dev  Before the PR #9436 adding`WiFi.begin();` before `WifiSetMode(WIFI_STA);` was a workaround to get it going. With the PR this workaround isn't working anymore and the crash i posted happens. So the PR introduces somehow this behaviour.

me-no-dev, 2024-03-29:
@Jason2866 please try the PR again. just pushed something to help this exact problem. Here is what I am testing and is not failing:
```cpp
  WiFi.mode(WIFI_MODE_STA);
  WiFi.disconnect(true);
  WiFi.enableIPv6();
  WiFi.begin("ssid","pass");
  delay(1000);
  WiFi.disconnect(true);
```

What is `WifiSetMode(WIFI_STA)`?

Jason2866, 2024-03-29:
`WifiSetMode(WIFI_STA) == WiFi.mode(WIFI_STA);` We use a wrapper...
Will test with the change.

me-no-dev, 2024-03-29:
@Jason2866 please enable verbose debug. I really can not tell what is going on. BTW all fine on my end
```
[   784][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[   857][W][STA.cpp:571] disconnect(): STA already disconnected.
[   858][V][STA.cpp:210] _onStaEvent(): STA Started
[   867][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[   875][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
STA Started
E (1758) wifi:NAN WiFi stop
[   887][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
[   919][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[   940][V][STA.cpp:210] _onStaEvent(): STA Started
[   945][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[   952][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
STA Started
[   999][V][STA.cpp:227] _onStaEvent(): STA Connected: SSID: nbis-test, BSSID: 74:4d:28:c6:7a:fc, Channel: 8, Auth: WPA_WPA2_PSK
  GPIO : BUS_TYPE[bus/unit][chan]
[  1012][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 13 - STA_CONNECTED
[  1022][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 13 - STA_CONNECTED
[  1025][V][NetworkInterface.cpp:65] _onIpEvent(): sta Got New IP: 192.168.254.84 MASK: 255.255.255.0 GW: 192.168.254.1
[  1041][V][STA.cpp:149] _onStaArduinoEvent(): Enabled IPv6 Link Local on sta
STA Connected
[  1051][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 16 - STA_GOT_IP
[  1060][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 16 - STA_GOT_IP
[  1067][D][STA.cpp:194] _onStaArduinoEvent(): STA IP: 192.168.254.84, MASK: 255.255.255.0, GW: 192.168.254.1
sta: <UP,nbis-test,CH:8,RSSI:-41,N,WPA_WPA2_PSK>
      ether 30:C6:F7:F4:2F:8C
      inet 192.168.254.84 netmask 255.255.255.0 broadcast 192.168.254.255
      gateway 192.168.254.1 dns 192.168.254.3
      inet6 fe80::32c6:f7ff:fef4:2f8c%st2 type LINK_LOCAL

[  2758][V][NetworkInterface.cpp:107] _onIpEvent(): IF sta Got IPv6: Interface: 0, IP Index: 0, Type: LINK_LOCAL, Zone: 3 (st2), Address: fe80:0000:0000:0000:32c6:f7ff:fef4:2f8c
[  2774][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 17 - STA_GOT_IP6
[  2782][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 17 - STA_GOT_IP6
sta: <UP,nbis-test,CH:8,RSSI:-41,N,WPA_WPA2_PSK>
      ether 30:C6:F7:F4:2F:8C
      inet 192.168.254.84 netmask 255.255.255.0 broadcast 192.168.254.255
      gateway 192.168.254.1 dns 192.168.254.3
      inet6 fe80::32c6:f7ff:fef4:2f8c%st2 type LINK_LOCAL

[  9013][V][STA.cpp:235] _onStaEvent(): STA Disconnected: SSID: nbis-test, BSSID: 74:4d:28:c6:7a:fc, Reason: 8
[  9023][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 14 - STA_DISCONNECTED
[  9031][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 14 - STA_DISCONNECTED
[  9039][W][STA.cpp:157] _onStaArduinoEvent(): Reason: 8 - ASSOC_LEAVE
STA Disconnected
E (9921) wifi:NAN WiFi stop
[  9051][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF

```

Jason2866, 2024-03-29:
okay that helps
```
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:344
load:0x40078000,len:14216
load:0x40080400,len:4
load:0x40080404,len:3232
entry 0x4008057c
[     1][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x40161d94
[    12][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x40161db0
[    26][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x40161dcc
[    39][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x40161de8
[    53][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x40161d94
[    66][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x40161db0
[    79][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x40161dcc
[    93][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x40161de8
[   106][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RX (2) successfully set to 0x40161d94
[   120][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_TX (3) successfully set to 0x40161db0
[   133][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_CTS (4) successfully set to 0x40161dcc
[   147][V][esp32-hal-periman.c:229] perimanSetBusDeinit(): Deinit function for type UART_RTS (5) successfully set to 0x40161de8
[   162][D][esp32-hal-cpu.c:259] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
[   171][W][esp32-hal-psram.c:67] psramInit(): PSRAM init failed!
[   189][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 3 successfully set to type UART_RX (2) with bus 0x3ffbdf84
[   199][V][esp32-hal-periman.c:154] perimanSetPinBus(): Pin 1 successfully set to type UART_TX (3) with bus 0x3ffbdf84
[   212][V][esp32-hal-uart.c:396] uartBegin(): UART0 baud(115200) Mode(800001c) rxPin(3) txPin(1)
[   223][V][esp32-hal-uart.c:482] uartBegin(): UART0 not installed. Starting installation
[   234][V][esp32-hal-uart.c:525] uartBegin(): UART0 initialization done.

00:00:00.037 HDW: ESP32-D0WDQ6 v1.0 
00:00:00.050 UFS: FlashFS mounted with 12596 kB free
00:00:00.062 CFG: Loaded from File, Count 12
[   287][E][vfs_api.cpp:105] open(): /debug does not exist, no permits for creation
[   296][E][vfs_api.cpp:105] open(): /debug does not exist, no permits for creation
[   306][E][vfs_api.cpp:105] open(): /debug.bec does not exist, no permits for creation
[   315][E][vfs_api.cpp:105] open(): /debug.bec does not exist, no permits for creation
[   325][E][vfs_api.cpp:105] open(): /debug.be does not exist, no permits for creation
[   334][E][vfs_api.cpp:105] open(): /debug.be does not exist, no permits for creation
00:00:00.144 BRY: Berry initialized, RAM used 4446 bytes
[   359][E][vfs_api.cpp:105] open(): /preinit.be does not exist, no permits for creation
[   370][E][vfs_api.cpp:105] open(): /preinit.bec does not exist, no permits for creation
00:00:00.169 Project sonoff - Tasmota Version 13.4.0.3(arduino30)-3_0_0(2024-03-29T15:39:33)
[   395][E][vfs_api.cpp:105] open(): /autoexec.be does not exist, no permits for creation
[   405][E][vfs_api.cpp:105] open(): /autoexec.bec does not exist, no permits for creation
[   541][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[   618][V][STA.cpp:210] _onStaEvent(): STA Started
[   623][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[   630][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[  7025][V][WiFiGeneric.cpp:77] _arduino_event_cb(): SCAN Done: ID: 128, Status: 0, Results: 9
[  7034][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 10 - SCAN_DONE
[  7513][W][STA.cpp:571] disconnect(): STA already disconnected.
[  7524][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
[  7737][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[  7747][V][STA.cpp:210] _onStaEvent(): STA Started
[  7752][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[  7759][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[  8046][E][STA.cpp:406] connect(): STA not started! You must call begin() first.
00:00:08.277 WIF: Connecting to AP1 Jason_Home_WLAN Channel 11 BSSId 00:A0:57:2A:BD:19 in mode 11n as sonoff-000000-7272...
00:00:15.819 WIF: Connect failed with AP timeout
00:00:29.824 WIF: Connect failed with AP timeout
[ 37509][V][WiFiGeneric.cpp:77] _arduino_event_cb(): SCAN Done: ID: 129, Status: 0, Results: 8
[ 37518][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 10 - SCAN_DONE
[ 38101][W][STA.cpp:571] disconnect(): STA already disconnected.
[ 38112][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
[ 38325][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[ 38335][V][STA.cpp:210] _onStaEvent(): STA Started
[ 38340][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[ 38347][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[ 38634][E][STA.cpp:406] connect(): STA not started! You must call begin() first.
00:00:38.865 WIF: Connecting to AP1 Jason_Home_WLAN Channel 11 BSSId 00:A0:57:2A:BD:19 in mode 11n as sonoff-000000-7272...
```

Jason2866, 2024-03-29:
@s-hadinger  You will get maaany warnings but it works with adding to the env build_flags
I have this env
```
[env:tasmota32-arduino30]
extends                 = env:arduino30
board                   = esp32
board_build.f_flash     = 80000000L
board_build.f_cpu       = 240000000L
board_build.flash_mode  = qio
build_unflags           = ${env:arduino30.build_unflags}
build_flags             = ${env:arduino30.build_flags}
                          -DFIRMWARE_ARDUINO30
                          -DCORE_DEBUG_LEVEL=5
monitor_filters         = esp32_exception_decoder
lib_ignore              = ${env:arduino30.lib_ignore}
```

me-no-dev, 2024-03-29:
yup. I see where it fails. just a sec

s-hadinger, 2024-03-29:
Let me try with some Tasmota logs in between to see why this happens.

me-no-dev, 2024-03-29:
@s-hadinger in a mean time I pushed an update to the PR that makes sure that STA and AP receive the STOP event. Don't know if this helps you

s-hadinger, 2024-03-29:
The event `[   618][V][STA.cpp:210] _onStaEvent(): STA Started` is not present in the normal Tasmota code, it was generated by the additional `WiFi.begin();` before `WifiSetMode(WIFI_STA);` (as far as I can see).

s-hadinger, 2024-03-29:
Now connection to Wifi does not succeed anymore and we got periodically:

```

00:00:28.238 WIF: Attempting connection...
[ 28443][W][STA.cpp:575] disconnect(): STA already disconnected.
[ 28454][V][STA.cpp:214] _onStaEvent(): STA Stopped
[ 28459][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 12 - STA_STOP
[ 28466][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[ 28477][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
[ 28491][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[ 28504][V][STA.cpp:210] _onStaEvent(): STA Started
[ 28509][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[ 28516][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[ 28524][E][STA.cpp:370] connect(): STA not started! You must call begin() first.
[ 28931][E][STA.cpp:410] connect(): STA not started! You must call begin() first.
```

Wifi state is still always stuck to `WL_STOPPED` which is obviously not good.

me-no-dev, 2024-03-29:
@s-hadinger ok, but how to reproduce? There is obviously something different between our build setups. Your errors do not make sense to what our code does and I can't come up with a way to reproduce a single issue you have. What do we do?

me-no-dev, 2024-03-29:
Here are some pointers:
- `STA_START` can happen only when WiFi STA was started (`WiFi.mode(WIFI_MODE_STA)` or `WiFi.STA.begin()` OORR direct IDF API calls were made)
- If either of the above get called, `_esp_netif` would not be NULL, therefore `You must call begin() first.` will never be the case. The ONLY  exception is if IDF APIs were called directly, in which case none of our states would be updated.

me-no-dev, 2024-04-02:
@s-hadinger @Jason2866 have you been able to figure out why your results are so different?

s-hadinger, 2024-04-02:
No time, sorry. I will investigate later this week

me-no-dev, 2024-04-02:
@s-hadinger in that case I will merge the WiFi.mode fix, so you can test directly on master

s-hadinger, 2024-04-05:
@me-no-dev I can easily reproduce the problem:

Here is the test code, in `setup()`
```C

  AddLog(LOG_LEVEL_INFO, ">>>: WiFi.mode(WIFI_MODE_STA) WiFi.status()=%i", WiFi.status());
  WiFi.mode(WIFI_MODE_STA);
  AddLog(LOG_LEVEL_INFO, ">>>: WiFi.disconnect(true) WiFi.status()=%i", WiFi.status());
  WiFi.disconnect(true);
  AddLog(LOG_LEVEL_INFO, ">>>: WiFi.enableIPv6() WiFi.status()=%i", WiFi.status());
  WiFi.enableIPv6();
  AddLog(LOG_LEVEL_INFO, ">>>: WiFi.begin() WiFi.status()=%i", WiFi.status());
  WiFi.begin("xxxx","xxxx");
  AddLog(LOG_LEVEL_INFO, ">>>: delay(5000) WiFi.status()=%i", WiFi.status());
  delay(5000);
  AddLog(LOG_LEVEL_INFO, ">>>: WiFi.disconnect(true) WiFi.status()=%i", WiFi.status());
  WiFi.disconnect(true);
  AddLog(LOG_LEVEL_INFO, ">>>: END WiFi.status()=%i", WiFi.status());
```

And the logs:
```
00:00:00.038-220/52 >>>: WiFi.mode(WIFI_MODE_STA) WiFi.status()=254
[   264][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
00:00:00.130-179/40 >>>: WiFi.disconnect(true) WiFi.status()=254
[   346][W][STA.cpp:548] disconnect(): STA already disconnected.
[   347][V][STA.cpp:210] _onStaEvent(): STA Started
[   362][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[   370][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[   379][V][STA.cpp:214] _onStaEvent(): STA Stopped
[   384][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 12 - STA_STOP
[   391][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[   404][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
00:00:00.196-208/49 >>>: WiFi.enableIPv6() WiFi.status()=254
00:00:00.196-208/49 >>>: WiFi.begin() WiFi.status()=254
[   422][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[   433][E][STA.cpp:383] connect(): STA not started! You must call begin() first.
[   435][V][STA.cpp:210] _onStaEvent(): STA Started
[   446][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[   453][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
00:00:00.244-178/40 >>>: delay(5000) WiFi.status()=254
00:00:00.245-178/40 >>>: WiFi.disconnect(true) WiFi.status()=6
[  5462][W][STA.cpp:548] disconnect(): STA already disconnected.
[  5475][V][STA.cpp:214] _onStaEvent(): STA Stopped
[  5480][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 12 - STA_STOP
[  5487][D][STA.cpp:133] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[  5499][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
00:00:00.291-208/49 >>>: END WiFi.status()=254
```

As you can see, `WiFi.status()` is stuck at `254` (`WL_STOPPED`) which breaks Tasmota

me-no-dev, 2024-04-05:
@s-hadinger looks like you are sampling the status at the wrong time. What happened in the delay(5000)? I see no logs from begin? See that before disconnect, the status is `WL_DISCONNECTED`?

Please add the following line [here](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/src/STA.cpp#L269):
```cpp
log_v("wl_status: %d", (int)status);
```

You will see when the status changes.

I will confirm all that once power is restored here and my WiFi is on

s-hadinger, 2024-04-05:
@me-no-dev Is there a way to have Arduino Core3 not define `extern WiFiClass WiFi;` so that we can swap-in our own implementation of a sub-class of `WiFiClass`?

me-no-dev, 2024-04-05:
@s-hadinger we could probably equalize the interfaces now. And I tried to not change the original interface too, so do you see any difference from before the network refactoring?

> Is there a way to have Arduino Core3 not define `extern WiFiClass WiFi;`

while we can add a define switch to remove it, `WiFi` is accessed internally, so I am not sure if it will work.

s-hadinger, 2024-04-05:
Ok, I still have a problem I can't explain:

The only call made for this log is `WiFi.mode(WIFI_MODE_STA);`

Here are the logs with added verbosity
```
[   162][E][STA.cpp:260] STAClass(): _sta_network_if: 0 this:0x3ffc4090
>>>= initWiFiEvents()
[   340][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 9 - WIFI_READY
[   367][E][STA.cpp:311] onEnable(): >>> STAClass::onEnable
[   509][V][STA.cpp:214] _onStaEvent(): STA Started
[   517][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 11 - STA_START
[   524][E][STA.cpp:129] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[   531][E][STA.cpp:130] _onStaArduinoEvent(): _sta_network_if: 0x3ffc4090
[   538][E][STA.cpp:135] _onStaArduinoEvent(): Arduino STA Event: 11 - STA_START
[   545][E][STA.cpp:136] _onStaArduinoEvent(): _sta_network_if: 0x3ffc4090
[   552][D][STA.cpp:276] _setStatus(): >>> _setStatus: 6 _sta_network_if: 0x3ffc4090
[   516][D][WiFiGeneric.cpp:584] mode(): >>> WiFiGenericClass::mode 2
[   566][D][STA.cpp:271] status(): >>> this: 0x3ffc4090 status: 6
[   619][V][STA.cpp:218] _onStaEvent(): STA Stopped
[   635][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 12 - STA_STOP
[   642][E][STA.cpp:129] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[   649][E][STA.cpp:130] _onStaArduinoEvent(): _sta_network_if: 0x3ffc4090
[   656][E][STA.cpp:135] _onStaArduinoEvent(): Arduino STA Event: 12 - STA_STOP
[   663][E][STA.cpp:136] _onStaArduinoEvent(): _sta_network_if: 0x3ffc4090
[   670][D][STA.cpp:276] _setStatus(): >>> _setStatus: 254 _sta_network_if: 0x3ffc4090

[   684][V][NetworkEvents.cpp:125] checkForEvent(): Network Event: 8 - WIFI_OFF
[   691][E][STA.cpp:129] _onStaArduinoEvent(): Arduino STA Event: 8 - WIFI_OFF
[   698][E][STA.cpp:130] _onStaArduinoEvent(): _sta_network_if: 0x3ffc4090
[   705][D][WiFiGeneric.cpp:505] mode(): >>> WiFiGenericClass::mode 12
[   711][D][WiFiGeneric.cpp:508] mode(): >>> WiFiGenericClass::mode 13
[   717][E][STA.cpp:319] onDisable(): >>> STAClass::onDisable
[   723][D][WiFiGeneric.cpp:511] mode(): >>> WiFiGenericClass::mode 14
[   729][D][WiFiGeneric.cpp:517] mode(): >>> WiFiGenericClass::mode 16

----- why is ~STAClass() even called ???

[   736][E][STA.cpp:266] ~STAClass(): _sta_network_if: 0x3ffc4090 this: 0x3ffb2130
CORRUPT HEAP: Bad head at 0x3ffd1a30. Expected 0xabba1234 got 0x3ffcda7c
CORRUPT HEAP: Bad tail at 0x3ffd1a60. Expected 0xbaad5678 got 0x3ffd1a28
CORRUPT HEAP: Bad tail at 0x3ffd1ba1. Expected 0xbaad5678 got 0x6cad5678
CORRUPT HEAP: Bad head at 0x3ffb6dcc. Expected 0xabba1234 got 0x3ffb639c
CORRUPT HEAP: Bad tail at 0x3ffd1b6c. Expected 0xbaad5678 got 0x3ffd1a28
CORRUPT HEAP: Bad head at 0x3ffd1cfc. Expected 0xabba1234 got 0x3ffcda7c
CORRUPT HEAP: Bad head at 0x3ffd19c0. Expected 0xabba1234 got 0x3ffcda7c
CORRUPT HEAP: Bad head at 0x3ffd1c98. Expected 0xabba1234 got 0x3ffcda7c
[   783][D][STA.cpp:271] status(): >>> this: 0x3ffc4090 status: 254
```

After `WiFi.mode(WIFI_MODE_STA);`, `STA_STOP` event is received, but then the destructor `~STAClass()` is called which sets `_sta_network_if` to NULL, and does not record future events anymore.

I still don't understand why `~STAClass()` is called since it's a member of a sub-class of `WiFiClass` and it's supposed to be static.

me-no-dev, 2024-04-05:
it's not the destructor that is called. You must have turned STA OFF (disconnect(true)/mode(WIFI_OFF)) and since STA is the only interface on, the whole stack is freed after WIFI_OFF is emited

[128 more replies]

closed 2024-06-03 by VojtechBartoska