espressif/arduino-esp32 #8803: OTA runtime password set has no effect.

zekageri, 2023-10-25:
### Board

ESP32-Wrover

### Device Description

no

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

PlatformIO

### Operating System

Windows10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I want to be able to dinamically change OTA password every day, but it has no effect.

### Sketch

```cpp
#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "..........";
const char* password = "..........";
boolean shouldResetPassword = true;
long startMS = 0;

void setup() {
  Serial.begin(115200);

  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);

  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  ArduinoOTA.setPassword("admin");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });

  ArduinoOTA.begin();

  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void setNewOTA_Password(){
    ArduinoOTA.setPassword("admin1");
    Serial.println("OTA new password set");
    shouldResetPassword = false;
}

void loop() {
  ArduinoOTA.handle();
  if( !shouldResetPassword ){ return; }
  if( millis() - startMS >= 10000 ){
      setNewOTA_Password();
  } 
}
```

### Debug Message

```plain
No debug message, the new password is not set.
```

### Other Steps to Reproduce

If i reinitialise OTA, my esp crashes.

void setNewOTA_Password(){
    ArduinoOTA.end();
    vTaskDelay(100); // Doesn't matter.

    ArduinoOTA.setPassword("admin1");

    Serial.println("OTA new password set");
    shouldResetPassword = false;

   ArduinoOTA.begin(); // Crash
}

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-25:
It has a check if OTA is already initialized: https://github.com/espressif/arduino-esp32/blob/master/libraries/ArduinoOTA/src/ArduinoOTA.cpp#L74
You can remove the check and see if it works. I can not currently recall why it's there

zekageri, 2023-10-25:
Oh. Thank you for the fast response. Will check it!

zekageri, 2023-10-27:
It seems to work if i remove the check.

Modified method

```cpp
ArduinoOTAClass& ArduinoOTAClass::setPassword(const char * password) {
    if (!_password.length() && password) {
        MD5Builder passmd5;
        passmd5.begin();
        passmd5.add(password);
        passmd5.calculate();
        _password = passmd5.toString();
    }
    return *this;
}
```

zekageri, 2024-01-31:
After the latest upgrade I modified the method again and it didn't work this time. The method is the same as last time but can't change the password.

zekageri, 2024-01-31:
I know that because other parts of my application which uses the same dinamically changing password has successfully changed it but the OTA does not. It remained the same before the password change.

hitecSmartHome, 2024-05-31:
I have followed the steps which involved the base library tweeking.
I have removed the check `!_initialized && ` from the setPassword method but it did not work.
My application also has a dinamically changing OTA password but it won't change at runtime.
I have removed the `initialized` check from every method in `ArduinoOTA.cpp`

me-no-dev, 2024-05-31:
this is probably due to MDNS caching on your computer. MDNS records have Time To Live and the host does not need to check the record for changes before half of that time passes.

hitecSmartHome, 2024-05-31:
Since I'm using PIO and I try to upload ota with mdns name specified, will it work if I type the esp32 IP instead of it's mdns name?
Does this not bypassing the mdns cache?

me-no-dev, 2024-05-31:
> I have removed the check !_initialized &&  from the setPassword method

This should have been enough then. I see nothing else in the code preventing the password change

hitecSmartHome, 2024-06-11:
Tried with IP address instead of mdns name now and it does not work.

hitecSmartHome, 2024-06-11:
I have several ESP32 on the network running test sketches for days.
If I want to update one via OTA with password specified and it can't upload. Complaining about authentication.
I'm sure the password change method is called because I'm logging when it happens and other parts of my sketch does get updated with the new password.

```cpp
Linking .pio\build\esp-wrover-kit\firmware.elf
Retrieving maximum program size .pio\build\esp-wrover-kit\firmware.elf
Checking size .pio\build\esp-wrover-kit\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]  12.6% (used 41160 bytes from 327680 bytes)
Flash: [======    ]  64.1% (used 2015781 bytes from 3145728 bytes)
Building .pio\build\esp-wrover-kit\firmware.bin
esptool.py v4.5.1
Creating esp32 image...
Merged 27 ELF sections
Successfully created esp32 image.
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, ftdi, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = espota
Uploading .pio\build\esp-wrover-kit\firmware.bin
15:59:04 [DEBUG]: Options: {'esp_ip': 'test3.local', 'host_ip': '0.0.0.0', 'esp_port': 3232, 'host_port': 28577, 'auth': 'admin_test_pw_2', 'image': '.pio\\build\\esp-wrover-kit\\firmware.bin', 'spiffs': False, 'debug': True, 'progress': True, 'timeout': 10}
15:59:04 [INFO]: Starting on 0.0.0.0:28577
15:59:04 [INFO]: Upload size: 2022352
Sending invitation to test3.local 
Authenticating...FAIL
15:59:04 [ERROR]: Authentication Failed
*** [upload] Error 1
=============================== [FAILED] Took 101.90 seconds ===============================
```

If I restart these ESP-s the password is changing correctly once.

closed 2024-06-13 by me-no-dev