espressif/arduino-esp32 #8807: Support of Zigbee protocol

P-R-O-C-H-Y, 2023-10-25:


P-R-O-C-H-Y, 2023-11-07:
Tasks have been updated.

Hedda, 2023-11-17:
@P-R-O-C-H-Y Off-topic; just wondering if you might be interested in looking at Zigbee for Tasmota? -> https://github.com/arendst/Tasmota/discussions/20045

For reference, Tasmota's Zigbee2Tasmota (Z2T) implementation already supports Texas Instruments ZNP and Silicon Labs EZSP Zigbee stacks/interfaces with respective Zigbee Coordinator adapter running Zigbee NCP (Network Co-Processor) host firmware:

https://tasmota.github.io/docs/Zigbee/

Tasmota itself is a arduino-esp32 based open source firmware for DIY devices (especially popular for home automation solutions):

https://tasmota.github.io/docs/

PS: While Tasmota itself is mature, the ESP32-C6 support in Tasmota is still experimental, however you can follow its progress here:

https://github.com/arendst/Tasmota/discussions/19350

P-R-O-C-H-Y, 2023-12-21:
Update: Ported 2 simple Zigbee examples from ESP-Zigbee-SDK (esp-idf) #9024.
Creating a wrapper or a library is not planned for now, so after merging the examples we are closing this issue.

fillibar, 2024-01-24:
With the Arduino examples, are there any plans to make them able to connect to another hub/controller and not just another esp32c6 acting as a coordinator?

While it looks like that might be possible, at this time they just hang (monitor shows it is stuck at "Network steering was not successful"). This seems very similar to the issue here:
https://github.com/espressif/esp-idf/issues/10662

With the IDF example a heavily revised esp_zb_task seemed to help (also updating the dependencies since the ones for those examples are a bit behind).

Making a similar change to the esp_zb_task in the .ino file did not work even after copying that section from code that works when flashed with the idf.

Another thing that might help is if Serial output was provided for more of the functions. Beyond just an initial bit of information when the ESP32C6 starts up, nothing further is recorded in the Arduino IDE's Serial Monitor (vastly less than what the idf's monitor shows).

Hedda, 2024-01-24:
> With the Arduino examples, are there any plans to make them able to connect to another hub/controller and not just another esp32c6 acting as a coordinator?

FYI, it seems that Espressif devloper(s) are working on an Zigbee NCP radio library for zigpy (Python based Zigbee framework with Zigvee Stack/CLI abstraction API) so can be used as a Zigbee Coordinator NCP for projects like Home Assistant and Domoticz that depend on zigpy. Check out:

https://github.com/zigpy/zigpy/issues/1052

and

https://github.com/zigpy/zigpy-cli/pull/42

Regardless, you should really be posting any Zigbee related question and requests to Espressif’s ESP Zigbee SDK repository instead:

https://github.com/espressif/esp-zigbee-sdk

https://github.com/espressif/esp-zigbee-sdk/issues

fillibar, 2024-01-24:
I figured this was the place to post about the Arduino question because this is where I found the links to the Arduino examples.

ilker-aktuna, 2024-01-25:
@Hedda 
since you asked to write on zigbee sdk repo, I created this issue:
https://github.com/espressif/esp-zigbee-sdk/issues/228

But nobody seems to be interested there...

PR #9024 (2024-02-01): Add Zigbee examples
## Description of Change
This PR adds 2 examples to demonstrate usage of Zigbee stack:

- Zigbee_Light_Bulb -> Zigbee end device with LED.
- Zigbee_Light_Switch -> Zigbee Coordinator switching the LED on/off of the end device.

Supported SoCs are ESP32C6 and ESP32-H2.

## Tests scenarios
Tested locally by flashing Zigbee_Light_Bulb to ESP32-C6 DevKitM and Zigbee_Light_Switch to ESP32-H2 DevKitM.
CI is skipping the examples for now, as we cannot specify the FQBN.

## Related links
Closes #8807 
~~Blocked by #9025 (new GPIO Interrupt enable/disable methods)~~ -> Merged
closed 2024-02-01 by me-no-dev