espressif/arduino-esp32 #8809: 3.0.0 Alpha2 :: Serial + RMT doesn't work.

SuGlider, 2023-10-25:
### Board

ESP32

### Device Description

DevKit

### Hardware Configuration

None

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Win11

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

Starting UART0 and then using RX/TX Pins for RMT and starting again UART0 fails.
It works with 2.0.14, but it doesn't with 3.0.0-Alpha2.

The execution sequence can be seen in the example bellow.

### Sketch

```cpp
void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("TESTING ... 1");
  Serial.flush();

  const uint8_t testedPin = 1; // ESP32 UART0 RX=3 | TX = 1
#if 0
  // Arduino Core 2.0.4 -- working fine.
  rmt_obj_t* rmt_send = NULL;
  if ((rmt_send = rmtInit(testedPin, RMT_TX_MODE, RMT_MEM_64)) == NULL) {
    Serial.println("init sender failed\n"); // RX(3) / TX(1) is being used now... it may not print anything.
  }
  float realTick = rmtSetTick(rmt_send, 100);
  Serial.printf("real tick set to: %fns\n", realTick); // not to be seen... as we use TX0 in RMT.
#else
  // Arduino Core 3.0.0-Alpha2 -- working fine.
  if (!rmtInit(testedPin, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
    Serial.println("init sender failed\n");
  }
  Serial.println("real tick set to: 100ns");
#endif
  delay(500);
  Serial.end();
  Serial.begin(115200);
  Serial.setPins(3, 1);
  Serial.println("TESTING ... 2");
  Serial.flush();
}

void loop() {
  Serial.println("Looping ...");
  delay(5000);
}
```

### Debug Message

```plain
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1416
load:0x40078000,len:14804
load:0x40080400,len:4
load:0x40080404,len:3356
entry 0x4008059c
TESTING ... 1
********************===> UART0 stops working and it doesn't recover
```

### Other Steps to Reproduce

None.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-10-25:
Expected Ouput (obtained with 2.0.14):
```
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0
TESTING ... 1
TESTING ... 2
Looping ...
Looping ...
Looping ...
````

lucasssvaz, 2023-10-26:
Update:
Same thing happens with SPI in some cases.

SuGlider, 2023-10-31:
> Update: Same thing happens with SPI in some cases. Still not sure why.

This shall be fully fixed by https://github.com/espressif/arduino-esp32/pull/8820
@lucasssvaz - please let me know about your CI testing.

closed 2023-11-08 by me-no-dev