espressif/arduino-esp32 #8810: Arduino as an ESP-IDF component - esp32-hal-rmt.c:540:11: error: 'rmt_rx_channel_config_t' has no member named 'intr_priority'

dsilletti, 2023-10-25:
### Board

ESP32-S3

### Device Description

DevKitC

### Hardware Configuration

N/A

### Version

latest master (checkout manually)

### IDE Name

Visual Studio Code

### Operating System

MacOS Sonoma 14.0

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Cannot compile Arduino projects in ESP-IDF because of this error: 
```
esp32-hal-rmt.c:540:11: error: 'rmt_rx_channel_config_t' has no member named 'intr_priority'
```

I have to comment the line 540 in esp32-hal-rmt.c to successfully compile:

```
rx_cfg.intr_priority = 0;
```

### Sketch

```cpp
Any sketch
```

### Debug Message

```plain
esp32-hal-rmt.c:540:11: error: 'rmt_rx_channel_config_t' has no member named 'intr_priority'
```

### Other Steps to Reproduce

Use Arduino as component (3.0.0 github master branch) in ESP-IDF 5.1.1

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-10-25:
Hi @dsilletti, this have been added recently to the ESP-IDF and ESP32 Arduino project.
You are using version of ESP-IDF 5.1.1 and this change will be available in next release 5.1.2. To be able to use ESP32 Arduino as component without changes, you need to use branch "release/v5.1" of ESP-IDF with latest changes.

lucasssvaz, 2023-10-26:
@dsilletti Try to run get.py again if you are using the master branch to get the latest libs

nicklasb, 2023-10-26:
Is ´libraries/ESPmDNS/src/ESPmDNS.h:46:10: fatal error: mdns.h: No such file or directory´ something similar?
The managed_component espressif__mdns is there, not sure if espressif__arduino-esp32 has to explicitly require it?

me-no-dev, 2023-10-27:
@nicklasb wrong thread and this has already been fixed in master

Breefield, 2023-10-27:
I am still seeing this on `6d9ebea5008be540aaea5ad3cffc6ceb583692e3`

Steps to repro: 
```
idf.py create-project test
cd test
mkdir -p components
cd components
git clone https://github.com/espressif/arduino-esp32.git arduino
cd arduino
git submodule update --init --recursive
cd ../..
idf.py menuconfig
echo "CONFIG_FREERTOS_HZ=1000" >> sdkconfig.defaults
idf.py build
```

me-no-dev, 2023-10-27:
@Breefield being on latest IDF branch `release/v5.1` is required

Breefield, 2023-10-27:
Ah wonderful, I was on branch `v5.1` not `release/v5.1`, that'd do it.

serdartoren, 2023-11-19:
I had the same problem on the release/v5.1 branch, when I upgraded my branch to ESP-IDF v5.1.2 the problem went away.

hurricanefrog, 2024-02-16:
> Is ´libraries/ESPmDNS/src/ESPmDNS.h:46:10: fatal error: mdns.h: No such file or directory´ something similar? The managed_component espressif__mdns is there, not sure if espressif__arduino-esp32 has to explicitly require it?

In case anyone else stumbles upon this in the feature after migrating an ESP-IDF v4.x.x project to the v5.1.x version, and they encounter problems with failing compilations due to missing headers:

Check if you have `set (COMPONENTS main)` in your projects CMakeLists.txt - somehow, the dependencies of the arduino component are not loaded when you have this. Remove/comment the line, run idf.py reconfigure, and it should work,