espressif/arduino-esp32 #8815: Compiler extra_flags parameter issue

pstolarz, 2023-10-28:
### Board

any

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

arduino-cli

### Operating System

Linux

### Flash frequency

n/a

### PSRAM enabled

yes

### Upload speed

n/a

### Description

Current version of the [`platform.txt`](https://github.com/espressif/arduino-esp32/blob/6d9ebea5008be540aaea5ad3cffc6ceb583692e3/platform.txt#L70) file contains new versions of  `compiler.c.extra_flags`, `compiler.cpp.extra_flags` and `compiler.S.extra_flags` parameters which are not empty (the change provided by 6f7a1ca76ada4d705d79ce863450c91349327a79). I've been using these parameters to pass extra flags for the C/C++ compiler for the `arduino-cli` tool. Currently I'm not able to do this, since my passed flags interfere with the flags in `platform.txt` causing compilation error.

Could you consider moving these flags from `compiler.XXX.extra_flags` to `compiler.XXX.flags` and leave the extra flags empty for exclusive user usage as it is for other Arduino cores?

### Sketch

```cpp
n/a
```

### Debug Message

```plain
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/xtensa-esp32-elf-g++ -DCONFIG_PWR_CTRL_ENABLED -DCONFIG_OVERDRIVE_ENABLED -DCONFIG_CRC16_ENABLED -DCONFIG_ITERATION_RETRIES=1 @/root/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-6b1f40b9bf/esp32/flags/cpp_flags -w -Os -DF_CPU=240000000L -DARDUINO=10607 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"esp32\"" -DARDUINO_PARTITION_default "-DARDUINO_HOST_OS=\"linux\"" "-DARDUINO_FQBN=\"esp32:esp32:esp32\"" -DESP32 -DCORE_DEBUG_LEVEL=0 -DARDUINO_RUNNING_CORE=1 -DARDUINO_EVENT_RUNNING_CORE=1 -DARDUINO_USB_CDC_ON_BOOT=0 @/root/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-6b1f40b9bf/esp32/flags/defines -I/home/ps/projects/c/arduino/OneWireNg/examples/arduino/DallasTemperature -iprefix /root/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-6b1f40b9bf/esp32/include/ @/root/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-6b1f40b9bf/esp32/flags/includes -I/root/.arduino15/packages/esp32/tools/esp32-arduino-libs/idf-release_v5.1-6b1f40b9bf/esp32/qio_qspi/include -I/root/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha2/cores/esp32 -I/root/.arduino15/packages/esp32/hardware/esp32/3.0.0-alpha2/variants/esp32 -I/root/Arduino/libraries/OneWireNg/src @/tmp/arduino/sketches/2EC09C26A960CEA4C81193530AEEE765/build_opt.h /tmp/arduino/sketches/2EC09C26A960CEA4C81193530AEEE765/sketch/DallasTemperature.ino.cpp -o /tmp/arduino/sketches/2EC09C26A960CEA4C81193530AEEE765/sketch/DallasTemperature.ino.cpp.o
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libc.a(lib_a-raise.o):(.literal+0x0): warning: _getpid_r is not implemented and will always fail
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libc.a(lib_a-raise.o):(.literal+0x4): warning: _kill_r is not implemented and will always fail
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libc.a(lib_a-syswrite.o):(.literal+0x4): warning: _write_r is not implemented and will always fail
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/libc.a(lib_a-syswrite.o):(.literal+0x0): warning: __getreent is not implemented and will always fail
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/lib/no-rtti/crt0.o:(.literal+0x0): undefined reference to `main'
| /root/.arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-12.2.0_20230208/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld: /tmp/cc5Yr0r9.o:(.literal._ZN9OneWireNg8iterator10searchStepEv[_ZN9OneWireNg8iterator10searchStepEv]+0x0): undefined reference to `_ZN9OneWireNg6searchERA8_hb'
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-10-28:
Create a file named `build_opt.h` in your sketch folder and put the flags in there. That way we do not need to change anything and you do not need to depend on platform.txt

pstolarz, 2023-10-28:
@me-no-dev `compiler.XXX.extra_flags` is a common and [documented](https://arduino.github.io/arduino-cli/dev/commands/arduino-cli_compile/) approach to provide extra flags to the compiler, whereas `build_opt.h` is not (it's only esp32 and stm32 cores specific). On my CI/CD env I'm testing ~20 cores so `build_opt.h` doesn't help me. For all cores known to me the `compiler.XXX.extra_flags` are empty and left for exclusive user usage.

me-no-dev, 2023-10-30:
Could you give me some context on what you are testing and how you are overwriting those flags? We could move them and uglify the platform more, but I would like to understand well what is going on and what are the possible use cases.

pstolarz, 2023-10-30:
Please look at [this workflow lines 65-67](https://github.com/pstolarz/OneWireNg/blob/0f02169d000b0f3d4b91faed477bb2f98f13102f/.github/workflows/arduino-builds.yml#L65). I'm passing extra flags via `--build-property "compiler.cpp.extra_flags=$BUILD_FLAGS"`, where `BUILD_FLAGS` is set above.

BTW I've overcome this issue by adding `-c` do `BUILD_FLAGS` but it's not a solution I like.

ArminJo, 2024-10-10:
@me-no-dev 
Hi me-no-dev,
it seems, that ALL library developers using CI/CD rely on passing extra flags by using `compiler.cpp.extra_flags`.

See: https://github.com/arduino/arduino-cli/issues/2683#issuecomment-2263603730

> Even though the idea behind the compiler.cpp.extra_flags property was to provided a dedicated property for sole use by the user to inject arbitrary flags into the C++ compilation command, the powerful and flexible Arduino boards platform framework doesn't impose any restrictions on platform developers. Unfortunately some platform developers have pointlessly (because they could instead have simply defined an arbitrary property for the purpose) made use of this property internally.

So ESP is the only core which "breaks" this implicit rule with its 3.x release, effectively disabling CI/CD for library developer.
I can see the reason for using the `compiler.cpp.extra_flags` in platform.txt, but I am sure you weren't aware of the drawbacks (poor library support in the future) of using this.
Can you make all voluntary developes a favour and move the three `-MMD -c` from *.extra_flags to *. flags?

me-no-dev, 2024-10-10:
will look into it

me-no-dev, 2024-10-21:
Will be fixed in the upcoming v3.1.0

closed 2024-10-23 by pstolarz