espressif/arduino-esp32 #8816: NimBLE + SPIFFS - SPIFFS will stop working over longer periods with fewer writes

BrugerX, 2023-10-30:
### Board

ESP32-S3

### Device Description

NodeMCU v1.1

### Hardware Configuration

I have three triggers that determine how my experiment works:

1) Printing_on GPIO: 19
2) Xperiment_on GPIO: 18
3) Restart GPIO: 17

Other than that, I have two LEDS connected:

1) XPERIMENT_ON_LED: 25
2) BLE_FLASH: 32

### Version

v2.0.14

### IDE Name

PlatformIO with CLion

### Operating System

Windows 10

### Flash frequency

Unknown, but flashing is nto the issue

### PSRAM enabled

yes

### Upload speed

115200

### Description

I am trying to create a BLE based soil water content datalogger.

Filesystem: I am using a custom-made wrapper class based around the SPIFFS filesystem. I have used it for a previous project on an esp32 with LoRa where it went through exhaustive unit testing.
BLE: I am using NimBLE-Arduino for my BLE stack module.

Expected behaviour:

When the experiment on:

The server is expected to go to sleep for 30 minutes at a time, wake up for a limited amount of time (5 seconds.) The server has a SWC-sensor attached, which it sends to the client.

The client is constantly on, because it is connected to a power outlet. I receives the SWC reading and stores it plus a timestamp, which it generates locally, to a SWC_Value_Array and a Timestamp_Value_Array.

Saving:

The way the saving works currently is for the client to load the whole file into an array in the heap. It then finds the index where it wishes to insert the value. It writes the value and increments a counter, that keeps track of where the next write-index is.

Each value array has a specific index counter.

The timestamps are saved as their bits in 4 unsigned chars. The SWC is saved in the format "XX.XX" as unsigned chars as well. All SWC values are formatted to be exactly 5 chars long. E.g "1.24" -> "01.24", "4.54" -> "04.54".

All arrays are initialized to be filled with "l"'s.

Interrupts:

Xperiment_on: If this is triggered to be true, we start the experiment. If it is true and triggered to be false, we stop scanning and we stop the experiment.

Restart: This will overwrite the value arrays with l's and trigger a system abort.

Print_on: This will start a while loop, which continues to print the value arrays as unsigned chars or longs. There are vTask delays inbetween to ensure no WDT is triggered.

Error:

I can write a lot of values in a row into the SPIFFS. That is, I can write 20 values in a matter of 2 minutes, and nothing bad happens.

However, if I write 7 values over a span of half an hour I get an erroneous output. It will succesfully read the SWC from the client, however the SWC_value_array turns into unintelligble characters. In the specific screenshot the value is written in the middle of the array. But all previous values have been replaced by unintelligble characters. (The first screenshot is the 6th write, the second is the 7th write. I have highlighted the part where the client receives the characteristic value of the SWC characteristic, and where it can be found in the value_array)

[image]

[image]

### Sketch

```cpp
1) Start NimBLE as a client
2) Read server response and store it in an unsigned char array in SPIFFS in the manner described above
3) Try to save this unsigned char with
4) Put the server to sleep after 5 seconds from 2) and have it wake up long after
5) Repeat from 2-5

Otherwise, I will share my main.cpp here. I am hoping someone can give me an "analytical" explanation, and I will try to test it out on my own codebase :)

bool SPIFFSFileManager::save_file_with_retries(const char *filePath, const unsigned char *dataToWrite,int size_of_file, int attempts, int mS_delay_between) {
    int success = 0;
    int retries = 0;

    SPIFFSFileManager& fileMan = SPIFFSFileManager::get_instance();

    while (retries < attempts) {
        // Try to save the file
        success = fileMan.save_file(filePath, dataToWrite,size_of_file);

        // Check if save was successful
        if (success) {
            log_i("Succesfully inserted value into value array.\n");
            return true;
        } else {
            log_i("Failed to insert value array into path %s for the %i 'th time. Retrying again in %i milliseconds",filePath,retries,mS_delay_between);
            vTaskDelay(mS_delay_between / portTICK_PERIOD_MS);
            retries++;
        }
    }

    return false;
}
```

### Debug Message

```plain
There was no debug message. See the screenshots above for the semantical error.

I even include multiple exceptions in my saving method:

bool SPIFFSFileManager::save_file_with_retries(const char *filePath, const unsigned char *dataToWrite,int size_of_file, int attempts, int mS_delay_between) {
    int success = 0;
    int retries = 0;

    SPIFFSFileManager& fileMan = SPIFFSFileManager::get_instance();

    while (retries < attempts) {
        // Try to save the file
        success = fileMan.save_file(filePath, dataToWrite,size_of_file);

        // Check if save was successful
        if (success) {
            log_i("Succesfully inserted value into value array.\n");
            return true;
        } else {
            log_i("Failed to insert value array into path %s for the %i 'th time. Retrying again in %i milliseconds",filePath,retries,mS_delay_between);
            vTaskDelay(mS_delay_between / portTICK_PERIOD_MS);
            retries++;
        }
    }

    return false;
}

And loading:

bool SPIFFSFileManager::load_file(const char * filePath, unsigned char * resultArray, size_t endIdx){
    File f1 = fileSystem.open(filePath,FILE_READ);
    if(!f1 || f1.isDirectory()){
        throw std::logic_error("− failed to open file for reading. You possibly tried to load a directory");
    }

    unsigned char res;

    for (int i = 0; i<endIdx; i++){
        if(!f1.available()){
            break;
        }
        res = f1.read();
        resultArray[i] = res;
    }

    resultArray[endIdx] = '\0';

    return true;

}

I have run unit tests, that confirm that these methods work without NimBLE activated.
```

### Other Steps to Reproduce

My Pio.ini file, if it helps:

; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32 @ 6.3.0
board = NodeMCU-32S
framework = arduino
monitor_speed = 115200
monitor_filters = esp32_exception_decoder
upload_port = COM3
monitor_port = COM3
test_port = COM3

debug_tool = esp-prog
debug_init_break = tbreak setup

lib_deps =
	h2zero/NimBLE-Arduino@^1.4.1
	extentsoftware/TBeamPower@^2.0.4
	milesburton/DallasTemperature@^3.11.0
	4-20ma/ModbusMaster@^2.0.1
	thebigpotatoe/Effortless-SPIFFS@^2.3.0
lib_ldf_mode = deep+
board_build.partitions = SPIFFSoneMPartition.csv
board_build.flash_mode = dio

build_type = debug

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-10-30:
This forum is for issues with the code in this repository.  It is not for issues with your code, or 3rd party libraries.  You can ask your question at https://esp32.com , but expecting someone to guess at your problem without providing something that can be copied and reproduced is not going to get you many answers.

SuGlider, 2023-11-09:
@BrugerX - The project seems very nice. If the issue is related to SPIFFS, maybe you can isolate the code, create some timed writing method and check if it fails. This could result in a smaller sketch that can be used to reproduce the issue.

As an alternative, I would suggest using LitleFS instead of SPIFFS. It may solve the issue with no collateral effect for the project.

LittleFS is crash proof so every append first makes a copy, appends to the copy, then replaces the original. In this way a crash during append can not corrupt the original file. It is a bit slower because of it, but it may be safer than SPIFFS.

VojtechBartoska, 2023-12-11:
Closing this as expired due to no answer from the author of the issue.

closed 2023-12-11 by VojtechBartoska