espressif/arduino-esp32 #8823: Increase the number of BLE Bluetooth connected devices for ESP32

Crazy-Feng, 2023-11-01:
### Related area

BT,ESP32,Arduino

### Hardware specification

ESP32 DevKitC

### Is your feature request related to a problem?

When I used arduino to increase the number of Bluetooth connections of ESP32, I found that the maximum number of connections can only reach 4, which is not consistent with the official document that the maximum number of connections is 9.

### Describe the solution you'd like

I think 9 BLE connections for ESP32, more would be nice.

### Describe alternatives you've considered

1.I delayed sending data from the four devices that could connect so that they could stagdle their data and not block the channel, but unfortunately the fifth Bluetooth device still couldn't connect.
2.I replaced the antenna, added more memory, and expanded the Bluetooth space, but unfortunately the fifth BLE device still couldn't connect.

### Additional context

The framework I am using is arduino.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-11-01:
This is compiled into the included idf libraries.  Each connection requires some memory to be allocated.  If you want more connections, you can [compile your own libraries](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#esp32-arduino-lib-builder), but there may not be enough memory to run BLE + WiFi with that many connections.

mrengineer7777, 2023-11-01:
@Crazy-Feng Please provide a link to the official documentation so we can correct it.

Crazy-Feng, 2023-11-02:
> This is compiled into the included idf libraries. Each connection requires some memory to be allocated. If you want more connections, you can [compile your own [...]

Thank you for your advice. I will look at the relevant materials now.

Crazy-Feng, 2023-11-02:
> @Crazy-Feng Please provide a link to the official documentation so we can correct it.

Hello, link here:https://docs.espressif.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/ble-bt.html#id14

Crazy-Feng, 2023-11-02:
@mrengineer7777 Hello, this is the reply from your engineer when I sent an email to your company asking for the maximum number of Bluetooth connections
[image]

wodeminzi, 2023-11-02:
Yes, yes, I also ran into the esp32 Bluetooth connection count of only 4

wodeminzi, 2023-11-02:
In platformio, arduino is used to develop the ESP32-WROOM-32UE chip, which is used as a gateway to issue broadcast packets for the ESP32C3 device to connect. However, when the number of Bluetooth device connections is not set, the ESP32-WROOM-32UE can only connect 3 ESP32C3 devices, and after code Settings (as shown in Figure 1 below), it can connect 4 ESP32C3 devices. But when I set the maximum number of connections to bt_cfg.ble_max_conn = 5; Even when set to 9, there were only four ESP32C3 devices that could be connected. After watching the link below https://docs.espressif.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/ble-bt.html#id14 (figure 2 as the central link It also changed the ACL value, still only four ESP32C3 devices can be connected. I would like to ask how the maximum connection number of 9 you said was realized.
[image]
[image]

wodeminzi, 2023-11-02:
[image]

Crazy-Feng, 2023-11-02:
@lbernstone Hello,sorry to bother you, I see that [the Arduino library generator](https://github.com/espressif/esp32-arduino-lib-builder) provided in [the file](https://docs.espressif.com/projects/arduino-esp32/en/latest/esp-idf_component.html#esp32-arduino-lib-builder) can only run on Ubuntu (32 and 64 bit), Raspberry Pi and MacOS. Unfortunately, my computer system is Windows. Is there an Arduino library generator for Windows?
[image]

lbernstone, 2023-11-02:
> @lbernstone Hello,sorry to bother you, I see that [the Arduino library generator](https://github.com/espressif/esp32-arduino-lib-builder) provided in [the [...]

You can either run a virtual machine or WSL to compile them.

Crazy-Feng, 2023-11-02:
> You can either run a virtual machine or WSL to compile them.

Thanks for your reply, I will try it.

Crazy-Feng, 2023-11-06:
@lbernstone Hello,I tried a few things， you mentioned in your previous suggestion that too many connections may lead to insufficient memory. Is there any solution for this? Is it possible to control the increase of memory in software? Or can you only increase the memory size in hardware?

SuGlider, 2023-11-09:
> @lbernstone Hello,sorry to bother you, I see that [the Arduino library generator](https://github.com/espressif/esp32-arduino-lib-builder) provided in [the [...]

No, Lib Builder can only be used in Unix-Like environments. 
With Windows, it may be possible under WSL Ubuntu Linux, but I'm not sure that it will work fine.

SuGlider, 2023-11-09:
> @lbernstone Hello,I tried a few things， you mentioned in your previous suggestion that too many connections may lead to insufficient memory. Is there any solution for this? Is it possible to control the increase of memory in software? Or can you only increase the memory size in hardware?

No possible to change the ESP32 SRAM space. The best case would to use a ESP32 with PSRAM. 
Some allocations may use PSRAM and let more HEAP space for the BLE driver.

lbernstone, 2023-11-09:
The [NimBLE library](https://github.com/h2zero/NimBLE-Arduino) uses significantly less memory than the arduino library.

Crazy-Feng, 2023-11-10:
> No possible to change the ESP32 SRAM space. The best case would to use a ESP32 with PSRAM. Some allocations may use PSRAM and let more HEAP space for the BLE driver.

I get it, then I will go to learn about ESP32 with PSRAM,Many thanks for your reply.

Crazy-Feng, 2023-11-10:
> The [NimBLE library](https://github.com/h2zero/NimBLE-Arduino) uses significantly less memory than the arduino library.

I'm going to learn the NimBLE library and see if it works for my current project,Many thanks for your reply

closed 2023-11-23 by Crazy-Feng