espressif/arduino-esp32 #8824: ESP32C3 sd card

plutphil, 2023-11-01:
### Board

esp32-c3-devkitc-02

### Device Description

I have this exact board connected with an sd card with spi.

### Hardware Configuration

#define PIN_SPI_CLK 6
#define PIN_SPI_MOSI 4
#define PIN_SPI_MISO 5
#define PIN_SPI_CS 4

### Version

v2.0.14

### IDE Name

arduino / platforio

### Operating System

win10

### Flash frequency

?

### PSRAM enabled

no

### Upload speed

921600

### Description

With IDF the sd card and this exact pin configuration it works. Arduino Framework doesn't work. I tried setting the frequency down to 2.000.000 Hz and 200.000 Hz but this did not help. I'm trying to get this project working.

### Sketch

```cpp
#include <Arduino.h>
//https://arduinogetstarted.com/tutorials/arduino-micro-sd-card

#include <SD.h>

#define PIN_SPI_CLK 6
#define PIN_SPI_MOSI 4
#define PIN_SPI_MISO 5
#define PIN_SPI_CS 4

File myFile;

void setup() {
  Serial.begin(115200);
  printf("mosi%d miso%d cs%d clk%d\n",MOSI,MISO,SPI_CS0_GPIO_NUM,SPI_CLK_GPIO_NUM);
  SPI.begin(PIN_SPI_CLK,PIN_SPI_MISO,PIN_SPI_MOSI,PIN_SPI_CS);
  while (!SD.begin(PIN_SPI_CS,SPI,20*100*1000)) {
    Serial.println(F("SD CARD FAILED, OR NOT PRESENT!"));
    delay(1000);
  }

  Serial.println(F("SD CARD INITIALIZED."));

  if (!SD.exists("arduino.txt")) {
    Serial.println(F("arduino.txt doesn't exist. Creating arduino.txt file..."));
    // create a new file by opening a new file and immediately close it
    myFile = SD.open("arduino.txt", FILE_WRITE);
    myFile.close();
  }

  // recheck if file is created or not
  if (SD.exists("arduino.txt"))
    Serial.println(F("arduino.txt exists on SD Card."));
  else
    Serial.println(F("arduino.txt doesn't exist on SD Card."));
}

void loop() {
}
```

### Debug Message

```plain
[    99][W][sd_diskio.cpp:174] sdCommand(): no token received
[   200][W][sd_diskio.cpp:174] sdCommand(): no token received
[   300][W][sd_diskio.cpp:174] sdCommand(): no token received
[   399][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[   399][W][sd_diskio.cpp:516] ff_sd_initialize(): GO_IDLE_STATE failed
[   400][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[   409][W][sd_diskio.cpp:174] sdCommand(): no token received
[   515][W][sd_diskio.cpp:174] sdCommand(): no token received
[   615][W][sd_diskio.cpp:174] sdCommand(): no token received
[   714][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
SD CARD FAILED, OR NOT PRESENT!
```

### Other Steps to Reproduce

https://github.com/espressif/arduino-esp32/issues/6237

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-01:
It works in IDF with the MOSI and CS on the same pin?

P-R-O-C-H-Y, 2023-11-02:
Hi @plutphil, the SD card normally works on C3. Are you sure about the pins as @lbernstone mentioned?

plutphil, 2023-11-02:
> It works in IDF with the MOSI and CS on the same pin?

no it doesn't, my mistake. I used these pins:
#define PIN_SPI_CLK 6
#define PIN_SPI_MOSI 7
#define PIN_SPI_MISO 5
#define PIN_SPI_CS 4
how could i not see this. it know works also with arduino.

closed 2023-11-02 by plutphil