espressif/arduino-esp32 #8825: Can't upload sketch using terminal

ahmettarhan, 2023-11-01:
### Board

ESP32-S3

### Device Description

Just esp32-s3 module

### Hardware Configuration

Module is connected to computer through UART 

### Version

v2.0.9

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

Not relevant

### PSRAM enabled

yes

### Upload speed

Not relevant

### Description

I am trying to upload the sketch to the eps32-s3 module using terminal commands. I have to define the architecture of board according to https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc. I follow the instructions given in above link but for me, the folder after hardware folder is named 2.0.9 (it is avr for arduino board). I don't know what to do. 

### Sketch

```cpp
arduino_debug --upload --board esp32:esp32s3 --port COM12 core.ino
```

### Debug Message

```plain
Error: esp32:esp32s3: Invalid board name, it should be of the form "package:arch:board" or "package:arch:board:options"
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-01:
--fqbn esp32:esp32:esp32s3

ahmettarhan, 2023-11-02:
That worked @lbernstone thanks!!

closed 2023-11-02 by ahmettarhan