espressif/arduino-esp32 #8826: Error that AP password is incorrect in 2.0.14

qlalfdu, 2023-11-02:
### Board

esap32 wroom32

### Device Description

esp32 development board

### Hardware Configuration

Use AP mode only

### Version

v2.0.14

### IDE Name

1.8.19

### Operating System

windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

115200

### Description

There was no problem in 2.0.9, but after updating to 2.0.14, connection is not possible because the AP password is incorrect.

If I revert to 2.0.9, it works fine.

### Sketch

```cpp
Not relevant to code
```

### Debug Message

```plain
Not a compilation problem
```

### Other Steps to Reproduce

If I revert to 2.0.9, it works fine.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-11-02:
Hi @qlalfdu, can you please better explain what is the issue? And if you can send sketch to reproduce it would be great.

mrengineer7777, 2023-11-02:
Is your WiFi security WEP, WPA, WPA2, WPA3?

qlalfdu, 2023-11-02:
> 안녕@qlalfdu, 문제가 무엇인지 더 자세히 설명해 주시겠어요? 그리고 재현을 위해 스케치를 보낼 수 있다면 좋을 것입니다.

After updating the sketch I worked on in version 2.0.9 to 2.0.14, I just recompiled it without modifying the code, uploaded it, and tried to connect to the AP, but the password was wrong and I couldn't connect.

If I download the version again to 2.0.9, it works fine.

They change the version to 2.0.14 again and upload it. At this time, I can access it by setting erase all flash before sketch upload: enable and uploading.

``

qlalfdu, 2023-11-03:
> Is your WiFi security WEP, WPA, WPA2, WPA3?

I don't know how to set it up in Arduino.

qlalfdu, 2023-11-03:
The big problem is that it is very difficult to access modules attached to other devices to erase the flash with a download tool.

closed 2023-11-03 by qlalfdu