espressif/arduino-esp32 #8828: Unable to connect to AP after updating to 2.0.14

qlalfdu, 2023-11-03:
### Board

esap32 wroom32

### Device Description

esp32 development board

### Hardware Configuration

Use AP mode only

### Version

v2.0.14

### IDE Name

arduino ide 1.8.19

### Operating System

windows 10

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm sorry.
I'm not familiar with it and I'm not good at English, so I closed the previously posted issue by mistake, so please understand that I'm uploading it again.

1, After updating the sketch I worked on in version 2.0.9 to 2.0.14, I just recompiled and uploaded it without modifying the code and tried to connect to the AP, but I could not connect because the password was wrong.

2. If re-download version 2.0.9, it will work fine.

3. Change the version to 2.0.14 again and upload. At this point, can access it by selecting erase all flash before sketch upload: Enabled and Uploading.

The big problem is that it is not easy for people to access what is installed on other devices.

### Sketch

```cpp
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h>

#include <DNSServer.h>
#include "mainpage.h"
const byte DNS_PORT = 53;
DNSServer dnsServer;
#include <WebServer.h>
WebServer server(80);
IPAddress apIP(8, 8, 8, 8);
IPAddress apsubnet(255, 255, 255, 0);

String apid = "ABCD";
String appw = "12341234";

void setup() {
Serial.begin(115200);
WiFi.disconnect(true);
WiFi.mode(WIFI_AP);
WiFi.softAP(apid.c_str(), appw.c_str());
WiFi.softAPConfig(apIP, apIP, apsubnet);
WiFi.softAPConfig(apIP, apIP, apsubnet);
dnsServer.start(DNS_PORT, "*", apIP);
server.on("/", handleRoot);
server.on("/generate_204", handleRoot); //Android captive portal. Maybe not needed. Might be handled by notFound handler.
server.on("/fwlink", handleRoot); //Microsoft captive portal. Maybe not needed. Might be handled by notFound handler.
server.onNotFound(handleNotFound);
server.begin();
}

void loop() {
dnsServer.processNextRequest();
server.handleClient();
}

void handleRoot() {
server.send_P(200, "text/html", mainpage);
}

void handleNotFound() {
String message = F("File Not Found\n\n");
for (uint8_t i = 0; i < server.args(); i++) {
message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
}
server.send(200, "text/plain", message);
}

///
mainpage.h

const char mainpage[] PROGMEM = R"=====(
<!DOCTYPE html>
<html>
<head>
<html>
<link rel="shortcut icon" href="#">
<meta charset="UTF-8">
</head>
<title>FanSetup</title>
<body>
<div>
hellow word!
</div>
</body>
</html>
)=====";
```

### Debug Message

```plain
Not a compilation problem
```

### Other Steps to Reproduce

If I revert to 2.0.9, it works fine.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-03:
I am unable to reproduce.  I get the captive portal as expected.  Please enable verbose debug logging and post the log.

JosephHewitt, 2023-11-04:
I also had the same problem. I was running code compiled on `arduino-esp32 2.0.12` which worked fine. I compiled the same project on `2.0.14` and was still able to connect to other APs, but when reconfiguring (not re-flashing) the device to create an AP, I had the same problem as reported above. I then recompiled and re-flashed the same project with 2.0.14 again with verbose logging to attempt to find the cause. The logging didn't show anything interesting, just the normal events as shown below:
```
[ 15812][V][WiFiGeneric.cpp:392] _arduino_event_cb(): AP Started
[ 15812][D][WiFiGeneric.cpp:1039] _eventCallback(): Arduino Event: 10 - AP_START
[ 15828][V][WiFiServer.h:42] WiFiServer(): WiFiServer::WiFiServer(port=80, ...)
```

When attempting to connect (using an iPhone on iOS 17.1) to the AP, I received a message stating the password was incorrect. No logs appeared over serial during this. The password was definitely correct since my phone had saved the credentials from previous times when it worked fine.

I used some basic code to set up an AP to isolate the issue, but then things worked as expected. I have not been able to reproduce the problem since, even when recompiling the same project with an old version and repeating the same steps which led to the initial issue.

My project also uses OTA updating and I performed some OTA updates when trying to reproduce and still could not reproduce.

I am using Arduino IDE 1.8.19 (esptool v4.5.1), flashing an ESP32D0WDQ5 (revision 1). I am not using PSRAM.

qlalfdu, 2023-11-06:
The important thing to note is that if you erase flash even once in 2.0.14, it is impossible to reproduce.
If I clear flash in 2.0.9 and don't erase it in 2.0.14, it plays.
In other words, the problem is that a device made with a version prior to 2.0.9 cannot be connected to the AP after an OTA update to version 2.0.14

SuGlider, 2023-11-07:
> The important thing to note is that if you erase flash even once in 2.0.14, it is impossible to reproduce. If I clear flash in 2.0.9 and don't erase it in 2.0.14, it plays. In other words, the [...]

This is interesting... 2.0.9 uses IDF 4.4.4 and 2.0.14 uses IDF 4.4.5.
But erasing the flash should work always.

Are you using the Arduino IDE Menu `Erase All Flash Before Sketch Upload: "Enabled"` to erase the flash?

lbernstone, 2023-11-07:
> This is interesting... 2.0.9 uses IDF 4.4.4 and 2.0.14 uses IDF 4.4.5. But erasing the flash should work always.
>
> Are you using the Arduino IDE Menu `Erase All Flash Before Sketch Upload: "Enabled"` [...]

My understanding is this is the error he is reporting.  If you update a device from 2.0.9 using OTA to a 2.0.14 firmware, something is invalid in the nvs such that AP mode does not work properly (bad lease info?).  It takes some time to build the needed resources, but I'll see if I can reproduce today.

SuGlider, 2023-11-07:
There is a way to erase WiFi information using IDF call `esp_wifi_restore()`... 
This call erases WiFi NVS related information (everything).

Maybe calling it before running AP mode again may solve the issue.

SuGlider, 2023-11-07:
In the very worst case, there is a way to erase all NVS, including data from other domains...
``` cpp
#include <nvs_flash.h>

void setup() {
  nvs_flash_erase();
  nvs_flash_init();
}
```

lbernstone, 2023-11-07:
I am still unable to reproduce.  @qlalfdu can you try with a different client and see if this may be specific to your pc/phone?

SuGlider, 2023-11-07:
@qlalfdu @lbernstone - I was able to reproduce it. As @qlalfdu said I can't connect to the AP because after using 2.0.14, my Android phone says that the WiFi password is wrong ( "12341234").

I could make it work by using `nvs_flash_erase() and nvs_flash_init()`, but not with `esp_wifi_restore()`.
This is code that ha worked for me:

``` cpp
#include "Arduino.h"
#include <WiFi.h>
#include <WiFiClient.h>

#include <nvs_flash.h>  // <<<<<<

#include <DNSServer.h>

void setup() {
  Serial.begin(115200);
  Serial.println("Trying to erse all NVS flash...");
  if (nvs_flash_erase() != ESP_OK) Serial.println("===> Error with Flash Erase.");
  if (nvs_flash_init() != ESP_OK) Serial.println("===> Error with Flash INIT.");
  Serial.println("Done with NVS flash...");

  WiFi.disconnect(true);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(apid.c_str(), appw.c_str());
  WiFi.softAPConfig(apIP, apIP, apsubnet);
  WiFi.softAPConfig(apIP, apIP, apsubnet);
  dnsServer.start(DNS_PORT, "*", apIP);

  // ..... and the rest of the code...
}

```

lbernstone, 2023-11-08:
That's a pretty big hammer you got there, @SuGlider 
The namespace for all this is nvs.net80211.  Here's what mine has after running OTAWebUpdater.ino.
```
Namespace: nvs.net80211
Entries: 27
key 'ap.sndchan', type '1' 
key 'ap.authmode', type '1' 
key 'ap.sae_h2e', type '1' 
key 'ap.chan', type '1' 
key 'ap.ssid', type '66' 
key 'ap.passwd', type '66' 
key 'ap.hidden', type '1' 
key 'ap.max.conn', type '1' 
key 'bcn.interval', type '2' 
key 'ap.p_cipher', type '1' 
key 'ap.ftm_r', type '1' 
key 'ap.pmf_e', type '1' 
key 'ap.pmf_r', type '1' 
key 'ap.pmk_info', type '66' 
```
To get the ap.passwd:
```
  prefs.begin("nvs.net80211");
  char* pw[128];
  prefs.getBytes("ap.passwd", pw, 128);
  Serial.printf("pw: %s\n", pw);
```
See if that has been corrupted, or if deleting that key or the namespace, lets you start over.

SuGlider, 2023-11-08:
@lbernstone @qlalfdu  @JosephHewitt 
Well.... The 2.0.14 reports that the password is still the same (from Preferences): "12341234", **BUT** it still fails saying that the authentication failed because of bad password.

Arduino layer (`WiFi.softAP(SSID, PWD)`) just sets the necessary fields and then calls IDF Function `esp_wifi_set_config()`.
Something may have changed when moving from IDF 4.4.4/4.4.5 (Core 2.0.0 to 2.0.13) to 4.4.6 (Core 2.0.14) related to it.

One way to avoid this problem would be to use Arduino Core up to 2.0.13 (IDF 4.4.5 is fine)

Another way would be using this "big hammer" with 2.0.14.
https://github.com/espressif/arduino-esp32/issues/8828#issuecomment-1799380630

SuGlider, 2023-11-08:
In summary: Use Arduino Core 2.0.13 that all will be fine.

20162026, 2023-11-08:
judging from idf changelog something in NVS was "fixed", unfortunately changes are inside of wifi binary blobs
https://github.com/espressif/esp-idf/commit/ad08d94ed3d709de95b60a8608272a676c3b02f7

Jason2866, 2023-11-28:
Needing to erase the whole flash to get reliable wifi or wifi function is nothing new.
This happened with many cores before already. I do have this often when building the Arduino framework and using different commits where the closed source wifi libs have changed in IDF.
After a full erase and disconnecting from power, it connects to wifi without any issue when power is connected.

VojtechBartoska, 2023-12-11:
Hello,

closing this as answered, if you need, please reopen the issue.

closed 2023-12-11 by VojtechBartoska

coratron, 2023-12-18:
Hi @VojtechBartoska , what's the solution to this problem? A full flash erase?

 Rolling back to 2.0.13 as proposed by @SuGlider  does not look like a long term solution, more like a workaround.

> judging from idf changelog something in NVS was "fixed", unfortunately changes are inside of wifi binary blobs [espressif/esp-idf@ad08d94](https://github.com/espressif/esp-idf/commit/ad08d94ed3d709de95b60a8608272a676c3b02f7)

I am not sure that solved it either as I still see the issue. I am using arduino as a component and esp-idf v4.4.6. I have compiled the project with the latest v4.4 release: Commit hash b1500bf9cb3424f2b0d94caad2480138e781c518

What happens is if I compile the project with the old password, while trying to connect to the esp32 as softAP the password is seen as incorrect. If I modify the password and recompile everything is ok. The password is defined as follows:
`
const char* password = "password1234";`

It definitely looks like some kind of nvs management issue, where the password gets written somewhere with `WiFi.softAP(ssid, password)` and the issues flow from there.

Jason2866, 2023-12-18:
@coratron At least a full erase of the NVS section. There is no other fix. It is not an Arduino issue. The problem is the changed code in IDF / closed source binary blobs.

Jason2866, 2023-12-18:
Still Arduino code can't change this behaviour. It is caused from IDF. The issue needs to be opened / transfered to IDF repo.

coratron, 2023-12-18:
@Jason2866 I am opening a ticket there and will reference this thread

Gixy31, 2024-03-22:
I just upgraded my configuration to 6.5.0 / 2.0.14 and after compilation and uploading of a previouly working application, felt into a huge problem of WiFi connection.
After two days of investigation, I finally found that during WiFi connection, the event handler received an SYSTEM_EVENT_STA_DISCONNECTED event before the SYSTEM_EVENT_STA_GOT_IP event, causing an infinite loop of connection/reconnection in my software. I put a simple workaround, trying to reconnect only after checking that WiFi was really disconnected, which should be the case when this event is triggered, but obviously is not with this new release.