espressif/arduino-esp32 #8830: About ESP32 Arduino Lib Builder

Mushan-bin, 2023-11-03:
### Related area

ESP32,Arduino

### Hardware specification

ESP32 DevKitC

### Is your feature request related to a problem?

About ESP32 Arduino Lib Builder Build on Ubuntu, execute sudo pip install --upgrade pip, the following problem appears.

### Describe the solution you'd like

I would like to know where to compile the Arduino Lib Builder for creating ESP32 on umbuntu to create a custom Bluetooth connection library and use it on arduino.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

Mushan-bin, 2023-11-03:
[image]
[image]

me-no-dev, 2023-11-03:
so you could not upgrade PIP for the lib-builder. Please search google on how to overcome the PIP issue and then if you fix it and could not continue for another reason, post an issue in the lib-builder repo and not here.

closed 2023-11-03 by me-no-dev