espressif/arduino-esp32 #8832: About ESP32 Arduino Lib Builder

Mushan-bin, 2023-11-03:
### Related area

arduino，esp32

### Hardware specification

ESP32 DevKitC

### Is your feature request related to a problem?

For the ESP32 Arduino Lib Builder Tested on Ubuntu, I used the following command sudo pip install --upgrade pip got the following error. In addition, I also want to ask how to use this library file in umbuntu, which one to compile to complete this library, and how to generate the library file that can be used by arduino at last.

### Describe the solution you'd like

Is there any tutorial about esp32-arduino-lib-builder, so that I can master it as soon as possible, thank you.

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

lbernstone, 2023-11-03:
> For the ESP32 Arduino Lib Builder Tested on Ubuntu, I used the following command sudo pip install --upgrade pip got the following error.

??
https://github.com/espressif/esp32-arduino-lib-builder#build-on-ubuntu-and-raspberry-pi
The assets will be in the out folder after compilation.

Mushan-bin, 2023-11-06:
@lbernstone Hello, I would like to ask if I want to create my own library through ESP32 Arduino lib-builder to change the maximum number of Bluetooth connections of ESP32 devices to more than 9, can it be realized?

Mushan-bin, 2023-11-06:
@lbernstone After I have done the following
sudo apt-get install git wget curl libssl-dev libncurses-dev flex bison gperf python python-pip python-setuptools  python-serial python-click python-cryptography python-future python-pyparsing python-pyelftools cmake ninja-build ccache  jq
sudo pip install --upgrade pip
git clone https://github.com/espressif/esp32-arduino-lib-builder
cd esp32-arduino-lib-builder
./build.sh
The resulting image is shown below
[image]
[image]
Excuse me, why is this vault empty？

Mushan-bin, 2023-11-06:
@lbernstone I want to build a library that can set the maximum number of connections for Bluetooth devices, and I would like to ask if there is a tutorial on building this library, which can let me quickly master, thank you.

closed 2025-01-15 by Parsaabasi