espressif/arduino-esp32 #8835: ESP32DEV board stuck in RTCWDT_RTC_RESET reboot after fresh PlatformIO install

justoke, 2023-11-04:
### Board

ESP DEV Module

### Device Description

esp32dev board which is being used to run a generic module with ESP32-WROOM-32 module

### Hardware Configuration

No.

### Version

v2.0.11

### IDE Name

PlatformIO

### Operating System

Windows 11

### Flash frequency

40MHz

### PSRAM enabled

yes

### Upload speed

115200

### Description

I've got a code base about 2 years old which has been working fine. I just installed fresh install of PlatformIO and VS Code on a new Windows PC.   Although the project builds and outputs the bin file, the command line has all the supporting files as usual, the binary once released to an ESP32 Dev module fails to run and just emits the RTCWDT_RTC_RESET in a continuous cycle. Only when I take a binary built on the old PC and flash the device does it work as it did before:

I am using min_spiffs.csv 

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x1E0000,
app1,     app,  ota_1,   0x1F0000,0x1E0000,
spiffs,   data, spiffs,  0x3D0000,0x20000,
coredump, data, coredump,0x3F0000,0x10000,
```

**Command line release**
```
"C:\Users\sb\.platformio\penv\Scripts\python.exe" 
"C:\Users\sb\.platformio\packages\tool-esptoolpy\esptool.py" --chip esp32 --port "COM3" 
--baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio 
--flash_freq 40m --flash_size 4MB 
0x1000 E:\PlatformIO\Projects\kctech\.pio\build\esp32dev_4M\bootloader.bin 
0x8000 E:\PlatformIO\Projects\kctech\.pio\build\esp32dev_4M\partitions.bin 
0xe000 C:\Users\sb\.platformio\packages\framework-arduinoespressif32\tools\partitions\boot_app0.bin 
0x10000 .pio\build\esp32dev_4M\firmware1.bin
```
**Build and Release**
```
Processing esp32dev_4M (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11)
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ strict
Building in release mode
Retrieving maximum program size .pio\build\esp32dev_4M\firmware.elf
Checking size .pio\build\esp32dev_4M\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [==        ]  17.8% (used 58416 bytes from 327680 bytes)
Flash: [========= ]  85.4% (used 1679413 bytes from 1966080 bytes)
Configuring upload protocol...
AVAILABLE: cmsis-dap, esp-bridge, esp-prog, espota, esptool, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa
CURRENT: upload_protocol = esptool
Looking for upload port...
Using manually specified: COM3
Uploading .pio\build\esp32dev_4M\firmware.bin
esptool.py v4.5.1
Serial port COM3
Connecting....
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: d4:d4:da:40:a1:98
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Flash will be erased from 0x00001000 to 0x00005fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x001abfff...
Compressed 17536 bytes to 12203...
Writing at 0x00001000... (100 %)
Wrote 17536 bytes (12203 compressed) at 0x00001000 in 0.7 seconds (effective 188.4 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 146...
Writing at 0x00008000... (100 %)
Wrote 3072 bytes (146 compressed) at 0x00008000 in 0.3 seconds (effective 82.2 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.3 seconds (effective 249.4 kbit/s)...
Hash of data verified.
Compressed 1685168 bytes to 1178648...
Writing at 0x00010000... (1 %)
.....
Writing at 0x001a605a... (100 %)
Wrote 1685168 bytes (1178648 compressed) at 0x00010000 in 32.9 seconds (effective 409.5 kbit/s)...
Hash of data verified.
```

**Reset cycle ad infinitum**

ets Jun  8 2016 00:22:57

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57
```

I have tried different boards, erasing the flash and using the Espressif Download tool as an alternative to PlatformIO just to try but the problem seems to be that the offsets in the binary are incorrect or something has changed that seems to have broken the release binary.

Any suggestions greatly appreciated. I'm stuck on this one for the last 24 hours without being any closer to a resolution.

### Sketch

```cpp
The most relevant would be to use the partitions configuration as this seems to be most relevant:
[env:esp32dev]
board_build.partitions = min_spiffs.csv
platform = espressif32
board = esp32dev
```

### Debug Message

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
```

### Other Steps to Reproduce

I see I have v2.0.11 of the ESP32 Arduino framework and have tried updating the platform, so not sure how I can get the latest version as running the "pio update" command indicates everything is up to date.
 
I am using the standard build and release workflows that PlatformIO provides. The code compiles, the bin and supporting files are created and the release step deploys the code. But the device does run the deployed code and stays stuck in this reset cycle. Using a binary created a few weeks ago and uploading using the same commands that are issued by the PlatformIO release workflow results in working device.

I tried a fresh installation on my laptop and the result is the same, the device does not start:
```
Processing esp32dev_4M (platform: espressif32; board: esp32dev; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32dev.html
PLATFORM: Espressif 32 (6.4.0) > Espressif ESP32 Dev Module
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20011.230801 (2.0.11) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-04:
Unless you are saying that a Hello World sketch causes this, you need to provide your code.  Something in your initialization is blocking the cpu.

justoke, 2023-11-04:
I think it is related to the partition scheme. There was no code change, all I had done was build a new PC after mine had died and I wasn't able to get any of my modules to boot up. I didn't recall the Espressif32 version I had previously. It turns out this is a common issue with Release 6.4.0 as I came across references to this reboot loop on the PlatformIO forum. The workaround is to pin it at 6.3.2
```
[env:esp32dev]
board_build.partitions = min_spiffs.csv
platform = espressif32 @ 6.3.2
board = esp32dev
```

This was the issue from another project that provided the version pin suggestion.
https://github.com/theelims/ESP32-sveltekit/issues/5

closed 2023-11-04 by justoke

lbernstone, 2023-11-04:
I would normally say let's try to fix it instead of workaround, but 3.0.0 will be out soon, and will have a whole new set of migration issues.

justoke, 2023-11-05:
I agree a fix would be better. As I have a commercial product that is being rolled out, I just don't have time to try and find the root cause. I'll revisit the 6.4.0 release at some point and see if I can offer any more insights. I did create a simple blink sketch using the same partition scheme and it did work using the 6.4.0 release. I use tasks and RTOS in my commercial code base and one of the issues I found while trying to figure out the issue, mentioned something about the CPU being maxed out and not servicing the IDLE-task. 
https://github.com/theelims/ESP32-sveltekit/issues/5#issuecomment-1714579292