espressif/arduino-esp32 #8837: A time less than 2017 makes `ESP32Time.getTime()` slow.

luckyabsoluter, 2023-11-05:
### Board

Board hardware or firmware ESP32Time RTC

### Device Description

ESP32

### Hardware Configuration

sorry. many pin attached.

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

40Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

A time less than 2017 year makes `ESP32Time.getTime()` slow. (about 5 seconds)
Is it a firmware issue?

### Sketch

```cpp
ESP32Time rtc;

void main() {
    Serial.begin(115200);
    rtc.setTime(2016, 1, 1, 1, 1, 1, 1);
}

void loop() {
    unsigned long st = millis();
    rtc.getTime();
    unsigned long et = millis() - st;
    Serial.println(et);
}
```
```

### Debug Message

```plain
.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-11-05:
Just out of curiosity and because we are already 2023, what is the use case to set the time to 7+ years ago?

luckyabsoluter, 2023-11-05:
@me-no-dev
My RTC module was reset and I struggled with a weird bug. The same project, same code but suddenly stopped working, and the problem became a mystery.

lbernstone, 2023-11-05:
Seems a lot more likely it is making the rtc grumpy, so it is slow to respond.  Look in their library.  ESP32 runs with a time back in the 1970s in many, many environments :)

luckyabsoluter, 2023-11-06:
My bad. I had doubts about the absence of a loop in the ESP32Time library, but after upgrading the version from 1.0.2 to 2.0.4, the problem was resolved without any issues.

closed 2023-11-06 by luckyabsoluter

6v6gt-duino, 2024-01-26:
I also found this at https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-time.c

I haven't traced back your function call but the implication from this is that the ESP32  may want to reject a time with a year part of  2016 or earlier after waiting a timeout period. This matches your problem description. 

```
bool getLocalTime(struct tm * info, uint32_t ms)
{
    uint32_t start = millis();
    time_t now;
    while((millis()-start) <= ms) {
        time(&now);
        localtime_r(&now, info);
        if(info->tm_year > (2016 - 1900)){
            return true;
        }
        delay(10);
    }
    return false;
}
```