espressif/arduino-esp32 #8838: BLECharacteristic::setValue()/notify() works after pairing, but never again in BLE HID sketch

afpineda, 2023-11-05:
### Board

Wemos Lolin32 Lite

### Device Description

I'm using a standard Wemos Lolin32 Lite board (ESP32) and nothing else.

### Hardware Configuration

Nothing attached to GPIO pins.

### Version

v2.0.11

### IDE Name

VSCode with Arduino extension

### Operating System

Windows 10

### Flash frequency

80 Mhz

### PSRAM enabled

no

### Upload speed

921600

### Description

The sketch is a test unit for a BLE HID gamepad, which sends an input report at timed intervals.
Just after pairing with Windows 10, the sketch works as expected. After reset or power-off, it does not work anymore. The device is properly connected but input reports are not received.
No error messages at the serial port.

The full sketch can be found here:

[https://github.com/afpineda/OpenSourceSimWheelESP32/tree/BugFix/src/QualityControls/UnitTests/ESPBLEimplTest](https://github.com/afpineda/OpenSourceSimWheelESP32/tree/BugFix/src/QualityControls/UnitTests/ESPBLEimplTest)

The expected behavior is described there, too.
I've tracked down the bug up to two lines at source file `hidImplementation_ESPBLE.cpp`:
```c++
void hidImplementation::reportInput(...)
{
  ...
  inputGamepad->setValue((const uint8_t *)report, GAMEPAD_REPORT_SIZE);
  inputGamepad->notify();
  ...
}
```

Those lines are executed, but nothing happens.

### Sketch

```cpp
void hidImplementation::reportInput(
        inputBitmap_t globalState,
        bool altEnabled,
        uint8_t POVstate = 0)
{
    if (connectionStatus.connected)
    {
        uint8_t report[GAMEPAD_REPORT_SIZE];
        if (altEnabled)
        {
            report[0] = 0;
            report[1] = 0;
            report[2] = 0;
            report[3] = 0;
            report[4] = 0;
            report[5] = 0;
            report[6] = 0;
            report[7] = 0;
            report[8] = ((uint8_t *)&globalState)[0];
            report[9] = ((uint8_t *)&globalState)[1];
            report[10] = ((uint8_t *)&globalState)[2];
            report[11] = ((uint8_t *)&globalState)[3];
            report[12] = ((uint8_t *)&globalState)[4];
            report[13] = ((uint8_t *)&globalState)[5];
            report[14] = ((uint8_t *)&globalState)[6];
            report[15] = ((uint8_t *)&globalState)[7];
        }
        else
        {
            report[0] = ((uint8_t *)&globalState)[0];
            report[1] = ((uint8_t *)&globalState)[1];
            report[2] = ((uint8_t *)&globalState)[2];
            report[3] = ((uint8_t *)&globalState)[3];
            report[4] = ((uint8_t *)&globalState)[4];
            report[5] = ((uint8_t *)&globalState)[5];
            report[6] = ((uint8_t *)&globalState)[6];
            report[7] = ((uint8_t *)&globalState)[7];
            report[8] = 0;
            report[9] = 0;
            report[10] = 0;
            report[11] = 0;
            report[12] = 0;
            report[13] = 0;
            report[14] = 0;
            report[15] = 0;
        }
        report[16] = (uint8_t)clutchState::combinedAxis;
        report[17] = (uint8_t)clutchState::leftAxis;
        report[18] = (uint8_t)clutchState::rightAxis;
        report[19] = POVstate;
        if (notifyConfigChanges) {
            report[19] |= (RID_FEATURE_CONFIG << 4);
            notifyConfigChanges = false;
        }
        inputGamepad->setValue((const uint8_t *)report, GAMEPAD_REPORT_SIZE);
        inputGamepad->notify();
    }
}

void hidImplementation::begin(
    std::string deviceName,
    std::string deviceManufacturer,
    bool enableAutoPowerOff)
{
    if (hid == nullptr)
    {
        // Auto power-off initialization
        if (enableAutoPowerOff)
        {
            esp_timer_create_args_t args;
            args.callback = &autoPowerOffCallback;
            args.arg = nullptr;
            args.name = nullptr;
            args.dispatch_method = ESP_TIMER_TASK;
            ESP_ERROR_CHECK(esp_timer_create(&args, &autoPowerOffTimer));
        }

        // Stack initialization
        NimBLEDevice::init(deviceName);
        pServer = NimBLEDevice::createServer();
        pServer->setCallbacks(&connectionStatus);

        // HID initialization
        hid = new NimBLEHIDDevice(pServer);
        if (!hid) {
            log_e("Unable to create HID device");
            abort();
        }
        hid->manufacturer()->setValue(deviceManufacturer); // Workaround for bug in `hid->manufacturer(deviceManufacturer)`
        hid->pnp(BLE_VENDOR_SOURCE, BLE_VENDOR_ID, BLE_PRODUCT_ID, PRODUCT_REVISION);
        hid->hidInfo(0x00, 0x01);
        NimBLECharacteristic* modelString_chr = hid->deviceInfo()->createCharacteristic((uint16_t)0x2A24,NIMBLE_PROPERTY::READ);
        modelString_chr->setValue(deviceName);
        hid->reportMap((uint8_t *)hid_descriptor, sizeof(hid_descriptor));

        // Create HID reports
        inputGamepad = hid->inputReport(RID_INPUT_GAMEPAD);
        configReport = hid->featureReport(RID_FEATURE_CONFIG);
        capabilitiesReport = hid->featureReport(RID_FEATURE_CAPABILITIES);
        if (!inputGamepad || !configReport || !capabilitiesReport) {
            log_e("Unable to create HID report characteristics");
            abort();
        }
        configReport->setCallbacks(&configFRCallbacks);
        capabilitiesReport->setCallbacks(&capabilitiesFRCallbacks);

        // Configure BLE advertising
        NimBLEAdvertising *pAdvertising = pServer->getAdvertising();
        pAdvertising->setAppearance(HID_GAMEPAD);
        pAdvertising->setScanResponse(true);
        pAdvertising->setMinPreferred(0x06);
        pAdvertising->setMinPreferred(0x12);
        pAdvertising->addServiceUUID(hid->hidService()->getUUID());
        pAdvertising->addServiceUUID(hid->batteryService()->getUUID());
        pAdvertising->addServiceUUID(hid->deviceInfo()->getUUID());

        // Configure BLE security
        NimBLESecurity *pSecurity = new NimBLESecurity();
        pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

        // Start services
        hid->startServices();
        hid->setBatteryLevel(UNKNOWN_BATTERY_LEVEL);
        connectionStatus.onDisconnect(pServer); // start advertising
    }
}
```

### Debug Message

```plain
I found no error messages. Serial output goes as expected:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
--START--
*** DISCOVERING ***
Advertising
--GO--
(Waiting for connection)
(Waiting for connection)
BLE connected
*** CONNECTED ***
reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)reportInput (connected)
...
```

### Other Steps to Reproduce

I found the same behavior both with 2.0.6 and 2.0.11 versions of Arduino-ESP32.
When replacing the Arduino-ESP32 BLE library with [NimBLE](https://github.com/h2zero/NimBLE-Arduino) it works as expected.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-11-09:
Thanks for reporting, but it seems to be a 3rd party HID library. 
I can help with Arduino Core issues, but in that case I need the smallest possible sample code that I can use to reproduce the issue.

Please create an Arduino sketch that only uses the BLE Arduino Library to reproduce the problem.

afpineda, 2023-11-10:
As you ask, this is a single file, Arduino-only, sketch which exhibits the same behavior:

```c++
/**
 * @author Ángel Fernández Pineda. Madrid. Spain.
 * @date 2023-11-10
 * @brief Unit Test for bug #8838
 */

#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLEHIDDevice.h>

// Constants
#define RID_INPUT_GAMEPAD 0x01
#define RID_FEATURE_CAPABILITIES 0x02
#define RID_FEATURE_CONFIG 0x03
#define GAMEPAD_REPORT_SIZE 20
#define CAPABILITIES_REPORT_SIZE 8
#define CONFIG_REPORT_SIZE 4
#define CONTROLLER_TYPE_GAMEPAD 0x05
#define BUTTON_COUNT 128
#define BLE_VENDOR_SOURCE 0x00
#define BLE_VENDOR_ID 0x1d50
#define BLE_PRODUCT_ID 0xffff
#define PRODUCT_REVISION 0x0100

// types
typedef uint64_t inputBitmap_t;

// HID report descriptor

static const uint8_t hid_descriptor[] = {
    0x05, 0x01,                    // UsagePage(Generic Desktop[1])
    0x09, CONTROLLER_TYPE_GAMEPAD, // UsageId
    0xA1, 0x01,                    // Collection(Application)

    // ___ INPUT REPORT ___
    0x85, RID_INPUT_GAMEPAD, //     ReportId

    //     Buttons (128 bits=16 bytes)
    0x05, 0x09, //     UsagePage(Button[9])
    0x19, 0x01, //     UsageIdMin(Button 1[1])
    0x29, 0x80, //     UsageIdMax(Button 128[128])
    0x15, 0x00, //     LogicalMinimum(0)
    0x25, 0x01, //     LogicalMaximum(1)
    0x95, 0x80, //     ReportCount(128)
    0x75, 0x01, //     ReportSize(1)
    0x81, 0x02, //     Input(Data, Variable, Absolute, NoWrap, Linear, PreferredState, NoNullPosition, BitField)

    //     axis (1 byte)
    0x05, 0x01, //     UsagePage(Generic Desktop[1])
    0x09, 0x35, //     UsageId(Rz[53])
    // 0x15, 0x00, //     LogicalMinimum
    // 0x25, CLUTCH_FULL_VALUE, //     LogicalMaximum
    0x26, 0xFE, 0x00, //     LogicalMaximum(254)
    0x95, 0x01,       //     ReportCount(1)
    0x75, 0x08,       //     ReportSize(8)
    0x81, 0x02,       //     Input(Data, Variable, Absolute, NoWrap, Linear, PreferredState, NoNullPosition, BitField)

    //     axis (1 byte)
    0x09, 0x34, //     UsageId(Ry[52])
    0x81, 0x02, //     Input(Data, Variable, Absolute, NoWrap, Linear, PreferredState, NoNullPosition, BitField)

    //     axis (1 byte)
    0x09, 0x33, //     UsageId(Rx[51])
    0x81, 0x02, //     Input(Data, Variable, Absolute, NoWrap, Linear, PreferredState, NoNullPosition, BitField)

[... 156 lines cut ...]
// ----------------------------------------------------------------------------
// Arduino entry points
// ----------------------------------------------------------------------------

void setup() {
  esp_log_level_set("*", ESP_LOG_VERBOSE);
  Serial.begin(115200);
  Serial.println("--START--");
  hid_begin("bug_8838", "afpineda");
  Serial.println("--GO--");
}

inputBitmap_t data = 0;

void loop() {
  if (!connectionStatus.connected)
    Serial.println("(Waiting for connection)");
  else {
    hid_reportInput(data);
    data = data + 1;
  }

  delay(1000);
}
```

afpineda, 2023-11-10:
**Sorry. The code I posted in the open message is wrong**. Not the buggy one.  My mistake.

afpineda, 2023-11-10:
**Note**: you may use this online app for testing:
[https://hardwaretester.com/gamepad](https://hardwaretester.com/gamepad)

SuGlider, 2023-11-10:
@afpineda - I have tried the code, but it sounds like it works fine.
I used an ESP32 WROOM DevKit, Windows 11, Turned on Debug level INFO, used Win Bluetooth Configurations to find the ESP32, Pair it and start seeing the Gamepad Tester displaying the data.

After that I've reset the ESP32 (using the EN button) and it reconnected immediately.
Another test was to remove the ESP32 power for about 5 minutes and power it back again.
Again, it reconnected immediately and the Gamepad Tester also started displaying data.
Even right after re-flashing the ESP32 using the Arduino IDE, it connects immediately after flashing is done.

I used Arduino Core 2.0.14. -- the only difference I found from the information about this issue.

I have attached the build folder with the bin file (below), in case you want to try it.
[arduino_build_97102.zip](https://github.com/espressif/arduino-esp32/files/13319412/arduino_build_97102.zip)

afpineda, 2023-11-10:
I'm clueless here. I compiled and uploaded (this time with Arduino-IDE) the test sketch to a ESP32-devkit-C v4 board using different core versions and tested them both in Windows 10 and 11 (different computers with different hardware):
- v2.0.14 / Windows 10: paired and connected, but not detected as a HID gamepad. **This is something new**.
- v2.0.14 / Windows 11: same behavior.
- v2.0.13 / Windows 10: same behavior.
- v2.0.13 / Windows 11: same behavior.

Then I uploaded your ZIP binaries (thanks for that) to the ESP32-devkit-C board:
- Windows 10: same behavior.
- Windows 11: same behavior.

No luck. Any ideas?

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

SuGlider, 2025-02-09:
Reopen to verify 2902 initialization in BLE HID Class.

lucasssvaz, 2025-08-30:
We recently moved from Bluedroid to NimBLE. Could you please test it with the newest version ?

afpineda, 2025-11-30:
**UPDATE**
The issue with the device not being discoverable is caused by **Windows 11 version 25H2**.
[Uninstalling the update](https://pureinfotech.com/uninstall-windows-11-25h2/#uninstall_windows11_25h2_windowsupdate) makes the device discoverable again.
However, it now gets stuck in the pairing process.

afpineda, 2026-01-22:
I updated to 3.3.6.

**Still not working**.
The device is discovered in Windows 11, but it fails to pair.