espressif/arduino-esp32 #8841: RMT test case Infinite Loop even with ERASED all Firmware

gasparo, 2023-11-06:
### Board

ESP32-C6

### Device Description

PIN 23 CONNECTED WS2812b.

### Hardware Configuration

PIN 23 CONNECTED WS2812b.

### Version

other

### IDE Name

Arduino IDE

### Operating System

MAC OS 14.0

### Flash frequency

80 MHZ

### PSRAM enabled

yes

### Upload speed

921600

### Description

I tested the RMT function, to control WS2812b. I tested the code, but, it started to FREEZE on the BOOT.

I try comment all the CODE (clean code), ENABLE "ERASE ALL FLASH BEFORE UPLOAD", and re-upload it. But the CRASH on BOOT continue in LOOP.

### Sketch

```cpp
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  https://www.arduino.cc/en/Tutorial/BuiltInExamples/Blink
*/

#define BUILTIN_RGBLED_PIN   23

#define NR_OF_LEDS  3
#define NR_OF_ALL_BITS 24*NR_OF_LEDS

//#include "Freenove_WS2812_Lib_for_ESP32.h"

#define CHANNEL		0

//rmt_data_t led_data[NR_OF_ALL_BITS];

//Freenove_ESP32_WS2812 strip = Freenove_ESP32_WS2812(LEDS_COUNT, LEDS_PIN, CHANNEL, TYPE_GRB);

 //WS2812B
// the setup function runs once when you press reset or power the board
void setup() {
  Serial.begin(9600);
  Serial.println("STARTING");  
  delay(5000);
  /*
  if (!rmtInit(BUILTIN_RGBLED_PIN, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
        Serial.println("init sender failed\n");
    }
    Serial.println("real tick set to: 100ns");
*/

  
  delay(5000);
  // initialize digital pin LED_BUILTIN as an output.
  //if (!rmtInit(BUILTIN_RGBLED_PIN, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
  //      Serial.println("init sender failed\n");
  //  }
  pinMode(15, OUTPUT);
  //while (!strip.begin());
  //strip.setBrightness(20);  
}

// the loop function runs over and over again forever
void loop() {
  //loopLedIR_ON_OFF();             // wait for a second
  //loopLed();
}

void loopLedIR_ON_OFF()
{
  digitalWrite(15, HIGH);  // turn the LED on (HIGH is the voltage level)
  delay(2000);       
  Serial.println("ON");               // wait for a second
  digitalWrite(15, LOW);   // turn the LED off by making the voltage LOW
  delay(2000);   
  Serial.println("OFF");  

}
  int color[] =  { 0x55, 0x11, 0x77 };  // Green Red Blue values
  int led_index = 1;

/*

void loopLed()
{

    // Init data with only one led ON
    int led, col, bit;
    int i=0;
    for (led=0; led<NR_OF_LEDS; led++) {
        for (col=0; col<3; col++ ) {
            for (bit=0; bit<8; bit++){
                if ( (color[col] & (1<<(7-bit))) && (led == led_index) ) {
                    led_data[i].level0 = 1;
                    led_data[i].duration0 = 8;
                    led_data[i].level1 = 0;
                    led_data[i].duration1 = 4;
                } else {
                    led_data[i].level0 = 1;
                    led_data[i].duration0 = 4;
                    led_data[i].level1 = 0;
                    led_data[i].duration1 = 8;
                }
                i++;
            }
        }
    }
    // make the led travel in the pannel
    if ((++led_index)>=NR_OF_LEDS) {
        led_index = 0;
    }
    // Send the data and wait until it is done
    rmtWrite(BUILTIN_RGBLED_PIN, led_data, NR_OF_ALL_BITS, RMT_WAIT_FOR_EVER);
    delay(100);
    Serial.println("LOOP");
}

*/
```

### Debug Message

```plain
14:04:24.612 -> 4087e3e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:04:24.643 -> 4087e400: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
14:04:24.643 -> 4087e420: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000002 0x00000000
14:04:24.643 -> 4087e440: 0x60002000 0x00000100 0x00000020 0x4001b1c0 0x70f220d4 0xb0666082 0xb5f6886d 0x00000000
14:04:24.643 -> 4087e460: 0x786a6558 0x00000000 0x00010000 0x200203e9 0x4086c410 0x40875728 0x00000594 0x000000ee
14:04:24.674 -> 4087e480: 0x0000000d 0x00ffff00 0x01000000 0x420038f0 0x7533885e 0x8c5a9bca 0x8a57e948 0x675777fb
14:04:24.674 -> 4087e4a0: 0xefe1f630 0x37537b19 0x8f3a9fa2 0x0a23ea69 0x1abef30c 0x00000000 0x00000000 0x00000000
14:04:24.674 -> 4087e4c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x4087e5f0 0x00000101 0x00000002
14:04:24.674 -> 4087e4e0: 0x8c5a9bca 0x8a57e948 0x675777fb 0xefe1f630 0x37537b19 0x8f3a9fa2 0x0a23ea69 0x1abef30c
14:04:24.706 -> 
14:04:24.706 -> 
14:04:24.706 ->
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-06:
The RMT API has changed in breaking ways. See the examples at https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/RMT
https://docs.espressif.com/projects/arduino-esp32/en/latest/migration_guides/2.x_to_3.0.html#rmt

SuGlider, 2023-11-07:
@gasparo - Indeed RMT isn't working properly for the C6 when using Arduino Core 3.0.0-Alpha2.
There is an error sent in the console:
```
E (83) rmt: rmt_new_tx_channel(211): invalid interrupt priority:1082651320
init sender failed
```

At this time, it is already fixed in ESP32 Arduino GitHub master branch.
You may want to try to fix it directly in `Arduino15/packages/esp32/hardware/esp32/3.0.0-alpha2/cores/esp32/esp32-hal-rmt.c` files by applying the path from https://github.com/espressif/arduino-esp32/pull/8813 (RMT TX interrupt priority)
Maybe also apply this for RX RMT, if necessary: https://github.com/espressif/arduino-esp32/commit/730d827553cf3ae357eb62a3b66f97b13a255177

Some information that may help you:
1- ESP32-C6 DevKit uses GPIO 8 for its builtin RGB LED. Please set `#define BUILTIN_RGBLED_PIN   8` in the `RMTWriteNeoPixel.ino` RMT example. In the `pins_arduino.h` file for all Espressif devkits, we define `PIN_NEOPIXEL` as being the RGBLED Pin, when it exists. You can also use it there, like `#define BUILTIN_RGBLED_PIN PIN_NEOPIXEL`

2- the `digìtalWrite(RGB_LED, HIGH | LOW)` only works when the pin used is the `RGB_BUILTIN` one defined in `pins_arduino.h`
For all other pins, you can use the new ESP32 Arduino function `neopixelWrite(WS2812b_Pin, Red, Green, Blue);`, Reg, Green and Blue shall be in the range of 0..255.
Blink Example using pin 23:
``` cpp
void blinkRGBLED(uint32_t count, uint32_t time_ms) {
  const uint8_t LED_Pin = 23;  // change it to any other valid GPIO
  for ( ; count; count--) {
    neopixelWrite(LED_Pin, 255, 255, 255);  // RGB = White (on)
    delay(time_ms);
    neopixelWrite(LED_Pin, 0, 0, 0); // RGB = Black (off)
    delay(time_ms);
  }
}
``` 
3- There is an issue with the function `neopixelWrite()` that will prevent it from working for more than one pin... If it is called with GPIO15 and latter with GPIO8, it will initialize only the first pin and not the second one... Therefore, at this time, it can only be used for a single GPIO...

It is possible to use the code from `RMTWriteNeoPixdel.ino` to create some better generic WS2812b function, such as:
I just tested it in the ESP32-C6 devkit and it works fine, even if there is a WS2812b LED connected to GPIO23.
Just remember to apply the fix from #8813 

``` cpp
#include "esp32-hal-rgb-led.h"

void my_neopixelWrite(uint8_t pin, uint8_t red_val, uint8_t green_val, uint8_t blue_val) {
  rmt_data_t led_data[24];
  if (!rmtInit(pin, RMT_TX_MODE, RMT_MEM_NUM_BLOCKS_1, 10000000)) {
    log_e("RGB LED driver initialization failed!");
    return;
  }

  int color[] = {green_val, red_val, blue_val};  // Color coding is in order GREEN, RED, BLUE
  int i = 0;
  for (int col = 0; col < 3; col++ ) {
    for (int bit = 0; bit < 8; bit++) {
      if ((color[col] & (1 << (7 - bit)))) {
        // HIGH bit
        led_data[i].level0 = 1; // T1H
        led_data[i].duration0 = 8; // 0.8us
        led_data[i].level1 = 0; // T1L
        led_data[i].duration1 = 4; // 0.4us
      } else {
        // LOW bit
        led_data[i].level0 = 1; // T0H
        led_data[i].duration0 = 4; // 0.4us
        led_data[i].level1 = 0; // T0L
        led_data[i].duration1 = 8; // 0.8us
      }
      i++;
    }
  }
  rmtWrite(pin, led_data, RMT_SYMBOLS_OF(led_data), RMT_WAIT_FOR_EVER);
}

void blinkRGBLED(uint32_t count, uint32_t time_ms) {
  const uint8_t LED_Pin = 23;  // change it to any other valid GPIO
  for ( ; count; count--) {
    my_neopixelWrite(LED_Pin, 255, 255, 255);  // RGB = White (on)
    Serial.println("HIGH");
    delay(time_ms);
    my_neopixelWrite(LED_Pin, 0, 0, 0); // RGB = Black (off)
    Serial.println("LOW");
    delay(time_ms);
  }
}

void setup() {
  // No need to initialize the RGB LED
  blinkRGBLED(5, 500);
}

// the loop function runs over and over again forever
void loop() {
#ifdef RGB_BUILTIN
  digitalWrite(RGB_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  digitalWrite(RGB_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);

  my_neopixelWrite(PIN_NEOPIXEL, RGB_BRIGHTNESS, 0, 0); // Red
  delay(1000);
  my_neopixelWrite(PIN_NEOPIXEL, 0, RGB_BRIGHTNESS, 0); // Green
  delay(1000);
  my_neopixelWrite(PIN_NEOPIXEL, 0, 0, RGB_BRIGHTNESS); // Blue
  delay(1000);
  my_neopixelWrite(PIN_NEOPIXEL, 0, 0, 0); // Off / black
  delay(1000);
#endif
}
```

I hope that this may help you.

closed 2023-11-09 by me-no-dev