espressif/arduino-esp32 #8843: esp.emac: no mem for receive buffer

hzhh110, 2023-11-07:
### Board

ESP32-WROOM-32e

### Device Description

platform = espressif32 @ ~5.2.0
for sdk = 4.4.2
Network cable allocation space is there is no memory, and can not be restored, I free FreeHeap after the same can not be restored, need to call which interface can be re-allocated?

[WiFiGeneric.cpp:1438] hostByName(): DNS Failed for ****

### Hardware Configuration

Eth - for ETH_PHY_LAN8720

### Version

latest master (checkout manually)

### IDE Name

Platfrom

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

esp.emac: no mem for receive buffer 
[WiFiGeneric.cpp:1438] hostByName(): DNS Failed for

When esp.emac: no mem for receive buffer is displayed, subsequent devices cannot be recovered, even if I release the stack

### Sketch

```cpp
esp.emac: no mem for receive buffer 
[WiFiGeneric.cpp:1438] hostByName(): DNS Failed for

When esp.emac: no mem for receive buffer is displayed, subsequent devices cannot be recovered, even if I release the stack
```

### Debug Message

```plain
esp.emac: no mem for receive buffer 
[WiFiGeneric.cpp:1438] hostByName(): DNS Failed for

When esp.emac: no mem for receive buffer is displayed, subsequent devices cannot be recovered, even if I release the stack
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hzhh110, 2023-11-07:
As long as I start using Bluetooth communication, the network cable can not connect to the Internet, but the LAN can communicate, or connect to the Internet line communication, I go to use Bluetooth again, after a while the same, and then I try to release the Bluetooth FreeHeap Size:179336 after 120 seconds, but the network cable keeps appearing hostByName(): DNS Failed for and cannot be recovered until the device is restarted

hzhh110, 2023-11-07:
Bluetooth once started to occupy a lot, is there a way to optimize and reduce?

FreeHeap BLE1 :191456
initBLEJH-P130-A0009
FreeHeap BLE2 :105288

void initBLE(const char *name)
{
DebugPrint("initBLE");
DebugPrintln(name);
BLEDevice::init(name);
#if isESP32
BLEDevice::setPower(ESP_PWR_LVL_P7,ESP_BLE_PWR_TYPE_DEFAULT);
#else
// #if isESP32_C3
BLEDevice::setPower(ESP_PWR_LVL_P18,ESP_BLE_PWR_TYPE_DEFAULT);
#endif
BLEServer *pServer = BLEDevice::createServer();
_pServer = pServer;
pServer->setCallbacks(new MyServerCallbacks());
BLEService *pService = pServer->createService(SERVICE_UUID);
pTXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_TX, BLECharacteristic::PROPERTY_NOTIFY);
pTXCharacteristic->addDescriptor(new BLE2902());
BLECharacteristic *pRXCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID_RX, BLECharacteristic::PROPERTY_WRITE);
pRXCharacteristic->setCallbacks(new MyCallbacks());
pService->start();
pServer->getAdvertising()->start();
}

hzhh110, 2023-11-08:
I wonder if it is possible that I set ETH.config(), then removed it, and then the router recorded it, causing the problem? And how do I get dns1 and dns2 in config(IPAddress local_ip, IPAddress gateway, IPAddress subnet, IPAddress dns1, IPAddress dns2)? Why am I getting Intranet ones and not something like 114.114.114.114 or 8.8.8.8?

hzhh110, 2023-11-08:
E (9267) esp.emac: no mem for receive buffer
E (9271) esp.emac: no mem for receive buffer
[WiFiGeneric.cpp:1438] hostByName(): DNS Failed

If [WiFiGeneric.cpp:1438] hostByName(): DNS Failed appears, it cannot be restored
You can take a router, and then the router is connected to the Internet at the beginning, after the router's Internet connection network cable is removed, you will find that you can not reconnect, has been DNS Failed

hzhh110, 2023-11-08:
In addition, when you set ETH.config() and WiFi.config(), you cannot connect to the Internet when dns1 and dns2 are 0

lbernstone, 2023-11-08:
[NimBLE](https://github.com/h2zero/NimBLE-Arduino) uses significantly less memory than the built-in library.

hzhh110, 2023-11-08:
> [NimBLE](https://github.com/h2zero/NimBLE-Arduino) uses significantly less memory than the built-in library.

Thank you, I will try this solution later, in fact, my biggest problem is the DNS Internet can not recover the problem, is there a way to solve this?

mrengineer7777, 2023-12-06:
Possibly related #8672

closed 2025-01-15 by Parsaabasi