espressif/arduino-esp32 #8847: How do I get the correct dns1 and dns2?

hzhh110, 2023-11-08:
### Related area

nodemcu-32s

### Hardware specification

esp32

### Is your feature request related to a problem?

How do I get the correct dns1 and dns2? Why are all the Intranet information such as 192.168.3.1 obtained? Even if my gateway is 192.168.2.1 but I get 192.168.3.1 or something like ETH.dnsIP(0).toString(); ETH.dnsIP(1).toString();

### Describe the solution you'd like

How do I get the correct dns1 and dns2? Why are all the Intranet information such as 192.168.3.1 obtained? Even if my gateway is 192.168.2.1 but I get 192.168.3.1 or something like ETH.dnsIP(0).toString(); ETH.dnsIP(1).toString();

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

SuGlider, 2023-11-08:
Is it a question about how to set DNS server IP when using Ethernet?
@me-no-dev - Can you help it?

me-no-dev, 2023-11-08:
@hzhh110 I can help if you provide a proper explanation of how and what network interfaces you have configured on the ESP, what are active and maybe a small example sketch. As-is, you just shoot out some IP addresses, without any context on what they actually are.

As a simple answer: the ESP32 has global DNS servers for all interfaces, it does not allow for separate DNS servers for each interface, so whichever interface has the highest priority, is the one that provided the DNS servers.

hzhh110, 2023-11-09:
[E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for 
And this is the most deadly, when I unplug the network cable on the router, and then plug it in,DNS will continue to fail, equivalent to the Internet completely failed, need to re-pull the device's network cable to restore normal

mrengineer7777, 2023-12-06:
Possibly related #8672

VojtechBartoska, 2023-12-12:
Hello @hzhh110, did those answers help you or you are still looking for some help?

VojtechBartoska, 2024-09-10:
Hello, this issue is quite old and there is no answer for some time.

To keep the issue tracker of this project manageable, I am closing this ticket as expired. 

If needed, you can reopen it. 

Thanks for understanding. 👍

closed 2024-09-10 by VojtechBartoska