espressif/arduino-esp32 #8855: I2C bus speed not correct on V3.0.0 Alpha2. Set to 400KHz, get 1.6MHz to 1.7MHz

tchilton, 2023-11-08:
### Board

ESP32 S2

### Device Description

Confirmed on ESP32-S2

### Hardware Configuration

Empty ESP32-S2-Saola-1

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 2.1.1

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

115200

### Description

Setting I2C bus speed to 400KHz results in I2C bus operating between 1.695MHz-1.785MHz
Bus speed also drops unexpectedly to 90.909KHz if talking to non-present devices, see bug report 8480 for more details on this.

This defect was identified whilst testing 3.0.0.Alpha2 to see if it impacted the 8480 defect. Alpha 2 seems to be more badly broken than 2.0.14, hence the separate bug report.

### Sketch

```cpp
Please see bug 8480, same sketch in use.

https://github.com/espressif/arduino-esp32/issues/8480
```

### Debug Message

```plain
No debug errors, code runs without error
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-11-08:
@me-no-dev PTAL. Thanks.

tchilton, 2023-11-09:
Further debugging against issue #8480 and the revised test code listed in that defect, seems to show that the register contents of 
 I2C_SCL_LOW_PERIOD_REG and I2C_SCL_HIGH_PERIOD_REG are incorrect in V3.0.0-Alpha 2

V2.0.14 yields clock register values (for 400KHz) of 

I2C IOBase = 0x60013000
I2C_SCL_LOW_PERIOD_REG       : 0x00000063
I2C_SCL_HIGH_PERIOD_REG      : 0x000C0034

V3.0.0-Alpha 2, for the same code and frequency gives

I2C IOBase : 0x60013000
I2C_SCL_LOW_PERIOD_REG       : 0x00000004
I2C_SCL_HIGH_PERIOD_REG      : 0x00004004

The very low clock cycle number will be why the frequency is so much higher

me-no-dev, 2023-11-09:
@tchilton can you try with github master? I added some fixes at some point, but I do not remember if it was before or after alpha2 release.

me-no-dev, 2023-11-09:
@tchilton this is the change: https://github.com/espressif/arduino-esp32/commit/93a118c8e40ecf7c5f93c3c6970a91466287e9f0

Of course... right after alpha2 release :)

tchilton, 2023-11-09:
@me-no-dev. 

Yep, of course its after the latest release, that's the live we live :-) I think the expression is "move fast and break things."

Dumb Question, what is the URL for Arduino IDE (or the Platform IO equivalent) for the Github Master ?
I've tried https://github.com/espressif/arduino-esp32/blob/master/package.json, which only goes as far as 2.0.14 

The standard link of https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json only goes as far as 3.0.0-Alpha2 which is what I've used for testing so far.

me-no-dev, 2023-11-10:
@tchilton check manual install instructions for your OS: https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation

tchilton, 2023-11-10:
@me-no-dev 

I thought I was using the latest code available with the boards manager URL being the _dev_index (https://espressif.github.io/arduino-esp32/package_esp32_dev_index.json), but having looked at its contents, its clearer that there are specific versions in there with 3.0.0-Alpha2 and there is no way to request the GitHub master branch. 

Secondly, the documentation with the title of "Windows (Manual installation)" is not at all clear that it gives a different outcome to the dev branch above. I suggest the doc is updated to say that this is the master branch.

Anyhow, having figured those things out, your request makes a lot more sense to me. I de-installed both Arduino and Espressif esp32 board support, manually installed the latest master using https://docs.espressif.com/projects/arduino-esp32/en/latest/installing.html#windows-manual-installation, recompiled test code (in defect #8480)

The relevant bits of the output are  :

ESP IDF Version String : v5.1.1-577-g6b1f40b9bf-dirty
ESP IDF Version Number : 327937

So, the IDF versions are identical to the _dev_ branch I used previously, however I'm happy to report that the clock rates can now be set and **mostly works** and values in I2C_SCL_LOW_PERIOD_REG and I2C_SCL_HIGH_PERIOD_REG look to be sensible now and scale with the different bitrates used in the tests.

Tested at 100KHz, got 100KHz, but it still drops to 1.25KHz due to defect #8480
Tested at 400KHz, got 400KHz, drops to 5KHz as above
**Tested at 1MHz, got 909KHz**, drops to 12.5KHz - No idea why this is off by 10% ??

So, defect #8480 remains in the current code base and the failure is always after 10 bytes were transmitted. I'll update the other defect accordingly. The workaround of changing clock rate and resetting it back continues to work in the mean-time.

I also tested in Platform IO as this looked to be a simpler method than the Arduino manual install method and its my preferred environment. **This FAILS to install on the current master,** but manually pushing back the platform_packages to previous versions will download and compiles properly

The branches of master and idf-release/v5.1 fails to download properly, giving the same errors during the install

```
MASTER branch

Resolving esp32-s2-saola-1-git-master dependencies...
Already up-to-date.
Updating metadata for the vscode IDE...
UserSideException: Processing esp32-s2-saola-1-git-master (platform: https://github.com/platformio/platform-espressif32.git; board: esp32-s2-saola-1; framework: arduino)
--------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s2-saola-1.html
PLATFORM: Espressif 32 (6.4.0+sha.f6ec392) > Espressif ESP32-S2-Saola-1
HARDWARE: ESP32S2 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.0.0+sha.6b287db 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
**KeyError: 'framework-arduinoespressif32-libs':
  File ".....\.platformio\penv\Lib\site-packages\platformio\builder\main.py", line 173:
    env.SConscript("$BUILD_SCRIPT")**
  File ".....\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File ".....\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File ".....\.platformio\platforms\espressif32@src-5f117260f75b328038ec9d3fd0e14a68\builder\main.py", line 346:
    target_elf = env.BuildProgram()
  File ".....\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)

<more related errors removed>
========================== [FAILED] Took 1.35 seconds ==========================

Environment                  Status    Duration
---------------------------  --------  ------------
esp32-s2-saola-1-git-master  FAILED    00:00:01.346
==================== 1 failed, 0 succeeded in 00:00:01.346 ====================
```

Reverting back to idf-release/v4.4 or release/v2.x both download and compile successfully, indicating this is in the upstream code. Either way the earlier releases  don't help in testing this bug.

Not sure if this is a known issue / if it needs a new defect raising - please advise.

tchilton, 2023-11-12:
Update following more reading and diagnostic on defect #8480. The 1Mbps data rate mismatch of 909KHz may not be a defect as the ESP32-S2 only shows supported data rates of 100KHz and 400KHz, so the 1Mbps (and 3.4MBps), although in the I2C standards do not seem to be supported by the ESP32S2 devices, which is a shame. 

@me-no-dev, please advise on the two open questions - the documentation defect and the #master branch not installing properly on PlatformIO. Do you want these raised as separate defects or are they already known about ? Once these are sorted, then the upstream fix on MASTER which sorts out the clock rate issue means that this defect can be closed

me-no-dev, 2023-11-13:
Fix for the issue here will be part of the next alpha/rc release of 3.0.0. As for PIO, this will probably be available after we release v3.0.0 final. PIO is a separate company that manages all that support.

I2C standard speeds are 100kHz and 400kHz. The higher speeds are different standards. However, the hardware on ESP32 and S2 can go even over 1MHz, but you will need to thinker manually with the registers. It was probably not stable enough for the IDF team to support it.

closed 2025-01-15 by Parsaabasi