espressif/arduino-esp32 #8857: [E][WiFiGeneric.cpp:1438] hostByName(): DNS Failed for  (There's a net in the back that can't be recovered)

hzhh110, 2023-11-09:
### Related area

ESP32-WROOM-32E

### Hardware specification

esp32

### Is your feature request related to a problem?

And this is the most deadly, when I unplug the network cable on the router, and then plug it in,DNS will continue to fail, equivalent to the Internet completely failed, need to re-pull the device's network cable to restore normal

### Describe the solution you'd like

And this is the most deadly, when I unplug the network cable on the router, and then plug it in,DNS will continue to fail, equivalent to the Internet completely failed, need to re-pull the device's network cable to restore normal

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

hzhh110, 2023-11-09:
platform = espressif32 @ ~5.2.0
SDK=4.4.2

hzhh110, 2023-11-09:
#define ETH_TYPE ETH_PHY_LAN8720
ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE);

hzhh110, 2023-11-09:
[image]

hzhh110, 2023-11-09:
The gold is the Internet network cable, the blue is the network cable connected to the ESP32 network port, when the initial connection, unplug the network cable after a while to plug in, the Internet will be completely unable to restore, unless the device restarts to restore

hzhh110, 2023-11-09:
wifi also has this problem

hzhh110, 2023-11-09:
It is easy to repeat the problem when wfi and network cable are connected, and as long as the problem occurs, it is basically impossible to recover

hzhh110, 2023-11-09:
The problem can be easily repeated when trying wifi(password error) and network cable connection, as long as the problem occurs, it is basically impossible to recover

me-no-dev, 2023-11-09:
Please try to clearly describe your problem. So far I understand that you have some issue with DNS and you are plugging and unplugging the cable, but why do you mention `wifi(password)`? Are you running both WiFi and Ethernet? What are you doing exactly, what do you expect to happen and what is happening at the end?

hzhh110, 2023-11-09:
> Please try to clearly describe your problem. So far I understand that you have some issue with DNS and you are plugging and unplugging the cable, but why do you mention `wifi(password)`? Are you [...]

1. The WiFi. Config and Eth. Config settings are on the one hand. If they are set, I cannot set DNS 1 and dns2 to 0, otherwise I cannot connect to the server. I can only set them to gateway.Get WiFi. DnsIP (0) and WiFi. DnsIP (1) and Eth. DnsIP (0) and Eth. DnsIP (1) can not be obtained, so I simply use gateway directly, this can be solved.
2. I don't set it by default. I directly connect to the WiFi. Begin and the Eth. Begin. However, many routers, as long as the [WiFiGeneric. CPP: 1438] hostByName (): DNS Failed for appears, basically there is no solution, and they will not be able to connect all the time.Cannot connect until the device is restarted
3. Reproduction method:
1. Set the wifi password to the wrong one and connect the network cable
2. Connect the wifi to a non-network and then connect the network cable
3. Connect the network cable to the router, then remove the network cable connecting the router to the Internet, plug it in again after a period of time, the mobile phone and so on are normal, and then the device has been unable to restore, can only restart the device (this wifi will also appear, but one of my three can be reconnected, two have been unable to connect).
1. The WiFi. Config and Eth. Config settings are on the one hand. If they are set, I cannot set DNS 1 and dns2 to 0, otherwise I cannot connect to the server. I can only set them to gateway.Get WiFi. DnsIP (0) and WiFi. DnsIP (1) and Eth. DnsIP (0) and Eth. DnsIP (1) can not be obtained, so I simply use gateway directly, this can be solved.
2. I don't set it by default. I directly connect to the WiFi. Begin and the Eth. Begin. However, many routers, as long as the [WiFiGeneric. CPP: 1438] hostByName (): DNS Failed for appears, basically there is no solution, and they will not be able to connect all the time.Cannot connect until the device is restarted
3. Reproduction method:
1. Set the wifi password to the wrong one and connect the network cable
2. Connect the wifi to a non-network and then connect the network cable
3. Connect the network cable to the router, then remove the network cable connecting the router to the Internet, plug it in again after a period of time, the mobile phone and so on are normal, and then the device has been unable to restore, can only restart the device (this wifi will also appear, but one of my three can be reconnected, two have been unable to connect).

hzhh110, 2023-11-09:
> Please try to clearly describe your problem. So far I understand that you have some issue with DNS and you are plugging and unplugging the cable, but why do you mention `wifi(password)`? Are you [...]

1. The WiFi. Config and Eth. Config settings are on the one hand. If they are set, I cannot set DNS 1 and dns2 to 0, otherwise I cannot connect to the server. I can only set them to gateway.Get WiFi. DnsIP (0) and WiFi. DnsIP (1) and Eth. DnsIP (0) and Eth. DnsIP (1) can not be obtained, so I simply use gateway directly, this can be solved.
2. I don't set it by default. I directly connect to the WiFi. Begin and the Eth. Begin. However, many routers, as long as the [WiFiGeneric. CPP: 1438] hostByName (): DNS Failed for appears, basically there is no solution, and they will not be able to connect all the time.Cannot connect until the device is restarted
3. Reproduction method:
(1)Set the wifi password to the wrong one and connect the network cable
(2) Connect the wifi to a non-network, and then connect the network cable.
(3) Connect the network cable to the router, then remove the network cable that connects the router to the Internet, plug it in after a period of time, the mobile phone and so on are normal, and then the device has been unable to restore, can only restart the device (this wifi will also appear, but one of my three can be reconnected, two have been unable to connect).

hzhh110, 2023-11-10:
You can try to enable wifi and network cable at the same time when the router and WiFi are in different gateways. When the wifi router has no network, it is easy to have DNS errors when the network cable is connected, that is, you can't connect it again. This is a big problem. Do you have DNS conflicts?Is the network cable shared with wifi?