espressif/arduino-esp32 #8858: There is still a problem with Parts while Trying to upload/download and delete in sub-folders

benni4000, 2023-11-09:
### Board

ESP32

### Device Description

There is still a problem with Parts while Trying to upload/download and delete in sub-folders

### Hardware Configuration

no

### Version

v2.0.14

### IDE Name

Not matter

### Operating System

Windows 7

### Flash frequency

80

### PSRAM enabled

no

### Upload speed

115200

### Description

There is still a problem with Parts while Trying to upload/download and delete in sub-folders

### Sketch

```cpp
not needed
```

### Debug Message

```plain
not needed
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

SuGlider, 2023-11-09:
@benni4000 - This issue seems incomplete as there is no enough information to reproduce it or even understand what that is about. Please explain it with more details.

benni4000, 2023-11-09:
Hello, create a sub folder on Root Directory and place a file there (S:\SubFolder\test.txt)Delete/Upload/Download arent working there.Because filepath is like (SubFolder\Subfolder\test.txt)Folder Name is double.When i am back on PC i can Show you a Foto or Video. Otherwise i try to fix it myself.Regards Benni--Diese Nachricht wurde von meinem Android Mobiltelefon mit WEB.DE Mail gesendet.

SuGlider, 2023-11-09:
I'm sorry. I'll need the photos or screenshots to see what is that about.
How is this related to Arduino?

closed 2025-01-15 by Parsaabasi