espressif/arduino-esp32 #8862: DNS WiFi conflicts with ETH

hzhh110, 2023-11-10:
### Board

ESP32-WROOM-32E

### Device Description

You can try to enable wifi and network cable at the same time when the router and WiFi are in different gateways. When the wifi router has no network, it is easy to have DNS errors when the network cable is connected, that is, you can't connect it again. This is a big problem. Do you have DNS conflicts?Is the network cable shared with wifi?

### Hardware Configuration

 begin with WiFi.begin and then try eth.begin
ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLK_MODE); 

### Version

v2.0.14

### IDE Name

Platfrom

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

DNS WiFi conflicts with ETH

### Sketch

```cpp
DNS WiFi conflicts with ETH
```

### Debug Message

```plain
DNS WiFi conflicts with ETH
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hzhh110, 2023-11-10:
​If the network cable has a network, but the WiFi does not have a network, (first WiFi begin, connect mqtt, and then WiFi begin), repeat the problem.

hzhh110, 2023-11-10:
If the WiFi has a network, but the network cable does not have a network, (first the network cable begins, connect mqtt, and then WiFi begins), the problem is repeated.
If the network cable has a network, but the WiFi does not have a network, (first WiFi begin, connect mqtt, and then WiFi begin), repeat the problem.

hzhh110, 2023-11-10:
I tried two versions platform = espressif32 @ ~5.2.0 & platform = espressif32 @ ~6.4.0

[env:nodemcu-32s]
platform = espressif32 @ ~6.4.0
board = nodemcu-32s
framework = arduino
lib_deps = 
	h2zero/NimBLE-Arduino@^1.4.1
	arduino-libraries/NTPClient@^3.1.0
	crankyoldgit/IRremoteESP8266 @^2.8.6
	rweather/Crypto@^0.2.0
	knolleary/PubSubClient@^2.8
	bblanchon/ArduinoJson@6.19.4
	mathertel/OneButton @ ^2.0.3
build_flags = 
	-D MQTT_MAX_PACKET_SIZE=1024
	-D MQTT_KEEPALIVE=60
board_build.partitions = esp32cam2.csv

me-no-dev, 2023-11-10:
ESP32's network stack uses common DNS servers for all client interfaces (STA, ETH and PPP). This API will be changed for 3.0.0 to reflect that fact.

hzhh110, 2023-11-11:
Looking forward to a faster solution. Can platformio be adapted as soon as possible when solving the problem? I am now using the bootloader corresponding to SDK4. I don't know if there will be a problem of returning to the version after the successful upgrade. I also need to check this, because SDK4 was upgraded to the firmware of SDK3 before.I am afraid that if it is SDK5, will this problem also occur? The upgrade will be successful, and then the firmware will roll back to the previous version of SDK4 after power failure. I have sent out a batch of goods and need to upgrade.

Jason2866, 2024-05-22:
@hzhh110 Platformio is not a product from espressif. So espressif people can't say when or even if core 3.0.0 will be supported with Platformio.

VojtechBartoska, 2024-05-23:
Hello, we have discussed this issue internally and there are no actions points and the behaviour is by design, closing this ticket. Thanks for understanding

closed 2024-05-23 by VojtechBartoska