espressif/arduino-esp32 #8869: DAC or ADC in example `libraries/I2S/examples/ADCPlotter/ADCPlotter.ino`?

beta-tester, 2023-11-10:
### Board

n/a

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

n/a

### Operating System

n/a

### Flash frequency

n/a

### PSRAM enabled

yes

### Upload speed

n/a

### Description

in line 4 of the example libraries/I2S/examples/ADCPlotter/ADCPlotter.ino:
https://github.com/espressif/arduino-esp32/blob/3.0.0-alpha2/libraries/I2S/examples/ADCPlotter/ADCPlotter.ino#L4

is it `Digital to Analog Converter (DAC)` or should it be `Analog to Digital Converter (ADC)` ?

### Sketch

```cpp
n/a
```

### Debug Message

```plain
n/a
```

### Other Steps to Reproduce

n/a

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-11-11:
it's an old library that no longer works and is also removed. Do you need any specific feature?

beta-tester, 2023-11-11:
no, i just was looking for an DAC example (that uses DAC with DMA similar to the esp-idf example [signal_generator](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/dac/dac_continuous/signal_generator)).
the search for DAC found the example ADCPlotter and i stumbled over that there was only ADC code but the description in the top of the file was talking about DAC.

i could not find any DAC example in the arduino-esp32 repository similar to the `signal_generator`

you can close this issue then, when you will drop ADCPlotter example.

me-no-dev, 2023-11-11:
Things have changed, but are still available in another form to do the same thing. @P-R-O-C-H-Y PTAL