espressif/arduino-esp32 #8870: ESP32-S3 SD NAND openNextFile returning no directories at top level

frankcohen, 2023-11-11:
### Board

Reflections board, ESP32-S3 Dev Module compatible

### Device Description

Reflections is my open-source ESP32-S3-Mini-1 based board for building mobile entertainment experiences. It has a NAND (SMT form of an SD) onboard. 
[image]
The NAND comes pre-formatted with Fat16.

### Hardware Configuration

SPI bus:
#define SPI_MOSI      35
#define SPI_MISO      37
#define SPI_SCK       36

Nand on the SPI bus:
#define NAND_SPI_CS   15

TFT display on an SPI bus:

#define Display_SPI_CS    12
#define Display_SPI_DC    5
#define Display_SPI_RST   0
#define Display_SPI_BK    6

GPS module:
#define RXPin         18
#define TXPin         17
#define GPSPower      21

// Audio
#define I2S_bclkPinP       9
#define I2S_wclkPinP      10
#define I2S_doutPinP       8
#define AudioPower         7

// I2C bus
#define I2CSDA        3
#define I2CSCL        4

### Version

v2.0.14

### IDE Name

Arduino IDE 2.2.1

### Operating System

MacOS 13.3.1 (Ventura)

### Flash frequency

80 Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

Using File file = root.openNextFile(); to iteratue through files and directories stored on the NAND/SD Fat 16 device. The top level, openNextFile does not return any directorie.

[... 302 lines cut ...]
    Serial.println("ListDir:");
    listDir(SD, "/", 0);
    Serial.println("ListDir done");
*/

}

void loop(){

}
```

### Debug Message

```plain
None, see Serial Monitor output
```

### Other Steps to Reproduce

None. However the work around appears to be putting the needed files and directory structure into a subdirectory of the root. openNextFile returns the file and directories inside the root ddirectory.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

frankcohen, 2023-11-11:
The entire repository, including Gerber, schematic, example code is at https://github.com/frankcohen/ReflectionsOS

me-no-dev, 2023-11-18:
@P-R-O-C-H-Y PTAL

closed 2025-01-15 by Parsaabasi