espressif/arduino-esp32 #8872: ESP32-C6 blinkRGB not working on Windows (3.0.0-alpha2)

dvelaren, 2023-11-12:
### Board

ESP32-C6

### Device Description

ESP32-C6-DevKitC-1

### Hardware Configuration

No

### Version

other

### IDE Name

Arduino IDE

### Operating System

Windows 11

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

BlinkRGB example does not work with latest esp32 core version. It gives the following error when compiled:
```
E (89) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
```

### Sketch

```cpp
/*
  BlinkRGB

  Demonstrates usage of onboard RGB LED on some ESP dev boards.

  Calling digitalWrite(RGB_BUILTIN, HIGH) will use hidden RGB driver.
    
  RGBLedWrite demonstrates controll of each channel:
  void neopixelWrite(uint8_t pin, uint8_t red_val, uint8_t green_val, uint8_t blue_val)

  WARNING: After using digitalWrite to drive RGB LED it will be impossible to drive the same pin
    with normal HIGH/LOW level
*/
//#define RGB_BRIGHTNESS 64 // Change white brightness (max 255)

// the setup function runs once when you press reset or power the board

void setup() {
  // No need to initialize the RGB LED
}

// the loop function runs over and over again forever
void loop() {
#ifdef RGB_BUILTIN
  digitalWrite(RGB_BUILTIN, HIGH);   // Turn the RGB LED white
  delay(1000);
  digitalWrite(RGB_BUILTIN, LOW);    // Turn the RGB LED off
  delay(1000);

  neopixelWrite(RGB_BUILTIN,RGB_BRIGHTNESS,0,0); // Red
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,RGB_BRIGHTNESS,0); // Green
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,RGB_BRIGHTNESS); // Blue
  delay(1000);
  neopixelWrite(RGB_BUILTIN,0,0,0); // Off / black
  delay(1000);
#endif
}
```

### Debug Message

```plain
Arduino FQBN      : esp32:esp32:esp32c6:UploadSpeed=921600,CDCOnBoot=default,CPUFreq=160,FlashFreq=80,FlashMode=qio,FlashSize=4M,PartitionScheme=default,DebugLevel=debug,EraseFlash=none,JTAGAdapter=default
============ Before Setup End ============
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   486252 B ( 474.9 KB)
  Free Bytes        :   457128 B ( 446.4 KB)
  Allocated Bytes   :    22764 B (  22.2 KB)
  Minimum Free Bytes:   451920 B ( 441.3 KB)
  Largest Free Block:   425972 B ( 416.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
============ After Setup End =============
```

### Other Steps to Reproduce

The example works if I use Ubuntu WSL with the same core version. Also if I set Core Debug Level to Debug, it works, but if I put it in None, it gives the previous rmt mentioned error.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

P-R-O-C-H-Y, 2023-11-13:
Hi @dvelaren, this have been already fixed in master branch by PRs #8845 and #8813. So the fixes will be included in next release :)

closed 2023-11-13 by P-R-O-C-H-Y

gauchl, 2023-11-13:
For your info, I have exactly the same error with **esp32-c3**, with  Tools-> Core Debug Level: "None".
On the terminal port I get : "rmt: ESP-ROM:esp32c3-```
api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x458
load:0x403cc710,len:0x814
load:0x403ce710,len:0x2878
entry 0x403cc710
E (84) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
E (1084) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
E (2084) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
E (3084) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
E (4084) rmt: rmt_new_tx_channel(210): invalid interrupt priority:-1515870811
```

**If I compile with  Tools-> Core Debug Level: "Debug", the RGB led works OK.**
On the terminal I get:
               ```
app0 : addr: 0x00010000, size:  1280.0 KB, type:  APP, subtype: OTA_0
               app1 : addr: 0x00150000, size:  1280.0 KB, type:  APP, subtype: OTA_1
             spiffs : addr: 0x00290000, size:  1408.0 KB, type: DATA, subtype: SPIFFS
           coredump : addr: 0x003F0000, size:    64.0 KB, type: DATA, subtype: COREDUMP
------------------------------------------
Software Info:
------------------------------------------
  Compile Date/Time : Nov 13 2023 08:33:33
  Compile Host OS   : windows
  ESP-IDF Version   : v5.1.1-577-g6b1f40b9bf-dirty
  Arduino Version   : 3.0.0
------------------------------------------
Board Info:
------------------------------------------
  Arduino Board     : ESP32C3_DEV
  Arduino Variant   : esp32c3
  Arduino FQBN      : esp32:esp32:esp32c3:JTAGAdapter=default,CDCOnBoot=default,PartitionScheme=default,CPUFreq=160,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=115200,DebugLevel=debug,EraseFlash=none
============ Before Setup End ============
=========== After Setup Start ============
INTERNAL Memory Info:
------------------------------------------
  Total Size        :   340904 B ( 332.9 KB)
  Free Bytes        :   314772 B ( 307.4 KB)
  Allocated Bytes   :    22488 B (  22.0 KB)
  Minimum Free Bytes:   309564 B ( 302.3 KB)
  Largest Free Block:   286708 B ( 280.0 KB)
------------------------------------------
GPIO Info:
------------------------------------------
============ After Setup End =============
```

P-R-O-C-H-Y, 2023-11-13:
You can try to use manual installation and use master branch to have the fixes included :)

Mark-Whitaker-a, 2023-11-26:
I received the same error, this is how I fixed it:
roll back to the esp32 by Espressif version 2.0.14 in the Board Manager.