espressif/arduino-esp32 #8875: Multiple Libraries no longer function with latest update

DansDesigns, 2023-11-12:
### Board

ESP32-WROOM-32 + ESP32-WROOM-32U + ESP32-WROOM-32E

### Device Description

n/a

### Hardware Configuration

n/a

### Version

latest master (checkout manually)

### IDE Name

Arduino IDE 1.8.19

### Operating System

Windows 10 + Linux Zorin-Core 16.2

### Flash frequency

80

### PSRAM enabled

yes

### Upload speed

921600

### Description

updated to latest versions of: Arduino-MAX17055_driver, Adafruit_HMC5883_U and XT_DAC_Audio by xtronical
all fail to build the included examples and my project code no longer compiles with the same errors, links are to GitHub issues on each repository:

[HMC5883](https://github.com/adafruit/Adafruit_HMC5883_Unified/issues/15)

[MAX-17055](https://github.com/AwotG/Arduino-MAX17055_Driver/issues/6)

[XT_DAC_Audio Error](https://github.com/DansDesigns/Ultra-Slim-Tricorder/blob/main/XT_DAC_Audio_Build_error.txt)
(had to post as txt on my GitHub as message was too long..)

### Sketch

```cpp
Included Examples in each library..
For XT_DAC_Audio, use example: playwav or wavspeed
```

### Debug Message

```plain
n/a
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-12:
How about if we just start making this a reference list and cross them off when they are fixed?  Can you change the title to "List of Libraries not working in 3.0.0"

atanisoft, 2023-11-14:
The log from XT_DAC_Audio appears to be from 2.0.13 and not 3.0.0. Additionally using low level direct register access is not guaranteed to work across versions of the underlying ESP-IDF or even across the supported esp32 variants.

lbernstone, 2023-11-14:
For 3rd party libraries, please open an issue on their site and link it here if possible (as @DansDesigns did in OP).  Some code obviously won't/can't be pulled forward, so this ends up being a cemetery of dead projects too.

aaroneiche, 2023-11-15:
I ran into an error using Adafruit's Neopixel library on the ESP32-C3. It occurs with the examples.
I opened an issue there: https://github.com/adafruit/Adafruit_NeoPixel/issues/373

VojtechBartoska, 2023-12-08:
**For reference linking other related issues/PRs:**
- [IRremoteESP8266](https://github.com/crankyoldgit/IRremoteESP8266/issues/2039)
- [ESPAsyncWebServer](https://github.com/me-no-dev/ESPAsyncWebServer/pull/1351)
- [FastLed](https://github.com/FastLED/FastLED/issues/1555)

Parsaabasi, 2025-01-15:
Hello, 

Due to the overwhelming volume of issues currently being addressed, we have decided to close the previously received tickets. If you still require assistance or if the issue persists, please don't hesitate to reopen the ticket.

Thanks.

closed 2025-01-15 by Parsaabasi

aaroneiche, 2025-01-15:
Respectfully, this is a bad practice. Issues should be closed when they're fixed or addressed. This is sweeping problems under the rug at best, and misrepresenting the state of the software at worst. I strongly recommend for the health of the community that existing issues are left open.

wareya, 2025-01-15:
Additionally, issue-openers cannot reopen their own issues like suggested.

me-no-dev, 2025-01-15:
There are a lot of stale issues that are either already fixed or not valid anymore. Keeping them all open just makes them overwhelming to handle. A new issue can be opened to let us know that the given issue still exists and help us move it up the queue. We are a small team, so having a huge backlog is a big burden and does not help to prioritize what is actually still a problem.

me-no-dev, 2025-01-15:
This is not something that we will do often, or hopefully ever again, but in order to help the community better, we need to do something to filter out what is still valid