espressif/arduino-esp32 #8877: when WiFi.setSleep(false) won't reconnect to WiFi AP

kakopappa, 2023-11-13:
### Board

ESP32 Dev Module

### Device Description

There is no hardware connected.

### Hardware Configuration

not necessary

### Version

latest development Release Candidate (RC-X)

### IDE Name

Arduino IDE

### Operating System

Windows 10

### Flash frequency

80Mhz

### PSRAM enabled

no

### Upload speed

11520

### Description

I have noticed that when I set `WiFi.setSleep(false);` ESP32 does not auto-reconnect to AP. Wifi goes from WL_CONNECTED -> WL_IDLE_STATUS 

I set `WiFi.setSleep(true);` Wifi goes from WL_CONNECTED -> WL_CONNECTION_LOST ->WL_NO_SSID_AVAIL->WL_CONNECTED

not sure the power saving mode affects the reconnection. 

### Sketch

```cpp
#define BAUD_RATE         115200

#define WIFI_SSID         "xx_IPhone"    
#define WIFI_PASS         "xxxx"

#include <WiFi.h>
  
// setup function for WiFi connection
void setupWiFi() {
  Serial.printf("\r\n[Wifi]: Connecting");

  WiFi.setSleep(false);
  WiFi.setAutoReconnect(true);
 
  WiFi.begin(WIFI_SSID, WIFI_PASS); 

  while (WiFi.status() != WL_CONNECTED) {
    Serial.printf(".");
    delay(250);
  }
  Serial.printf("connected!\r\n[WiFi]: IP-Address is %s\r\n", WiFi.localIP().toString().c_str());
}
 

// main setup function
void setup() {
  //esp_log_level_set("*", ESP_LOG_VERBOSE); 
  Serial.begin(BAUD_RATE); Serial.printf("\r\n\r\n");
  setupWiFi(); 
}
 
const char* wl_status_to_string(wl_status_t status) {
  switch (status) {
    case WL_NO_SHIELD: return "WL_NO_SHIELD";
    case WL_IDLE_STATUS: return "WL_IDLE_STATUS";
    case WL_NO_SSID_AVAIL: return "WL_NO_SSID_AVAIL";
    case WL_SCAN_COMPLETED: return "WL_SCAN_COMPLETED";
    case WL_CONNECTED: return "WL_CONNECTED";
    case WL_CONNECT_FAILED: return "WL_CONNECT_FAILED";
    case WL_CONNECTION_LOST: return "WL_CONNECTION_LOST";
    case WL_DISCONNECTED: return "WL_DISCONNECTED";
  }

  return "NO";
}

String lastStatus = "";

void loop() {   
    String currentStatus =  wl_status_to_string(WiFi.status());

    if(currentStatus != lastStatus) {
      Serial.print("currentStatus:");
      Serial.println(currentStatus);

      Serial.print("lastStatus:");
      Serial.println(lastStatus);

      lastStatus = currentStatus;
    }
}
```

### Debug Message

```plain
When WiFi.setSleep(true);

14:55:32.320 -> [   488][V][WiFiGeneric.cpp:357] _arduino_event_cb(): STA Connected: SSID: Aruna_IPhone, BSSID: 32:6b:ae:58:2e:93, Channel: 11, Auth: WPA2_PSK
14:55:32.320 -> [   490][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
14:55:32.491 -> ..[  1006][V][WiFiGeneric.cpp:371] _arduino_event_cb(): STA Got New IP:172.20.10.8
14:55:32.848 -> [  1007][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
14:55:32.848 -> [  1010][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 172.20.10.8, MASK: 255.255.255.240, GW: 172.20.10.1
14:55:32.971 -> connected!
14:55:32.971 -> [WiFi]: IP-Address is 172.20.10.8
14:55:33.003 -> =========== After Setup Start ============
14:55:33.003 -> INTERNAL Memory Info:
14:55:33.003 -> ------------------------------------------
14:55:33.003 ->   Total Size        :   347872 B ( 339.7 KB)
14:55:33.003 ->   Free Bytes        :   263124 B ( 257.0 KB)
14:55:33.003 ->   Allocated Bytes   :    74804 B (  73.1 KB)
14:55:33.003 ->   Minimum Free Bytes:   259620 B ( 253.5 KB)
14:55:33.003 ->   Largest Free Block:   110580 B ( 108.0 KB)
14:55:33.003 -> ------------------------------------------
14:55:33.046 -> GPIO Info:
14:55:33.046 -> ------------------------------------------
14:55:33.046 ->                   1 : UART_TX
14:55:33.046 ->                   3 : UART_RX
14:55:33.046 -> ============ After Setup End =============
14:55:33.046 -> currentStatus:WL_CONNECTED
14:55:33.046 -> lastStatus:
14:55:47.052 -> [ 15241][V][WiFiGeneric.cpp:364] _arduino_event_cb(): STA Disconnected: SSID: Aruna_IPhone, BSSID: 32:6b:ae:58:2e:93, Reason: 200
14:55:47.053 -> [ 15242][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
14:55:47.099 -> [ 15249][W][WiFiGeneric.cpp:1064] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
14:55:47.099 -> [ 15256][D][WiFiGeneric.cpp:1084] _eventCallback(): WiFi Reconnect Running
14:55:47.099 -> currentStatus:[ 15265][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
14:55:47.099 -> WL_CONNECTION_LOST
14:55:47.099 -> lastStatus:WL_CONNECTED
14:55:49.518 -> [ 17691][V][WiFiGeneric.cpp:364] _arduino_event_cb(): STA Disconnected: SSID: Aruna_IPhone, BSSID: 00:00:00:00:00:00, Reason: 201
14:55:49.518 -> [ 17692][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
14:55:49.518 -> [ 17699][W][WiFiGeneric.cpp:1064] _eventCallback(): Reason: 201 - NO_AP_FOUND
14:55:49.518 -> [ 17706][D][WiFiGeneric.cpp:1088] _eventCallback(): WiFi AutoReconnect Running
14:55:49.563 -> currentStatus:[ 17715][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
14:55:49.563 -> WL_NO_SSID_AVAIL
14:55:49.563 -> lastStatus:WL_CONNECTION_LOST
14:55:49.642 -> [ 17805][V][WiFiGeneric.cpp:357] _arduino_event_cb(): STA Connected: SSID: Aruna_IPhone, BSSID: d2:71:55:2c:21:34, Channel: 11, Auth: WPA2_PSK
14:55:49.642 -> [ 17806][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 4 - STA_CONNECTED
14:55:49.642 -> currentStatus:WL_IDLE_STATUS
14:55:49.642 -> lastStatus:WL_NO_SSID_AVAIL
14:55:50.170 -> [ 18335][V][WiFiGeneric.cpp:371] _arduino_event_cb(): STA Got Same IP:172.20.10.8
14:55:50.170 -> [ 18336][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 7 - STA_GOT_IP
14:55:50.170 -> [ 18339][D][WiFiGeneric.cpp:1102] _eventCallback(): STA IP: 172.20.10.8, MASK: 255.255.255.240, GW: 172.20.10.1
14:55:50.170 -> currentStatus:WL_CONNECTED
14:55:50.170 -> lastStatus:WL_IDLE_STATUS

***** Successfully reconnect

When WiFi.setSleep(false);

14:58:42.479 -> [WiFi]: IP-Address is 172.20.10.8
14:58:42.479 -> =========== After Setup Start ============
14:58:42.479 -> INTERNAL Memory Info:
14:58:42.479 -> ------------------------------------------
14:58:42.479 ->   Total Size        :   347872 B ( 339.7 KB)
14:58:42.479 ->   Free Bytes        :   263112 B ( 256.9 KB)
14:58:42.479 ->   Allocated Bytes   :    74816 B (  73.1 KB)
14:58:42.479 ->   Minimum Free Bytes:   259632 B ( 253.5 KB)
14:58:42.479 ->   Largest Free Block:   110580 B ( 108.0 KB)
14:58:42.524 -> ------------------------------------------
14:58:42.524 -> GPIO Info:
14:58:42.524 -> ------------------------------------------
14:58:42.524 ->                   1 : UART_TX
14:58:42.524 ->                   3 : UART_RX
14:58:42.524 -> ============ After Setup End =============
14:58:42.524 -> currentStatus:WL_CONNECTED
14:58:42.524 -> lastStatus:
14:58:45.208 -> [  4146][V][WiFiGeneric.cpp:364] _arduino_event_cb(): STA Disconnected: SSID: Aruna_IPhone, BSSID: 5a:32:f8:9d:65:7b, Reason: 2
14:58:45.208 -> [  4147][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 5 - STA_DISCONNECTED
14:58:45.208 -> [  4154][W][WiFiGeneric.cpp:1064] _eventCallback(): Reason: 2 - AUTH_EXPIRE
14:58:45.208 -> [  4161][D][WiFiGeneric.cpp:1084] _eventCallback(): WiFi Reconnect Running
14:58:45.208 -> [  4170][V][WiFiGeneric.cpp:96] set_esp_interface_ip(): Configuring Station static IP: 0.0.0.0, MASK: 0.0.0.0, GW: 0.0.0.0
15:00:45.214 -> [124148][V][WiFiGeneric.cpp:376] _arduino_event_cb(): STA IP Lost
15:00:45.214 -> [124148][D][WiFiGeneric.cpp:1042] _eventCallback(): Arduino Event: 9 - STA_LOST_IP
15:00:45.214 -> currentStatus:WL_IDLE_STATUS
15:00:45.214 -> lastStatus:WL_CONNECTED

***** does not reconnect
```

### Other Steps to Reproduce

tested on Core SDK 2.0.14 .. same 

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

kakopappa, 2023-11-13:
line [L963 ](https://github.com/espressif/arduino-esp32/blob/7c79cdd6e3f12c2ce61275b062ca73c27de37f9b/libraries/WiFi/src/WiFiGeneric.cpp#L963
) seems intentionally blank. Setting

`WiFiSTAClass::_setStatus(WL_CONNECTION_LOST);`

seems to solve the problem though.

me-no-dev, 2023-11-13:
Yes... that is indeed strange. TBH we have had many issues about disconnect reason and how it's handled. This code have been revised many many times. Let's leave this as-is and allow the community to chime in.

TD-er, 2023-11-22:
> line [L963 ](https://github.com/espressif/arduino-esp32/blob/7c79cdd6e3f12c2ce61275b062ca73c27de37f9b/libraries/WiFi/src/WiFiGeneric.cpp#L963) seems intentionally blank. Setting
> 
> `WiFiSTAClass::_setStatus(WL_CONNECTION_LOST);`
> 
> seems to solve the problem though.

Thank you!!!!!!!!!!
This has been bugging me for ages.
This does cause all kinds of strange issues when this happens.

I will test it here on a few boards, but I'm quite sure this will fix a lot of very strange reconnect issues since some APs give this disconnect issue also when they try to suggest you to connect to another AP.
e.g. mesh-enabled APs which find your RSSI values to be a bit too low or when they are already quite busy with connected clients.
I know at least Fritzbox does this every now and then. AVM assured me their APs should be able to handle > 32 clients but I never managed to do so.

Could be this is also given as disconnect reason when an AP switches channel, so it might also happen with less crowded networks.

Anyway, I've seen this reason far too often when looking into WiFi issues that I'm very positive your suggested fix will help here.

closed 2025-01-15 by Parsaabasi