espressif/arduino-esp32 #8883: ESP32 LAN8720

hzhh110, 2023-11-14:
### Board

ESP32-WROOM-32E

### Device Description

ESP32-WROOM-32E

### Hardware Configuration

 isHaveEthernet = ETH.begin(ETH_ADDR, ETH_POWER_PIN, ETH_MDC_PIN, ETH_MDIO_PIN, ETH_TYPE, ETH_CLOCK_GPIO17_OUT);

### Version

v2.0.11

### IDE Name

Platfrom

### Operating System

macOS

### Flash frequency

40Mhz

### PSRAM enabled

yes

### Upload speed

115200

### Description

[The device has not been able to use the network cable normally, and other devices are normal. What is the reason for this generally](url)

https://github.com/espressif/arduino-esp32/assets/19906209/9d3f7a6c-15b5-4777-8988-d43a91dff110

[](url)

### Sketch

```cpp
The device has not been able to use the network cable normally, and other devices are normal. What is the reason for this generally
```

### Debug Message

```plain
The device has not been able to use the network cable normally, and other devices are normal. What is the reason for this generally
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

hzhh110, 2023-11-14:
There is also a yellow light is not on, the green light is on, changed a lot of time, LAN8720 still has a lot of resistors and capacitors, connected to the network cable or so.

me-no-dev, 2023-11-14:
LAN8720... what board is this? You custom one? How abour you post some serial output with debug on. It would probably tell you what is going on. Looks like your LAN module is constantly restarting or something. Can't tell more based on the little info you gave.

hzhh110, 2023-11-14:
[JH-IR12-A V2 1.pdf](https://github.com/espressif/arduino-esp32/files/13346885/JH-IR12-A.V2.1.pdf)
Most of this circuit is normally used, that is, the network cable problem feedback is more

sauttefk, 2023-11-14:
The assembly plan isn't very helpful to analyze your problem.

mrengineer7777, 2023-11-16:
@hzhh110 The LAN8720 module can be hard to work with.  https://github.com/espressif/arduino-esp32/discussions/7701

I ended up designing a custom PCB that uses a Microchip LAN chip.  Specifically KSZ8091RNBIA.  Works great!

TD-er, 2023-11-22:
Are you using the correct PHY address?
There are some pins on the LAN8720 chip which boot state determines the internal configuration of the chip among which is the address.
Thus this PHY address may differ per board.

- Common Waveshare LAN8720 PHY breakout board (and clones) use PHY address 1.
- Olimex ESP32 EVB REV B IoT LAN8710 PHY Board with CAN use PHY address 0.
- Other LAN8720 breakouts often use PHY address 0.

Also if the voltage supplied to the LAN8720 is fluctuating too much, the LAN adapter will toggle between 10 and 100 Mbps.
The LAN chip does require quite a bit of current when active, so it can also be a bad connection, thin wiring, etc.

closed 2025-01-15 by Parsaabasi