espressif/arduino-esp32 #8887: please make ESP32 able to read from USB or BT HID keyboard (standard C/C++ libs, e.g. scanf(), gets(), std::cin )

dsyleixa, 2023-11-15:
### Related area

USB, BT

### Hardware specification

all ESP32 types

### Is your feature request related to a problem?

keyboard-input e.g. for passwords or Serial communication without  Serial connection to a PC

### Describe the solution you'd like

please make ESP32 able to read user input from USB or BT HID keyboard (standard C/C++ libs, e.g. scanf(), gets(), std::cin ).
if BT then first pairing function like usual for BT,
if USB then via USB-host  at least for szandard UBB-HID-keyboard and possibly also for USB wireless keyboards (e.g., Logitech, Rii) via a USB plug with w/l transmitter.
After plugin/pairing the keyboard-inout detection should work via plug+play and out of the box with C/C++ stdio.h or std:cin libs  like commonly known for Windows/Linux-PCs or SoCs (e.g., RaspberryPi).

### Describe alternatives you've considered

N/A

### Additional context

N/A

### I have checked existing list of Feature requests and the Contribution Guide

- [X] I confirm I have checked existing list of Feature requests and Contribution Guide.

me-no-dev, 2023-11-15:
1. This requires HID HOST, which is not implemented in Arduino
2. We provide Arduino APIs here. For such "standard" APIs, you need to look into ESP-IDF or Crosstool-NG
3. What you ask for is more application level than Arduino core.

dsyleixa, 2023-11-15:
as to
This requires HID HOST, which is not implemented in Arduino:
then the request is about to implement it to Arduino, IDF is no option (finally this was posted here to the  arduino-esp32  repo).

closed 2025-01-15 by Parsaabasi