espressif/arduino-esp32 #8898: Object too Large for 'dram0_0_seg'

gfvalvo, 2023-11-17:
### Board

Adafruit ESP32 Feather Huzzah

### Device Description

Adafruit ESP32 Feather Huzzah

### Hardware Configuration

No

### Version

v2.0.11

### IDE Name

Arduino 1.8.19

### Operating System

Windows 10

### Flash frequency

80 MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

I’m in the process of porting a code that contains a vary large class to ESP32. In fact, the sizeof() compiler directive reports that its size is 132220 bytes.

When building a minimal ESP32 program with empty setup() and loop() instructions, the Arduino IDE reports:
“Global variables use 20968 bytes (6%) of dynamic memory, leaving 306712 bytes for local variables. Maximum is 327680 bytes.”

So, it seemed reasonable that I could create one object of the class in question. It turns out I can do it in PSRAM, but not in the regular ESP32 RAM.

If I try to do it as a global variable, the IDE reports that the object won't fit in `dram0_0_seg'.

And, if I try to do it dynamically, I get a null pointer returned by the malloc() function or an abort if I try the C++ ‘new’ operator.

I know very little about memory partitioning or the various memory segments used by the link. However, the error message mentions `dram0_0_seg' and `.dram0.bss'. That leads me to believe that the problem is trying to put the object into a memory segment that’s too small.

Is there any way to use a different “segment” or change the size of `dram0_0_seg'?

Thanks.

### Sketch

```cpp
volatile uint8_t bigObject[132220];

void setup() {
  bigObject[0] = 100;
}

void loop() {
}
```

### Debug Message

```plain
d:/arduino-1.8.19/portable/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: C:\Users\GFV\AppData\Local\Temp\arduino_build_279698/sketch_nov17b.ino.elf section `.dram0.bss' will not fit in region `dram0_0_seg'
d:/arduino-1.8.19/portable/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
d:/arduino-1.8.19/portable/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: DRAM segment data does not fit.
d:/arduino-1.8.19/portable/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_0_seg' overflowed by 28608 bytes
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board Adafruit ESP32 Feather.
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

lbernstone, 2023-11-17:
You'll need contiguous DRAM to make that pointer. There just isn't enough contiguous space in arduino-esp32 for that- only about 100K.  Newer variants have a better layout- the S3 will get you about 250K.
https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32/api-guides/memory-types.html

gfvalvo, 2023-11-18:
> You'll need contiguous DRAM to make that pointer.

Ahhh ... so it's a matter of contiguous available DRAM, not just the absolute number of bytes available. Thanks.

lbernstone, 2023-11-18:
The pointer is an address is memory.  The allocation is telling it how far from that address it extends.

VojtechBartoska, 2023-12-12:
Closing as answered, if needed you can reopen.

closed 2023-12-12 by VojtechBartoska