espressif/arduino-esp32 #8901: esp_deep_sleep_enable_gpio_wakeup for light sleep only

illusionmanager, 2023-11-20:
### Board

esp32 c3 Lolin

### Device Description

plain lolin c3 pico board

### Hardware Configuration

no

### Version

latest master (checkout manually)

### IDE Name

arduine ide

### Operating System

windows 10

### Flash frequency

40 Mhz ? (i don't know)

### PSRAM enabled

yes

### Upload speed

115200

### Description

I was trying the example code ExternalWakeUp.ino, but that doesn't compile on esp32c3. Then I discovered that we have to use esp_deep_sleep_enable_gpio_wakeup instead. But reading the code esp_Sleep.h that it  tells me that for this function:
* Each GPIO supports wakeup function, which can be triggered on either low level
 * or high level. Unlike EXT0 and EXT1 wakeup sources, this method can be used
 * both for all IOs: RTC IOs and digital IOs. It can only be used to wakeup from
 * light sleep though.
 it only works for light sleep. How do I get it to in deepsleep with external gpio?

### Sketch

```cpp
ExternalWakeUp.ino (standard example)
```

### Debug Message

```plain
none
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

illusionmanager, 2023-11-20:
i  guess it is working, so it is just an error in the documentation? At least it works with just one pin

lucasssvaz, 2023-11-20:
@illusionmanager What part of the documentation are you specifically referring to ?

illusionmanager, 2023-11-20:
and may I suggest to make changes to the example program so it doesn't use a function that isn't available for esp32c3

lbernstone, 2023-11-20:
They are counter-indicated for the c3.
https://github.com/espressif/arduino-esp32/tree/master/libraries/ESP32/examples/DeepSleep/ExternalWakeUp

lbernstone, 2023-11-20:
The documentation says it can also be used for RTC controlled gpio in deep sleep.
https://docs.espressif.com/projects/esp-idf/en/v5.1.1/esp32c3/api-reference/system/sleep_modes.html#gpio-wakeup

illusionmanager, 2023-11-20:
that part of the documentation does indeed mention the deep sleep capabilities, so it comes down to the esp_sleep.h file where the comment "It can only be used to wakeup from light sleep though." should be removed.

illusionmanager, 2023-11-20:
can I suggest a change to the example program so it reads something like this (But I made up ESP32_C3, perhaps there is a better one)

#ifdef ESP32_C3
esp_deep_sleep_enable_gpio_wakeup((1ULL << GPIO_NUM_4), ESP_GPIO_WAKEUP_GPIO_HIGH);
#else
esp_sleep_enable_ext0_wakeup(GPIO_NUM_33,1); //1 = High, 0 = Low
#endif

closed 2023-11-20 by illusionmanager