espressif/arduino-esp32 #8902: ***ERROR*** A stack overflow in task sys_evt has been detected

createskyblue, 2023-11-20:
### Board

ESP32 PICO V3-02

### Device Description

On the self-made development board

### Hardware Configuration

[image]

### Version

latest master (checkout manually)

### IDE Name

PlatfornIO

### Operating System

Windows 10

### Flash frequency

80MHz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Using Painless_ Mesh 1.5.0 library for Mesh networking experiments, while using serial port 0 to send and receive instructions. It has been found that when running the same code with different configuration files, the child nodes and main nodes are physically far away, it is easy to crash. But if the nodes are relatively close, the crash described below will not occur for a long time. Currently, we have not further identified the specific location of the error

使用Painless_Mesh-1.5.0库进行Mesh组网实验，同时使用串口0收发指令。发现当运行相同代码，不同配置文件的子节点和主节点物理较远的时候，容易发生崩溃。但是如果节点们比较近，则长时间都不会发生下面描述的崩溃。暂时没有进一步定位到错误的具体位置

### Sketch

```cpp
Unable to provide
```

### Debug Message

```plain
***ERROR*** A stack overflow in task sys_evt has been detected.

***ERROR*** A stack overflow in task sys_evt has been detected.

Backtrace: 0x401142e6:0x3ffb4a40 0x40110efe:0x3ffb4a70 0x40110f64:0x3ffb4ab0 0x400836f6:0x3ffb4ad0 0x40083775:0x3ffb4af0 0x40113ce2:0x3ffb4b20 0x401139dc:0x3ffb4b50 0x401139fd:0x3ffb4b70 0x401135d2:0x3ffb4b90 0x4011397b:0x3ffb4c00 0x40083b09:0x3ffb4c50 0x40085da8:0x3ffb4c70 0x40083a65:0x3ffb4d30 0x4008d399:0x3ffb4d50 0x4009091d:0x3ffb4d70 0x4008f0dd:0x3ffb4df0 0x4008d4a8:
0x3ffb4e20 0x4008d458:0x3f40b683 |<-CORRUPTED

ELF file SHA256: 15683556465e6a0d

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
lo
ad:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

setLogLevel: ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE |
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (sys_evt) 
Core  0 register dump:
PC      : 0x4009072f  PS      : 0x00060036  A0      : 0x8008efec  A1      : 0x3ffb4a40  
A2      : 0x3ffbf5d8  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060023  
A6      : 0x00060023  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff  
A10     : 0x3ffbf5c4  A11     : 0x00000001  A12     : 0x00060021  A13     : 0x80000000  
A14     : 0x007bf5d8  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0x00000000  LBEG    : 0x4008b441  LEND    : 0x4008b451  LCOUNT  : 0xffffffff  

Backtrace: 0x4009072c:0x3ffb4a40 0x4008efe9:0x3ffb4a80 0x40
08d4a8:0x3ffb4ab0 0x4008d458:0xa5a5a5a5 |<-CORRUPTED

ELF file SHA256: 15683556465e6a0d

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
lo
ad:0x40078000,len:13964
load:0x40080400,len:3600
entry 0x400805f0

setLogLevel: ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE |

Decoder:
START:
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/vfs/vfs.c:88
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:296
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/startup.c:328
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/arch/xtensa/debug_helpers.c:59
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/arch/xtensa/debug_helpers.c:76
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/uart_ll.h:872
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/uart_ll.h:136
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/uart_ll.h:493
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/brownout.c:86
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/hal/esp32/include/hal/uart_ll.h:120
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/panic_handler.c:219
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/xtensa_vectors.S:703
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/panic.c:408
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/esp_system.c:137
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:407
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3505
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
/Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:231
END
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-11-20:
Seems not an Arduino project. This github is for issues with Arduino

me-no-dev, 2023-11-20:
you used an external library and provided no code that can help to reproduce. Can hardly help here...

createskyblue, 2023-11-20:
> you used an external library and provided no code that can help to reproduce. Can hardly help here...
I will organize the code later and separate the minimum reproduction code from the business. Because some of the current code does not belong to me personally 😢

me-no-dev, 2023-11-20:
your memory get's corrupted. We do not need your whole project, just the camera initialization code and configuration, so we could check if it could be anything on our end. Just the info above is way not enough to come to any conclusion.

createskyblue, 2023-11-20:
> your memory get's corrupted. We do not need your whole project, just the camera initialization code and configuration, so we could check if it could be anything on our end. Just the info above is way not enough to come to any conclusion.

There is no camera used in the project, only the wifi STA+AP function is used, and in the pianless mesh, there should only be a TCP connection opened. I also noticed that the automatic reset after the crash trigger will trigger a new crash during the initial mesh (the crash report above occurs in chronological order).

 I will organize the relevant code as soon as possible.
(About 12 hours later?)

createskyblue, 2023-12-15:
```
setLogLevel: ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE |
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (sys_evt)
Core  0 register dump:
PC      : 0x40092903  PS      : 0x00060536  A0      : 0x800911c0  A1      : 0x3ffb4e50
A2      : 0x3ffbf5f8  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060523
A6      : 0x00060523  A7      : 0x0000cdcd  A8      : 0x0000abab  A9      : 0xffffffff
A10     : 0x3ffbdc6c  A11     : 0x3ffc5f00  A12     : 0x3ffc1a0c  A13     : 0x3ffc1b34
A14     : 0x007bf5f8  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4008b994  LEND    : 0x4008b9aa  LCOUNT  : 0xffffffff

Backtrace: 0x40092900:0x3ffb4e50 0x400911bd:0x3ffb4e90 0x4008f67c:0x3ffb4ec0 0x4008f62c:0xa5a5a5a5 |<-CORRUPTED

  #0  0x40092900:0x3ffb4e50 in compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:301
  #1  0x400911bd:0x3ffb4e90 in vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
      (inlined by) vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3458
  #2  0x4008f67c:0x3ffb4ec0 in _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
  #3  0x4008f62c:0xa5a5a5a5 in _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:231
  
  
  
  setLogLevel: ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE |
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: Stack canary watchpoint triggered (sys_evt)
Core  0 register dump:
PC      : 0x40092903  PS      : 0x00060936  A0      : 0x800911c0  A1      : 0x3ffb4970
A2      : 0x3ffbf5f8  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060923
A6      : 0x00060923  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x3ffbdc6c  A11     : 0x3ffc5f00  A12     : 0x3ffc1a0c  A13     : 0x3ffc1b34
A14     : 0x007bf5f8  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4008b994  LEND    : 0x4008b9aa  LCOUNT  : 0xffffffff

Backtrace: 0x40092900:0x3ffb4970 0x400911bd:0x3ffb49b0 0x4008f67c:0x3ffb49e0 0x4008f62c:0xa5a5a5a5 |<-CORRUPTED

  #0  0x40092900:0x3ffb4970 in compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:301
  #1  0x400911bd:0x3ffb49b0 in vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
      (inlined by) vTaskSwitchContext at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/tasks.c:3458
  #2  0x4008f67c:0x3ffb49e0 in _frxt_dispatch at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:436
  #3  0x4008f62c:0xa5a5a5a5 in _frxt_int_exit at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/portasm.S:231
  
  
  setLogLevel: ERROR | MESH_STATUS | CONNECTION | SYNC | COMMUNICATION | GENERAL | MSG_TYPES | REMOTE |
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception).
Debug exception reason: Stack canary watchpoint triggered (sys_evt)
Core  0 register dump:
PC      : 0x40092903  PS      : 0x00060336  A0      : 0x800911c0  A1      : 0x3ffb4970  
A2      : 0x3ffbf5f8  A3      : 0xb33fffff  A4      : 0x0000abab  A5      : 0x00060323  
A6      : 0x00060323  A7      : 0x0000cdcd  A8      : 0xb33fffff  A9      : 0xffffffff
A10     : 0x3ffbdc6c  A11     : 0x3ffc5f00  A12     : 0x3ffc1a0c  A13     : 0x3ffc1b34
A14     : 0x007bf5f8  A15     : 0x003fffff  SAR     : 0x00000004  EXCCAUSE: 0x00000001
EXCVADDR: 0x00000000  LBEG    : 0x4008b994  LEND    : 0x4008b9aa  LCOUNT  : 0xffffffff

Backtrace: 0x40092900:0x3ffb4970 0x400911bd:0x3ffb49b0 0x4008f67c:0x3ffb49e0 0x4008f62c:0xa5a5a5a5 |<-CORRUPTED

  #0  0x40092900:0x3ffb4970 in compare_and_set_native at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/compare_set.h:25
      (inlined by) spinlock_acquire at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_hw_support/include/soc/spinlock.h:103
      (inlined by) xPortEnterCriticalTimeout at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/port.c:301
  #1  0x400911bd:0x3ffb49b0 in vPortEnterCritical at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/freertos/port/xtensa/include/freertos/portmacro.h:578
[... cut ...]
The approximate cause of the crash is that I connected to the AP opened by Esp32. When the issue occurs, it must be temporarily resolved by power cycling. Otherwise, there is a probability of triggering this problem each time I connect to the AP, and it may not necessarily be the same triggering method mentioned earlier.
The same code runs well on ESP32-C3, which is confusing to me.
If I discover a more effective triggering method, I will continue to report it.

崩溃大致触发的原因是我连接了Esp32开启的AP，当问题触发后必须要通过断电复位才能暂时摆脱问题，否则每次连接AP的时候就有概率触发这个问题，不一定是前面提到的触发方式
相同的代码在ESP32-C3中运行良好，这让我感到很困惑
如果我发现更有效的触发方式，我将继续报告