espressif/arduino-esp32 #8904: ESP32C3 EEPROM problem with size >= 8192?

JimDrewGH, 2023-11-21:
### Board

ESP32C3 Dev Module

### Device Description

Custom board and also tried another C3 dev board (and multiple different skus of ESP32C3 modules).

### Hardware Configuration

N/A

### Version

v2.0.13

### IDE Name

IDE 2.2.1

### Operating System

Windows 10 Pro

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

When simply replacing the size to use for the EEPROM from 4096 to 8192, the EEPROM no longer holds the written value!

Changing EEPROM.begin(4096) to EEPROM.begin(8192) initializes the EEPROM, and all values can be read just fine, but for some reason any EEPROM.write(x,y) no longer actually stores any data permanently.  You can read back a change, and it shows correct - however, when you power cycle that change is gone.

I discovered this because I wanted to increase the size of some storage space I needed and so i just changed the size and discovered the issue.

### Sketch

```cpp
This works:

EEPROM.begin(4096);

This "works", but anything written using EEPROM.write does not permanently store the data.
```

### Debug Message

```plain
N/A
```

### Other Steps to Reproduce

Nothing magical, just simply changing the size value is enough to show the issue.

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-11-21:
EEPROM library on ESP32 uses the NVS as backend, which limits the size to a bit under 2KB. Do you need 4K? Can you split it? I suggest you have a look at the `Preferences` library. It's pretty nice ;)

lbernstone, 2023-11-21:
It's a blob, it is not limited on size.  But, it is using the nvs partition, so if there is not space there, begin will throw an error.  I'll bet it is returning a failure here.

JimDrewGH, 2023-11-21:
This is interesting.  I did notice something today that is rather puzzling.  I went back to 4096 and SOME of my devices with the exact same code does not work - and ALL of those particular devices also have the fuse burned to prevent downloading code via USB.  I am not sure if that is a coincidence or a fluke.  I actually need 8K of EEPROM space.  I could use SSPIF if absolutely necessary, but that would mean a complete re-write of quite a bit of code.  I ported this code from the ESP8266 and everything seemed fine until I needed more EEPROM space.  I will take a look at what "Preferences" is.

JimDrewGH, 2023-11-21:
The EEPROM.begin(8192) was not failing, which is why I was confused about what was going on.  I did find that if you select the ERASE ALL FLASH BEFORE UPLOAD SKETCH option before compiling/uploading the sketch then the EEPROM works again as expected (up to 4K it seems).  I am not thrilled about having to re-write code just for simple single byte read/writes to permanent storage, but I guess Preferences is likely going to be the requirement moving forwards.

I think there should be something mentioned SOMEWHERE about EEPROM library being limited in size, especially since it doesn't fail on the begin().

lbernstone, 2023-11-21:
I don't suppose you ever read [this](https://github.com/espressif/arduino-esp32/tree/master/libraries/EEPROM).
In any case, nvs is a shared space, so other code may be consuming some of that space.  It is COW, so you will always need twice the space to put any key pair in it.  Because EEPROM is a monolithic blob, it will always write the whole thing on every single change.  For a blob of 8192, that actually means three sectors will be written on every single change of the contents.  This is remarkable inefficient, and will use up your flash cells much more quickly than is healthy.
I get that it won't help with your existing devices, but the nvs partition can be [sized](https://docs.espressif.com/projects/arduino-esp32/en/latest/tutorials/partition_table.html#using-a-custom-partition-scheme) to whatever you would like on a fresh upload.

JimDrewGH, 2023-11-21:
No, I have never seen that info before.  I have had no need to go searching for a replacement because EEPROM has worked for years without any changes required.  I would expect at very least for it to return with a failure if there was going to be an issue!

I don't trust any type of non-volatile storage system, so my EEPROM data is loaded into RAM as an exact copy.  Updates to the actual EEPROM content only occur when there is a change to a particular entry (single byte) and this happens only during the product's configuration, so performance is never an issue as I am working out of RAM unless something actually needs to change.  There might be a few hundred total write accesses to the EEPROM during the lifecycle of the product, so wear is also not  a concern.  I just expected EEPROM to work as it always has in the past, and to return an error if something is not going to work - like everything does.

me-no-dev, 2023-11-22:
> I just expected EEPROM to work as it always has in the past

In our case, this is true. Maybe you just used EEPROM on other devices before?

Anyway... EEPROM is something that comes from AVR and other chips that have dedicated EEPROM areas. We do not have such thing, so we try to emulate it with what we have. NVS(Preferences) is the better option, because it auto-manages the flash and prevents killing a sector because of too many writes.

In your case, it seems that you will not be overwriting the flash many times, therefore it might be beneficial to separate a 16K partition for EEPROM and access that directly (not going through EEPROM or Preferences)

SuGlider, 2023-11-28:
Another suggestion for storing data in flash (EEPROM - which is just "emulating it in the flash") would be to use LittleFS instead of the EEPROOM library. You could create a file with pairs using JSON format (key, value). There are also some good open JSON libraries for Arduino (https://github.com/arduino-libraries/Arduino_JSON)

[Sketch example](https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/LITTLEFS_test.ino) - must use the correct [partition file](https://github.com/espressif/arduino-esp32/blob/master/libraries/LittleFS/examples/LITTLEFS_test/partitions.csv)

SuGlider, 2023-11-28:
Or just create a binary file with all the 8192 entries in `uint32` format (4 bytes)... It would be something like a 32KB file.
A function to read /write to/from an array and that should be all.

JimDrewGH, 2023-12-11:
I ended up converting everything over to use Preferences.  I just closed my eyes and pretended I am not wasting resources.  :)

I would highly recommend that there is SOME information SOMEWHERE that deprecates EEPROM, because that does not show up when the compiler runs under the Arduino IDE (either 1.x or 2.x) or PlatformIO.  I think this would at least alert someone that there is going to a problem instead of having to track down the issue like I did.

closed 2023-12-11 by JimDrewGH