espressif/arduino-esp32 #8906: Incorrect GPIO state during deep sleep

ViniP8, 2023-11-21:
### Board

ESP32 S2 

### Device Description

Lolin S2 Mini

### Hardware Configuration

No

### Version

v2.0.11

### IDE Name

Arduino IDE

### Operating System

Ubuntu 22.04

### Flash frequency

Default

### PSRAM enabled

yes

### Upload speed

115200

### Description

I'm trying to set the state of GPIO 13 as LOW, and GPIO 10 as HIGH and keep them during deep sleep, but when entering deep sleep the pins are floating (Pin 10 with ~2V, Pin 13 with ~1V)

### Sketch

```cpp
void setup() {
  Serial.begin(115200);
  pinMode(15, OUTPUT);
  digitalWrite(15, 1);  //Builtin led on
  delay(500);
  deep_sleep_gpio_init();
}

void loop() {
}

void deep_sleep_gpio_init(void) {
  changePin(13, LOW);
  changePin(10, HIGH);
  gpio_deep_sleep_hold_en();
  delay(500);
  esp_deep_sleep_start();
}

void changePin(int pin, bool value) {
  gpio_hold_dis((gpio_num_t)pin);
  pinMode(pin, OUTPUT);
  digitalWrite(pin, value);
  delay(10);
  gpio_hold_en((gpio_num_t)pin);
}
```

### Debug Message

```plain
-
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-11-21:
Have you tried with latest core 2.0.14?

ViniP8, 2023-11-21:
Yes, with 2.0.14 the problem persists

I suspect it is a hardware defect, the code below also results in Pin 13 outputing 2V and Pin 10 outputing 1V (after the 3s delay. If I only enable Pin 13, it ouputs 3.3V correctly). The strange thing is that until yesterday it was working and nothing was changed, and the consumption on these pins was less than 5mA

`  pinMode(15, OUTPUT);
  digitalWrite(15, 1);
  pinMode(10, OUTPUT);
  digitalWrite(10, HIGH);
  delay(3000);
  pinMode(13, OUTPUT);
  digitalWrite(13, LOW);
  delay(1000);`

lucasssvaz, 2023-11-21:
I've just tested with a regular ESP32S2 devkit and it works as expected. Unfortunately I don't have the Lolin board to test it. If anyone from the community could test it please, it would be much appreciated.

TD-er, 2023-11-22:
Is it possible those pins are being slightly pulled-up?
For example some external circuit pulling as some other parts are being powered compared to when the board is running.
Or a present pull-up resistor?

If so, then I guess you will for sure have some leak current and thus not the lowest possible deep sleep current.

I don't know how the ESP does keep the pin locked in a specific state during deep sleep.
If this is done by switching some pull-up or -down resistor over the GPIO pin, then I can imagine it could end up at those "in-between" states when another external pull-down or -up resistor is present.

ViniP8, 2023-11-22:
The board was being used without other components when testing the above code. 

I re-soldered the board and the problem stopped occouring with pins 10 and 13 and it started to happen to pins 8 and 10 (1V and 2V when setting one to HIGH and another to LOW, with the board heating a bit). The fault was probably in the soldering or in the hardware itself, I have already ordered another one for testing

As it apparently is not a software error, feel free to close the issue

Thanks to everyone who responded!

closed 2023-11-22 by Jason2866