espressif/arduino-esp32 #8911: Unable to receive images when BLE is used

Abhay-2412, 2023-11-23:
### Board

ESP32 wrover

### Device Description

ESP32 wrover with ov5640 camera and some i2c devices

### Hardware Configuration

camera and gpio expander 

### Version

v1.0.6

### IDE Name

Arduino IDE

### Operating System

windows 11

### Flash frequency

80Mhz

### PSRAM enabled

yes

### Upload speed

921600

### Description

Greetings,

I have successfully implemented a codebase that facilitates the reception of images captured by an ESP32 WROVER module within my application via WiFi. Initially, the WiFi credentials were hardcoded for connectivity. I am now seeking to enhance the code by integrating Bluetooth Low Energy (BLE) functionality to dynamically update the WiFi credentials. However, upon enabling BLE, I encountered an issue where the image reception functionality ceased to operate as expected.

To address this, I took measures to gracefully deinitialize BLE immediately after obtaining the necessary credentials, followed by a deliberate delay to ensure proper synchronization. Despite these efforts, the image reception functionality remains compromised.

I am seeking guidance on a more effective approach to address this issue, ensuring a seamless transition from BLE-based credential updates to the established image reception flow. Your insights and recommendations would be greatly appreciated.

Thank you.

### Sketch

```cpp
will provide if necessary
```

### Debug Message

```plain
no useful logs
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

Jason2866, 2023-11-23:
Arduino core 1.0.6 is not supported anymore. Please try with actual core 2.0.14

SuGlider, 2023-11-24:
> I am now seeking to enhance the code by integrating Bluetooth Low Energy (BLE) functionality to dynamically update the WiFi credentials. However, upon enabling BLE, I encountered an issue where the image reception functionality ceased to operate as expected.

It sounds like a lack of memory issue (HEAP or STACK).
Given that the board used is an ESP32 WROVER (with PSRAM), I'd say that moving to Arduino Core 2.0.14 would fix the issue, given that the memory management is better in this version when compared to Arduino Core 1.0.6.

Abhay-2412, 2023-12-06:
hello i followed your suggestion and shifted to 2.0.14 but now iam getting below errors

```
12:52:38.077 -> E (71005) cam_hal: cam_dma_config(330): DMA buffer 96000 Byte malloc failed, the current largest free block:65524 Byte
12:52:38.077 -> E (71005) cam_hal: cam_config(399): cam_dma_config failed
12:52:38.077 -> CORRUPT HEAP: Bad head at 0x3f8009c8. Expected 0xabba1234 got 0x3f800014
12:52:38.077 -> 
12:52:38.077 -> assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)
12:52:38.125 -> 
12:52:38.125 -> 
12:52:38.125 -> Backtrace: 0x40083fa5:0x3ffcdf50 0x40095c85:0x3ffcdf70 0x4009b44d:0x3ffcdf90 0x4009b045:0x3ffce0c0 0x40084515:0x3ffce0e0 0x4009b47d:0x3ffce100 0x400d5c77:0x3ffce120 0x400d640b:0x3ffce140 0x400d6510:0x3ffce160 0x400d3b5b:0x3ffce1a0 0x400d56df:0x3ffce1e0 0x400e2aaa:0x3ffce250
12:52:38.125 -> 
12:52:38.125 -> 
12:52:38.125 -> 
12:52:38.125 -> 
12:52:38.125 -> ELF file SHA256: e8104db5fea11e34
12:52:38.125 -> 
12:52:38.223 -> Rebooting...
```

me-no-dev, 2023-12-06:
you are running out of memory. You have PSRAM on your board?

me-no-dev, 2023-12-06:
you don't do it by `psramInit()`, but by enabling it in the board menu. Your log above shows that PSRAM is not enabled.

SuGlider, 2023-12-07:
>12:52:38.077 -> E (71005) cam_hal: cam_dma_config(330): DMA buffer 96000 Byte malloc failed, the current largest free block:65524 Byte
12:52:38.077 -> E (71005) cam_hal: cam_config(399): cam_dma_config failed

DMA is usually allocated only in SRAM (HEAP). I'd say that allocating 96000 bytes should be done somehow by spliting it into Lines of maybe 480 pixels? 16 bit per pixel at 96000 means 100 lines in an image.

The conitnuous maximum block is about 64KB. 
You can try to find tha place where the allocation of this 96000 bytes is happening and change it to less, if possible.

VojtechBartoska, 2024-01-31:
Closing as expired, no answer.

closed 2024-01-31 by VojtechBartoska