espressif/arduino-esp32 #8913: WDT triggers when running empty sketch on core 0, not on core 1

lwalbert, 2023-11-23:
### Board

ESP32 WROOM-32

### Device Description

Devkit C

### Hardware Configuration

NA

### Version

v2.0.14

### IDE Name

Arduino IDE  1.8.13

### Operating System

macOS Catalina 10.15.7

### Flash frequency

80MHz

### PSRAM enabled

no

### Upload speed

921600

### Description

Hi,
When selecting "Arduino runs on:Core0" from tools menu in Arduino IDE and downloading a empty sketch, the WDT gets triggered on only on core0.

Selecting "Arduino runs on:Core1" and downloading the empty sketch, everything seems to run fine.
Adding a delay(1) ,  and downloading to core0 solved the issue of WDT triggering.

I was expecting an empty sketch to "run" behave similar on both cores without WDT triggering.

maybe a priority bug when core0 is selected?

### Sketch

```cpp
void setup() {
  // put your setup code here, to run once:
Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
//delay(1);
}
```

### Debug Message

```plain
E (10090) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (10090) task_wdt:  - IDLE (CPU 0)
E (10090) task_wdt: Tasks currently running:
E (10090) task_wdt: CPU 0: loopTask
E (10090) task_wdt: CPU 1: IDLE
E (10090) task_wdt: Aborting.

abort() was called at PC 0x400d84e5 on core 0

Backtrace: 0x4008340d:0x3ffbe9cc |<-CORRUPTED

ELF file SHA256: 355049df55a228ff

Rebooting...
```

### I have checked existing issues, online documentation and the Troubleshooting Guide

- [X] I confirm I have checked existing issues, online documentation and Troubleshooting guide.

me-no-dev, 2023-11-23:
yes and that is OK. WDT is disabled on Core1 to help Arduino users use blocking code in their sketches. For Core0 WDT is left enabled, because networking is running there. If you add `delay(5);` in the loop, it will stop triggering

closed 2023-11-23 by me-no-dev

anettleship, 2023-12-15:
@me-no-dev I'm trying to puzzle through this issue on another project... https://github.com/interactionresearchstudio/ESP32-SOCKETIO/issues/63

Essentially, it's a completed project, not updated in a few years. They provide pre-built binaries and their own uploader, which works fine, but when I compile from source and upload to my esp32, I get a watchdog error and the device resets.

My question: where in the arduino-esp32 code is the part which sets the watchdog to be enabled on Core0, and disabled on Core1?

I'm trying to better my understanding in general, and also to discover if it is possible that the error I am seeing below is a result of the watchdog being previously disabled for both cores when the ESP32-SOCKETIO project was in development, and when the stock binaries were built, but now enabled for Core0, resulting in the behaviour that is the root cause of my bug.

```
E (43774) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (43774) task_wdt:  - async_tcp (CPU 0/1)
E (43774) task_wdt: Tasks currently running:
E (43774) task_wdt: CPU 0: IDLE
E (43774) task_wdt: CPU 1: **loopTask**
E (43774) task_wdt: Aborting.

abort() was called at PC 0x400f3d08 on core 0

Backtrace: 0x40083a75:0x3ffbecbc |<-CORRUPTED

ELF file SHA256: a3d64e267fd50348

Rebooting...
ets Jun  8 2016 00:22:57

```

Note, I've uploaded with settings that specify running on Core 1 (the default) and on Core 0, and if I switch to Core0, I can see the loopTask running on the correct core, but core 0 still calls the abort() and reboots the device:

```
E (18008) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (18008) task_wdt:  - IDLE (CPU 0)
E (18008) task_wdt: Tasks currently running:
E (18008) task_wdt: CPU 0: **loopTask**
E (18008) task_wdt: CPU 1: IDLE
E (18008) task_wdt: Aborting.

abort() was called at PC 0x400f3f70 on core 0

Backtrace: 0x40083a75:0x3ffbecbc |<-CORRUPTED

ELF file SHA256: 21361f46454692e7

Rebooting...
ets Jun  8 2016 00:22:57

```

I have found I can prevent the reboot by importing the watchdog module esp_task_wdt.h and disabling the watch dog, or setting a longer timeout. However, if I am to justify adding this to the project's codebase, I need to be able to demonstrate understanding of why this change is necessary!

Thanks very much for your assistance

me-no-dev, 2023-12-15:
@anettleship by default WDT on Core0 is enabled and on Core1 disabled. It is disabled only so that you can do Arduino shenanigans on Core 1. Core0 stays guarded to ensure that WiFi/BT are working properly and have time to handle the traffic.
You can turn off WDT on core0, but I do not recommend it. API is here: https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal.h#L110